OpenZWave/open-zwave #1892: Sunricher Z-Wave RGBW Wall Controller Discussion

petergebruers, 2019-07-21:
I was asked by @gizmocuz (Domoticz) to have a look at this RGBW remote control device known as "RGBW Wall Controller" manufacturer type designation SR-ZV9003T4-RGBW-EU. Also Sold under different brand names.

I have added a device configuration for it to get recognized and allow for setup of this device.

Node Identity Codes: 0330:0300:a107

Info and manual

https://products.z-wavealliance.org/products/2945

This device warrants some discussion because it is somewhat special: it does not have any physical output, nor does it have anything that can be (virtually) controlled. It can be seen as a "remote controller for an RGBW controller".

What makes it special? These are the device CCs that the node can send:

COMMAND_CLASS_SWITCH_MULTILEVEL
COMMAND_CLASS_COLOR
COMMAND_CLASS_SCENE_ACTIVATION

It does not implement these CCs as "controlled", only as "controlling" which means it can only use those CCs to send data to another device, implementing those CCs

How does that work? The manual actually quite clearly lists what it does.

The buttons "1" to "4" can send 2x click to Lifeline, and OpenZWave 1.6 supports Central Scene and will report this double click. There is no "single click" because this has a different meaning... It selects a group (2 to 5) to send Multilevel and Color commands to.

Press a number button (eg "1") to activate group.

1.Press and hold down button "1" to switch the associated devices with a Basic Set commands.
2.Press PWM1-PWM4 buttons, direct control of associated devices using Color Switch Set commands.
3.Press and hold down PWM1-PWM4 buttons, direct control of associated devices using Color Switch Start Level Change commands.
4.Short touch or slide touch the “Color Wheel”, direct control of associated devices using Color Switch Set commands.
5.Press and hold down brightness button, direct control of associated devices using Multilevel Switch Start Level Change commands.
6.Short press "play/pause" button, direct control of associated devices using Configuration Set commands with Parameter Number = 5, Size = 1, toggling through values (1-0-2-0-3-0-4-0-5-0-6-0-1-...). Press and hold down "play/pause" button direct control of associated devices using Configuration Set commands with Parameter Number = 6, Size = 1 with varying values.

Note that the play/pause button sends a parameter change to 5 and 6, which only works on (matching) RGBW Sunricher RGBW devices! On those devices the button cycles through predefined programs. I have not tested this.

I've tested this, by associating it with a FIBARO RGBW Controller FGRGBWM-441.

If you use the device in this way, the controlled node FGRGBWM-441 will reflect what the panel has sent (and that can be picked up by the software based on OpenZWave).

But what if you would like your gateway software to use such a device, without having to buy an RGBW module?

At the moment, that is not possible. Although you can associate a one of the groups 2-5 to the controller, the data received by OpenZWave will be unhandled, because OpenZWave does not "act like a RGBW module".

It would be possible to support this, because OZW can decode the received data, but there are no facilities to pass this information to the host application (eg Domoticz).

On a more technical level, if you add controller (1.1) to group 2 and trigger the color wheel, OpenZWave receives valid data:

2019-07-09 10:28:00.831 Detail, Node002,   Received: 0x01, 0x16, 0x00, 0x04, 0x00, 0x02, 0x0e, 0x60, 0x0d, 0x00, 0x01, 0x33, 0x05, 0x03, 0x02, 0xd8, 0x03, 0x00, 0x04, 0x28, 0x01, 0xca, 0x00, 0x86

This is a ColorCmd_Set = 0x05 - but this is normally a command sent by OZW, not handled by OZW, so it is not decoded and no values are created for this.

Note: dimming (brightness) uses "Start Level Change" and "Stop Level Change" - they are timing sensitive.

For sake of completeness, the "basic" CC does have code to handle received Set commands, OZW_Log.txt snippet:

2019-07-09 10:58:40.420 Detail, Node002,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x02, 0x07, 0x60, 0x0d, 0x00, 0x01, 0x20, 0x01, 0xff, 0xca, 0x00, 0x89
2019-07-09 10:58:40.420 Detail, 
2019-07-09 10:58:40.420 Info, Node002, MultiChannelEncap with endpoint set to 0 - Send to Root Device
2019-07-09 10:58:40.420 Info, Node002, Received Basic set from node 2: level=255.  Sending event notification.
2019-07-09 10:58:40.420 Detail, Node002, Notification: NodeEvent
2019-07-09 10:58:40.420 Info, Notification: Node Event Home e9e2c3c3 Node 2 Status 255 Genre basic Class NO OPERATION Instance 1 Index 0 Type bool

To summarize: at the moment, the "Sunricher Z-Wave RGBW Wall Controller" can be added to OpenZWave to control an RGBW device. It cannot be used as a standalone device to directly control an application based on OpenZWave.

Fishwaldo, 2019-07-22:
OZW was never written with the intention to behave as a Node, but as the gateway, hence why it doesn't generally handle SET messages. 

While it could be done, I'm not sure its worth the effort at this point.

gizmocuz, 2019-07-22:
Thanks Peter for the research!

I can understand the point of Fishwaldo.

It would have been nice as it is a node with lots of possibilities...

petergebruers, 2019-07-22:
Yes, while it could be done, I guess there are more pressing issues...

I am not sure if we should keep this issue "open", to warn users about the specifics of this device. I am kind of pro keeping it open. For example, a Domoticz user might buy this device and expect to "see" it do things in Domoticz (when used standalone).

We could also argue that "the device is compatible with OZW with its intended use of controlling another RGBW module"... And close this issue.

gizmocuz, 2019-07-22:
Well of course it is Domoticz specific... Anyone could buy this device.
Apparently it is mentioned/compatible already with all Home Automation applications:

https://www.robbshop.nl/touch-schakelaar-rgbw-robb-smarrt

Sure there are more pressing issues, i would close it and maybe in the feature when received ColorCmd_Set work open it again ;)

petergebruers, 2019-10-03:
Okay, nobody ever complained about this device, or maybe they did, but did not find this topic :wink: More than 2 months have passed, I think it is time to close this issue.

closed 2019-10-03 by petergebruers