OpenZWave/open-zwave #1895: Qubino ZMNHADX problem with end points and groups

GravityRZ, 2019-07-26:
Also filed this one under domoticz but i realize it might be zwave related

i am experiencing problems with a qubino ZMNHADX module

it was working perfectly but i had a power outage.
after that the switch is still working but the zwave groups seem to be empty.
i can however not add controller 1 to  any group anymore(i need 1 in group 1 and group nine)

i can add any other module to the group but the controller(1) i can not add)

eg with module 11 i can not add 1 to group 1 and 9 but i can add lets say 2(which does not exist)

i was running v4.10998 so i upgraded to the latest v4.11025 but the problem is still there

is there another way to add groups because from the user interface it is not working.

second question, where are those groups stored, inside the module because it makes no sense that a power outage can cause this

also stopped/started domoticz, disabled/enabled zwave controller and switched to polling but nothing helps

since i can not add the temp sensor to group 9 I went back to enable polling on this device
if i enable polling on this device AND i also enable endpoint I2 and I3 (without this temp is not coming in) i can get the temperature reading back in domoticz.

so for know the big problem is gone but it is really strange that i can not add controller 1 to any of the groups of this qubino device
[image]
[image]

possibly related to other endpoint problems with this device

petergebruers, 2019-07-26:
For reference. We have dicussed this module before, about one month ago you said this module was working correctly, see

https://github.com/OpenZWave/open-zwave/issues/1820

I do not have such a module, I cannot test...

I would first try to find out if the "unable to add controller to group 9" is a domoticz or ozw problem. Please enable debug logging. Then use both domoticz and ozw control panel. If it works on ozwcp then it is domoticz issue. Please drag and drop problematic OZW_log.txt here so we can have a look.

Groups are stored in non volatile memory of the device. A power cut is unlikely to change that but "never say never".

I am on holiday, my response to this issue may be slow...

GravityRZ, 2019-07-26:
ok i understand no hurry.
this is what is log in domoticz.log
`2019-07-26 14:45:57.657  Status: OpenZWave: added node: 1 (0x01) instance 0 in group: 1 of node: 11 (0x0b)`

this however does not reflect in the groups screen (still not 1 in group 1 of module 11)

there is nothing in the OZW_log, only this but not at the exact time

`2019-07-26 14:45:49.560 Always, 
2019-07-26 14:46:36.303 Error, Node011, ERROR: Dropping command, expected response not received after 1 attempt(s)
2019-07-26 14:46:36.303 Always, 
2019-07-26 14:46:36.303 Always, Dumping queued log messages
`

also it makes no difference if i add the controller to the group using domoticz or ozwpanel, same result.

petergebruers, 2019-07-28:
I have had a look at the device manual, and it seems like a complex device, having different groups per endpoint... For example root device has groups 1 to 9 and 9 indeed says "multilevel sensor report (triggered at change of temperature sensor) up to 16 nodes." but then endpoint 4 has only "Group 2: multilevel sensor report (triggered at change of temperature sensor) up to 16 nodes.". That seems like an odd setup imho but might not be relevant at all to this issue. Anyway, afaik you cannot select group 2 of endpoint 4, only the 9 groups on "root" will be available

I'll try to add some devices to my test network and see if I can detect a similar problem...

I'll tag @Fishwaldo - he might own this device...

GravityRZ, 2019-07-28:
ok thanks Peter. i know that in the beginning(zwave v1.4) it also took a lot of time mapping the endpoints to get things to work. back then i discovered that i could only get the temp sensor to work if i enabled the I2 and I3 end points en put the device to polling. after a while it also started top work like it should (controller 1 to group 1 and 9)  and i could disable I2,I3(do not use them) and disabled polling. now we are back to that same point again. thanks for your help

petergebruers, 2019-07-28:
I have been testing the official open-zwave-control-panel and I see at least one bug. It has to do with how it treats "multi instance" associations. The effect of that is, when you add "1" to a group it may actually be stored as "1.1" which is not exactly the same but let's call them "equivalent" (little white lie). If it is stored that way, ozwcp won't show it, event when it's correctly stored on the device.

This does not explain your problem but I'll have to improve ozwcp...

The bug is in there since... I think... Switch to OZW 1.6 but I am not sure. Anyway it is not new so the version of ozwcp embedded in Domoticz probably does not handle this either. Maybe on OZW 1.4 this worked...

I can't say if the "group editor" in Domoticz handles this, I'll have a look after I've fixed ozwcp.

I can think of 2 possible ways to add your controller to that group... But it is a bit cumbersome.

- If you have a test system, or a virtual machine, or some setup that allows you to run the Domoticz 1.4 stable, then you could attach your Z-Wave dongle and use it to set group 9
- If you own a PC you can use the Silabs "PC Controller" and use its group editor. But then you'll have to learn a new tool

Because the groups are stored on the device, as long as you do not try to edit them with Domoticz Dev or ozwcp it should "stick".

I am talking out loud... I am only trying to find possible workarounds.

petergebruers, 2019-07-29:
I've been debugging ozwcp a bit... It has issues with groups so in this is it is not a good test, not until I've merged the PR. I can merge my own PR but maybe I'll do one more test before I do that.

@GravityRZ do you have the means to build ozwcp from source? Have you ever built OpenZWave and/or the control panel? You do not have to do that, but I would be interested in the results...

Debugging this helps me understand the possible pitfalls with multi-channel association and I'll have a look at the Domoticz group editor this week. That said, I don't think there are big issues in OpenZWave itself, instead I think it is the editing of the associations that fails. Or maybe it a peculiarity of this device...

GravityRZ, 2019-07-30:
Hi Peter. I do not have the meand or skills to build it myself.

petergebruers, 2019-08-02:
Two things...

1) about Domoticz

I have checked the group editor in Domoticz, and because it allows you to enter the node and endpoint ID instead of offering a selection list (like ozwcp) it does not have any issue.

You can try to enter
1
1.0
1.1

The only possible issue I see is you have to click the  blue "Refresh" button, bottom right on that page. The result of the change does not appear automatically.

I expect domoticz to show "1.1" because OZW has some logic to change endpoint 0 to 1.

Can you please try that?

2) The config file already specifies to automatically add the controller to group 9, it is this part of the config file:

`<Group auto="true" index="9" label="Multilevel sensor report" max_associations="16"/>`

Should work out of the box so to speak.

If the you were unable to set the association (and make the node report temperature) please try these steps to re-interview the device:

- Disable polling of the sensor.
- In domoticz, hardware, select OpenZWave en click "setup"
- Select the controller node and below that set "2. Enable Debug: " to "enabled"
- select the offending node and click the "Refresh Node Info" button
- wait 30 seconds and click "refresh"

Now do something to make it report temperature eg heat or cool the sensor...

Please zip the OZW_Log.txt and drop it here if it still does not work.

GravityRZ, 2019-08-09:
just got back from holliday.
i tried the first part but really there is no way to add either 1, 1.0 or 1.1 using either the ozwpanel or through domoticz and refresh

here is the ozw_log file

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3486181/OZW_Log.txt)

GravityRZ, 2019-08-09:
here is a second file
after the refresh i added 1 to both group 1 and group 9 of module 11
after that i changed the temp.
it looks like it is adding the controller to the groups but it is not visible in domoticz nor does it seem to work
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3486362/OZW_Log.txt)

petergebruers, 2019-08-10:
Thank you for the logs. I'll try to have a look at them today.

petergebruers, 2019-08-11:
Hi! Short version of my analysis:

I would argue nothing is wrong with OpenZWave, Domoticz or the config file... I think the real cause of the problems might be this: you said: "it was working perfectly but i had a power outage."

Based on the log info, it seems the device does not accept the "multi channel set" command which it clearly should and which also worked about a month ago. This is pure speculation, but maybe the (sudden) loss of power caused the dive to write something unexpected to its non volatile memory, which causes the abnormal behaviour(s). It is rare, but I've heard stories of this before.

The device always responds by saying its groups are empty.

What I would do is exclude and include this device, node 11... Which is annoying but I cannot think of anything else. The manual is specific about how to do "Module Exclusion/Reset". I quote the important parts from the manual

"Connect module to power supply"
"press service button S for more than 6 second or press push button I1 five times within 3s (5 times change switch state within 3 seconds) in the first 60 seconds after the module is connected to the power supply."

That is important, because they continue to say:
"By this function all parameters of the module are set to default values and own ID is deleted.
If service button S is pressed more than 2 and less than 6 seconds (or if push button I1 is pressed three times within 3s) module is excluded, but configuration parameters are not set to default values."

Long story:

Let's have a look at the first log file

```
2019-08-09 13:35:01.910 Info, Node011, Loading Cache for node 11: Manufacturer=Qubino (Goap), Product=ZMNHADx Flush 1 Relay
2019-08-09 13:35:01.910 Info, Node011, Node Identity Codes: 0159:0002:0052
...
2019-08-09 13:35:09.139 Detail, Node011, QueryStage_Associations
2019-08-09 13:35:09.139 Info, Node011, Number of association groups reported for node 11 is 9.
...
2019-08-09 13:35:09.267 Detail, Node011,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x0b, 0x05, 0x8e, 0x03, 0x01, 0x01, 0x00, 0x73
2019-08-09 13:35:09.267 Detail, 
2019-08-09 13:35:09.267 Info, Node011, Response RTT 127 Average Response RTT 127
2019-08-09 13:35:09.267 Info, Node011, Received Multi Instance Association report from node 11, group 1
2019-08-09 13:35:09.267 Info, Node011,   The group contains:
2019-08-09 13:35:09.267 Info, Node011, Get MultiChannelAssociation for group 2 of node 11
...
019-08-09 13:35:11.897 Detail, Node011,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x0b, 0x05, 0x8e, 0x03, 0x09, 0x10, 0x00, 0x6a
2019-08-09 13:35:11.897 Detail, 
2019-08-09 13:35:11.897 Info, Node011, Response RTT 189 Average Response RTT 191
2019-08-09 13:35:11.897 Info, Node011, Received Multi Instance Association report from node 11, group 9
2019-08-09 13:35:11.897 Info, Node011,   The group contains:
2019-08-09 13:35:11.897 Info, Node011, Querying associations for node 11 is complete.
```

Manual decode of packet:

0x01, 0x0b, 0x00, 0x04, 0x00, 0x0b, 0x05, 
SOF, LEN = 11, request, data, rxStatus, sourceNode 11, cmdLength 5,

0x8e, 0x03, 0x01, 0x01, 0x00, 0x73
MULTI_CHANNEL_ASSOCIATION, MULTI_CHANNEL_ASSOCIATION_REPORT, Grouping Identifier 1, Max Nodes Supported 1, Reports to Follow 0

Conclusion, at 13:35:11.897 Group 1 and 9 are empty

OZW proceeds to auto-add node 1 endpoint 1 (ignore "instance 1" in the log message, it is endpoint 1)

```
2019-08-09 13:35:11.897 Info, Node011, Adding the controller to group 1 (Lifeline) of node 11
2019-08-09 13:35:11.897 Info, Node011, MultiChannelAssociation::Set - Adding instance 1 on node 1 to group 1 of node 11
2019-08-09 13:35:11.897 Detail, Node011, Queuing (Send) MultiChannelAssociationCmd_Set (Node=11): 0x01, 0x0d, 0x00, 0x13, 0x0b, 0x06, 0x8e, 0x01, 0x01, 0x00, 0x01, 0x01, 0x25, 0x7d, 0x3a
...
2019-08-09 13:35:11.897 Info, Node011, Adding the controller to group 9 (Multilevel sensor report) of node 11
2019-08-09 13:35:11.897 Info, Node011, MultiChannelAssociation::Set - Adding instance 1 on node 1 to group 9 of node 11
...
```

Send seems OK:

```
2019-08-09 13:35:12.277 Info, Node011, Sending (Send) message (Callback ID=0x7d, Expected Reply=0x13) - MultiChannelAssociationCmd_Set (Node=11): 0x01, 0x0d, 0x00, 0x13, 0x0b, 0x06, 0x8e, 0x01, 0x01, 0x00, 0x01, 0x01, 0x25, 0x7d, 0x3a
2019-08-09 13:35:12.277 Info, Node011, Encrypted Flag is 0
[... 11 lines cut ...]
MULTI_CHANNEL_ASSOCIATION_SET_MARKER, Node 1, Endpoint 1, txOptions, funcID

Seems correct to me.

OZW now verifies by reading back the grouo:

```
2019-08-09 13:35:12.402 Info, Node011, Sending (Send) message (Callback ID=0x7e, Expected Reply=0x04) - MultiChannelAssociationCmd_Get (Node=11): 0x01, 0x0a, 0x00, 0x13, 0x0b, 0x03, 0x8e, 0x02, 0x01, 0x25, 0x7e, 0x38
...
2019-08-09 13:35:12.587 Detail, Node011,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x0b, 0x05, 0x8e, 0x03, 0x01, 0x01, 0x00, 0x73
2019-08-09 13:35:12.587 Detail, 
2019-08-09 13:35:12.587 Info, Node011, Response RTT 184 Average Response RTT 186
2019-08-09 13:35:12.587 Info, Node011, Received Multi Instance Association report from node 11, group 1
2019-08-09 13:35:12.587 Info, Node011,   The group contains:
2019-08-09 13:35:12.587 Detail, Node011,   Expected reply and command class was received
```

Manual decode...

0x05, 0x8e, 0x03, 0x01, 0x01, 0x00
length 5, MCA, Report, Grouping Identifier 1, Max Nodes Supported 1, Reports to Follow 0

The group is empty! This would be easier to debug if the device could report why it did not accept the "set", but there is no such thing (a firmware developer could send debug output over UART but that is of no help here).

The other logs add the same "1.1" and the result is always an empty report.

GravityRZ, 2019-08-11:
ok thanks Peter.
i rather not exclude/include because the device id will change and i do not want that before.

i remember that when i upgraded from zwave 1.4 to 1.6 version there were a lot of problems in the beginnig which forced met to go right back to 1.4
one of the things i noticed one time with the earlier zw1.6 versions is that the controller is was displayed as 1.1.

woul di t be an id to make a backup, reinstall aan old 1.4 version and see if the groups appear over there.
this way we can rule out if it is caused by the migration to 1.6

also please note that i can add any other number to the groups and they will appear on node 11
only number 1 i can not add

petergebruers, 2019-08-12:
> i rather not exclude/include because the device id will change and i do not want that before.

I understand. There might be a trick with powering down Domoticz, resetting the module and then do "replace failed node" to keep the node ID but I haven't tried that myself so don't do that. I am only mentioning this because it is an interesting idea

> i remember that when i upgraded from zwave 1.4 to 1.6 version there were a lot of problems in the beginnig which forced met to go right back to 1.4
> one of the things i noticed one time with the earlier zw1.6 versions is that the controller is was displayed as 1.1.

Might be related, but indeed Domoticz should display "1.1" - even when you only type "1" in the browser, OZW adds the ".1". In fact, due to the "auto" addition to lifeline you should not need to type anything

> woul di t be an id to make a backup, reinstall aan old 1.4 version and see if the groups appear over there.
> this way we can rule out if it is caused by the migration to 1.6

Yes, but seems like a lot of work, unless you have a second system or virtual machine to run that 1.4 on.

> also please note that i can add any other number to the groups and they will appear on node 11
> only number 1 i can not add

Yes. It is almost as if it specifically does not accept "1.1". 

I am running out of ideas... Going to call help from Justin

@Fishwaldo - do you happen to own such a module? If so, would it be possible to test if OZW adds the controller to group 1 and 9 (as set in the config file)?

GravityRZ, 2019-08-12:
well with a synology package it is really easy to back and forth.

i installed V4.10734(last build wilt zw 1.4)
also restored from a genuine 1.4 backup to make sure i do not get devices which were created under zw1.6

i was right. the groups pop up right away.
[image]

i think when i upgrade to 1.6 from this version it will still work but i have learned that this causes problems later on.
so from here i uninstall this version, reinstall 11025 and do a restore from a 11025 backup

do you want me to test anything while i am still running this 10734 zw1.4 version?

i already tried removing and adding node 1 to group 1 and 9 and that works like expected so under zw1.4 everything works like it should

petergebruers, 2019-08-12:
Thanks!
Okay, synology package, good to know...
I would stop this test and go back to your 1.6 production version

> i already tried removing and adding node 1 to group 1 and 9 and that works like expected so under zw1.4 everything works like it should

This may seem like a strange thing to say, but it does not surprise me and it is actually not completely ok 😄 

Let me explain... You see "1" and not "1.1" - there is a fundamental difference. The short version has a different meaning and is much older, it works with any device. But there might be other problems, especially with devices having multiple instances eg your I2 I3

I have thought about this a while ago and what you can do under 1.6 is adding a special flag called "ForceInstances false", this will force 1.6 to write "short" (aka single channel) associations, without a dot

You can try this: stop domoticz. in your ozwcache file, find the node, then search command class 142. Under that element, add the compatibility flag, like this:

```
  <CommandClass id="142">
    <Compatibility>
      <ForceInstances>false</ForceInstances>
    </Compatibility>
```

GravityRZ, 2019-08-12:
ok i will try that.
in case you need it. here is the OZW log from 1.4
excluded 1 from group 1 and 9
added 1 to group 1 and nine and did a temp change.

[OZW_Log14.txt](https://github.com/OpenZWave/open-zwave/files/3491411/OZW_Log14.txt)

will change back to  11025 and do the things you suggested.

petergebruers, 2019-08-12:
BTW for that change to be permanent, it would have to go in the ZMNHADx.xml file but let's find out first if it helps. Damn, it is sooo difficult for me to debug devices that I do not own... Sometimes I buy one, but in this case, I really have got enough relays.

GravityRZ, 2019-08-12:
ok. back at 11025, changed the things you said and YES. groups appear and temp is coming in.

this was in the cache file before i made the change

i removed the "<Compatibility />"  andthen added you compatability entry. is this correct or should i leave this one in as well

`            <CommandClass id="142" name="COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION">
                <Compatibility />
                <State>
                    <CCVersion>1</CCVersion>
                    <InNif>true</InNif>
                    <StaticRequests>5</StaticRequests>
                </State>
                <Instance index="1" endpoint="1" />
                <Instance index="2" endpoint="2" />
                <Instance index="3" endpoint="3" />
                <Instance index="4" endpoint="4" />
                <Associations num_groups="9">
                    <Group index="1" max_associations="1" label="Lifeline" auto="true" multiInstance="true">
                        <Node id="1" />
                    </Group>
                    <Group index="2" max_associations="16" label="basic on/off (Output Q)" auto="true" multiInstance="true">
                        <Node id="1" />
                    </Group>
                    <Group index="3" max_associations="16" label="Input I2 basic report" auto="false" multiInstance="true" />
                    <Group index="4" max_associations="16" label="Input I2 notification report" auto="false" multiInstance="true" />
                    <Group index="5" max_associations="16" label="Input I2 binary sensor report" auto="false" multiInstance="true" />
                    <Group index="6" max_associations="16" label="Input I3 basic report" auto="false" multiInstance="true" />
                    <Group index="7" max_associations="16" label="Input I3 notification report" auto="false" multiInstance="true" />
                    <Group index="8" max_associations="16" label="Input I3 binary sensor report" auto="false" multiInstance="true" />
                    <Group index="9" max_associations="16" label="Multilevel sensor report" auto="true" multiInstance="true">
                        <Node id="1" />
                    </Group>
                </Associations>
            </CommandClass>
        </CommandClasses>`

GravityRZ, 2019-08-12:
ok so what do  i need to change in the config file
entry 142 is not in there at all

after that i will do  a remove cache startt domoticz and see if it is included from the config file

petergebruers, 2019-08-12:
Two things...

- The groups are stored on the device itself. So unless you remove them, then test if the node really stops reporting state and temperature, the "1" may still be in there because you added it under OZW 1.4!
- The config file has the older `<CommandClass id="133">` - you can change that 133 into 142 and put the compatibility flag below that element.

GravityRZ, 2019-08-12:
ok changed the config file, deleted the cache file and it still works.

also removed node 1 from group 1 and 9: temperature reading stops
added node 1 to group 1 and 9: temperature starts coming in again.

petergebruers, 2019-08-12:
Thank you for reporting back. I could change the config file in the repo... But I am not confident about that I2 and I3 stuff. Do I recall correctly that you cannot test this (because you don't have physical access to those inputs)?

I would really like to test that device without that flag, but instead use "1.0" as node+endpoint identifier. But right now that is quite impossible, OZW (hard coded) changes that to "1.1". It might seem like a silly difference but I've checked several devices on my Fibaro HomeCenter 2 and when you add the controller, it consistently uses "1.0".  Sorry for talking out loud, this multi channel stuff is confusing.

Let me sleep on it...

Fishwaldo, 2019-08-13:
@GravityRZ what hardware/firmware version are you running? (The SxPx portion of the model number?) 

It works without issues on the version that Quibino sent me. Maybe there is a firmware bug etc.

Fishwaldo, 2019-08-13:
The version I’ve tested is S6P2.

GravityRZ, 2019-08-13:
[image]
S3P1

Fishwaldo, 2019-08-13:
@petrovcicklemen hi Klemen - is this a known issue? It’s not accepting a multichannel Association on Group 1 (But does work on with a standard association)

petrovcicklemen, 2019-08-14:
Hi all. I kindly ask you give me some time, until I can check this next monday, since we have national holiday till the end of this week.

petrovcicklemen, 2019-08-14:
> it also took a lot of time mapping the endpoints to get things to work. back then i discovered that i could only get the temp sensor to work if i enabled the I2 and I3 end points en put the device to polling. after a while it also started to work like it should (controller 1 to group 1 and 9) and i could disable I2,I3(do not use them) and disabled polling. now we are back to that same point again. thanks for your help

Meanwhile, I can comment on the quoted text. 

As you may or may not know, the majority of Qubino devices have multiple configurations. In their basic/default configurations (no additional inputs, no temp sensor), they are represented as non-multichannel devices. If the user enables additional inputs or connects, for example, the temperature sensor, these devices become multi channel (each inputs is represented by its own endpoint and so is the temperature sensor). 

When the device is included in its basic configuration, an "ordinary" lifeline is usually configured using ASSOCIATION_SET(groupId=1, nodeId=1). This results in the device sending its unsolicited messages (output state, power consumption, ambient temperature measurements) non MULT_CHANNEL encapsulated. If, however, the user enables any of the inputs or connects a temperature sensor, a MULTI_CHANNEL lifeline must be configured, using MULTI_CHANNEL_ASSOCIATION_SET(groupid=1, multichannelnodeid=1, endpointid=0), since this results in the unsolicited messages also being MC encapsulated (these messages now also contain the information of the source endpoint) and map to corresponding widgets, created by the gateway.

In the case described above, enabling input I2 and I3 results in the device representing itself as a multi channel device (CC_MULTI_CHANNEL is added to the NIF). The same applies, if the temp sensor is connected. The temperature sensor is mapped to the root and has its own endpoint. This endpoint can be enumerated differently, depending on the additionaly enabled inputs. If no inputs are enabled, then endpoint2 belongs to the temp sensor. If I2 is enabled, enpoint3 belongs to the temp sensor. If I2 and I3 are enabled, endpoint4 belongs to the endpoint.

So, basically, there could be three things which cause the temperature not being updated:
- a bug in the firmware
- an incorrectly configured lifeline (e.g. widget is created for endpoint4/temp, but the ordinary lifeline is configured, which results in the device sending non MC encapsulated reports, hence not updating the widget).
- a missing widget for a selected property

petergebruers, 2019-08-14:
Thank you @petrovcicklemen for explaining that. Not sure yet how to handle the different behavior. Try MCA and if that fails SCA?

Question: you say multichannelnodeid=1, endpointid=0 and OZW has used endpoint 1 instead of 0, it is hardcoded to do that if MCA CC exists. I've thought this to be an anomaly but it has been like that for quite a while (I only recently started contributing to OZW). Would ep 0 versus ep 1 make a difference? If I understand the spec correctly, ep 0 would be the equivalent of root.

petrovcicklemen, 2019-08-15:
> Thank you @petrovcicklemen for explaining that. Not sure yet how to handle the different behavior. Try MCA and if that fails SCA?

"If such a controlling node does not implement any End Points, it MUST NOT advertise the Multi Channel Command Class in its Node Information Frame (NIF) or S0/S2 Commands Supported Report Command. (3.2.2, SDS13783)" 

According to the specs text above, I think one way to go about this would be to check if the NIF contains CC_MULTI_CHANNEL. If it does, configure MCA, otherwise SCA.

> Question: you say multichannelnodeid=1, endpointid=0 and OZW has used endpoint 1 instead of 0, it is hardcoded to do that if MCA CC exists. I've thought this to be an anomaly but it has been like [...]

I think this has something to do with the following:

"End Point 1 MUST implement the primary application functionality of the actual Multi Channel
device. For backwards compatibility, the Root Device MUST mirror the application functionality of End Point 1 (3.2.1, SDS13783)"

Since the functionalities are mirrored, they send the same reports, but from different source endpoints. I think it then depends for which endpoint (root or endpoint1) the gateways/platforms create the widgets, that are then updated with the received reports.

petrovcicklemen, 2019-08-19:
I've tested this now and for me it also worked.

@GravityRZ : I suggest that you contact [Qubino support](https://support.qubino.com/support/home) and we can connect remotely to you, if you'd be ok with that, to see what is going on.

Fishwaldo, 2019-08-19:
> According to the specs text above, I think one way to go about this would be to check if the NIF contains CC_MULTI_CHANNEL. If it does, configure MCA, otherwise SCA.

we already do this. 

> "End Point 1 MUST implement the primary application functionality of the actual Multi Channel
device. For backwards compatibility, the Root Device MUST mirror the application functionality of End Point 1 (3.2.1, SDS13783)"

If only all vendors were compatible with this... we would be good. :) Unfortuantly there are a number out there that have different functionality on root versus endpoint 1 :(

Anyway - Closing this as it doesn't seem to be OZW related.

GravityRZ, 2019-08-19:
@petrovcicklemen 
Do you mean you tested it with that exact firmware or just with the latest firmware?

how would remote connecting work

> I've tested this now and for me it also worked.
> 
> @GravityRZ : I suggest that you contact [Qubino support](https://support.qubino.com/support/home) and we can connect remotely to you, if you'd be ok with that, to see what is going on.

petrovcicklemen, 2019-08-19:
@GravityRZ : With the latest. I was just interested in seeing, how the lifeline is configured, since this is usually the cause for measurements not being sent or the widget being updated.

My colleague would connect to your PC via Teamviewer, install the PC Controller application (mentioned above) and try to configure the appropriate lifeline that way.

GravityRZ, 2019-08-20:
would it be possible that you send me the software so i can try myself.
it does not seem to  be that difficult if you tell me what to do
the reason asking is that the zwave stick is connected to my synology so i need to reconnect it to my pc manually.

petrovcicklemen, 2019-08-20:
No problem. I'll ask my colleague to prepare a gif file, which shows the steps. You'll also receive the app.

GravityRZ, 2019-08-23:
just filed a ticket. thanks for the help

GravityRZ, 2019-08-30:
i was able to remove standard association and multi channel association 1 from the module and added
multi channel association 1.1 following the instructions @petrovcicklemen send me.

restarted domoticz and noticed that 1 is not visible anymore on group 1 nor can i add it(or 1.0 or 1.1)
hopefully this means 1.1 is set

controller 1 is still visible on group 9 and temperature is still coming in.

my worries: if i delete the controller from group9 i will loose temp reading again(because it is still the old 1 from v1.4)

how can i make sure the module is sending out multi assoc readings?
also i still can not add 1, 1.0 or 1.1 to either of the groups

so what does this change mean. that it is set permanent at 1.1 and it does not matter if i add groups or not under domoticz 

so at the moment i can switch on the module and get temp readings with this settings in the groups

if i normally can not do this when 1 is not in the lifeline then it probably works

do i need to remove the module from the cachefile and let it be added again(what happens with the groups)
[image]

[image]

GravityRZ, 2019-08-31:
to correct the issue i went back to v1.4, added 1 to group 1,2 and 9
uninstalled v1.4 and installed latest v1.6(11230) now everything is back but i think the problem is the same

i should be able to remove 1, add it again so it appears as 1.1 but i can not

please note a new node i added(19) it was standard put in 1 and i added 1 to group3(it appears as 1.1)

petergebruers, 2019-09-04:
@GravityRZ thank you for your detailed description, I understand what you are saying but I still have  not found the time to look at it in great detail. I am sorry about that... I'll come back to this.

GravityRZ, 2019-09-05:
ok no problem.

i deleted the cache file and recreated it
config file only has 133 in it.
also the force instances option is not in the config file

here is the part which is in the cache file. please note it has both 133 and 142 in it
probably because i enabled 1 from 1.4 build and 1.1 by using the zwave controller software
still it is strange that i can click 1 under group 1 and 9 and it dissapears but when i try to add it i can not add 1, 1.0 or 1.1 anymore while the unit seems to recognise multi channel association

>          <CommandClass id="133" name="COMMAND_CLASS_ASSOCIATION">
>                 <Compatibility />
>                 <State>
>                     <CCVersion>1</CCVersion>
>                     <InNif>true</InNif>
>                     <StaticRequests>5</StaticRequests>
>                 </State>
>                 <Instance index="1" endpoint="1" />
>                 <Instance index="2" endpoint="2" />
>                 <Instance index="3" endpoint="3" />
>                 <Instance index="4" endpoint="4" />
>                 <Associations num_groups="9">
>                     <Group index="1" max_associations="1" label="Lifeline" auto="true" multiInstance="true">
>                         <Node id="1" />
>                     </Group>
>                     <Group index="2" max_associations="16" label="basic on/off (Output Q)" auto="true" multiInstance="true">
>                         <Node id="1" />
>                     </Group>
>                     <Group index="3" max_associations="16" label="Input I2 basic report" auto="false" multiInstance="true" />
>                     <Group index="4" max_associations="16" label="Input I2 notification report" auto="false" multiInstance="true" />
>                     <Group index="5" max_associations="16" label="Input I2 binary sensor report" auto="false" multiInstance="true" />
>                     <Group index="6" max_associations="16" label="Input I3 basic report" auto="false" multiInstance="true" />
>                     <Group index="7" max_associations="16" label="Input I3 notification report" auto="false" multiInstance="true" />
>                     <Group index="8" max_associations="16" label="Input I3 binary sensor report" auto="false" multiInstance="true" />
>                     <Group index="9" max_associations="16" label="Multilevel sensor report" auto="true" multiInstance="true">
>                         <Node id="1" />
>                     </Group>
>                 </Associations>
>             </CommandClass>
>             <CommandClass id="134" name="COMMAND_CLASS_VERSION">
>                 <Compatibility />
>                 <State>
>                     <CCVersion>1</CCVersion>
>                     <InNif>true</InNif>
>                     <StaticRequests>5</StaticRequests>
>                 </State>
>                 <Instance index="1" endpoint="1" />
>                 <Instance index="2" endpoint="2" />
>                 <Instance index="3" endpoint="3" />
>                 <Instance index="4" endpoint="4" />
>                 <Value type="string" genre="system" instance="1" index="0" label="Library Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="3">
>                     <Help>Z-Wave Library Version</Help>
>                 </Value>
>                 <Value type="string" genre="system" instance="2" index="0" label="Library Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="Unknown">
>                     <Help>Z-Wave Library Version</Help>
>                 </Value>
>                 <Value type="string" genre="system" instance="3" index="0" label="Library Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="Unknown">
>                     <Help>Z-Wave Library Version</Help>
>                 </Value>
>                 <Value type="string" genre="system" instance="4" index="0" label="Library Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="Unknown">
>                     <Help>Z-Wave Library Version</Help>
>                 </Value>
>                 <Value type="string" genre="system" instance="1" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="4.05">
>                     <Help>Z-Wave Protocol Version</Help>
>                 </Value>
>                 <Value type="string" genre="system" instance="2" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="Unknown">
>                     <Help>Z-Wave Protocol Version</Help>
>                 </Value>
>                 <Value type="string" genre="system" instance="3" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="Unknown">
>                     <Help>Z-Wave Protocol Version</Help>
[... 19 lines cut ...]
>                 <State>
>                     <CCVersion>1</CCVersion>
>                     <InNif>true</InNif>
>                     <StaticRequests>5</StaticRequests>
>                 </State>
>                 <Instance index="1" endpoint="1" />
>                 <Instance index="2" endpoint="2" />
[... cut ...]
>                     <Group index="6" max_associations="16" label="Input I3 basic report" auto="false" multiInstance="true" />
>                     <Group index="7" max_associations="16" label="Input I3 notification report" auto="false" multiInstance="true" />
>                     <Group index="8" max_associations="16" label="Input I3 binary sensor report" auto="false" multiInstance="true" />
>                     <Group index="9" max_associations="16" label="Multilevel sensor report" auto="true" multiInstance="true">
>                         <Node id="1" />
>

petergebruers, 2019-09-06:
Ah, yes, we discussed "ForceInstances" (see <https://github.com/OpenZWave/open-zwave/issues/1895#issuecomment-520328820)> and because it is not in the config/qubino/ZMNHADx.xml file, it won't be in the ozwcache file if you refresh the node or delete that file.

I am still discussing this issue with @Fishwaldo - we would like to have a fix that does not require changes to config files. But it is tricky, we do not want to break support for existing devices.

I can update the config with that flag and post that here. Would you be willing to test it?

petergebruers, 2019-09-06:
Here's the file. Please stop Domoticz, unzip, overwrite the existing file in your domoticz/config/qubino folder with this one, start Domoticz and do "refresh node info" on this node (in hardware, openzwave panel of domoticz).
[ZMNHADx.xml.zip](https://github.com/OpenZWave/open-zwave/files/3582946/ZMNHADx.xml.zip)

GravityRZ, 2019-09-06:
yes with this file  only entry 142 is in the cache file

also i can remove 1 from groups(only tested on group2 to be save) and add it again(but it appears as 1, not 1.1)

so it looks like it is only using multichannel but still no 1.1

petergebruers, 2019-09-06:
Thank you for reporting back.

> yes with this file only entry 142 is in the cache file

Good.

> also i can remove 1 from groups(only tested on group2 to be save) and add it again(but it appears as 1, not 1.1)

Good.

> so it looks like it is only using multichannel but still no 1.

Correct! To be more precise: 1.1 won't be accepted by the device and 1.0 cannot be set by OpenZWave...

 Technically speaking, it uses "Multi Channel Assocation" but this bit in OpenZWave is hardcoded (simplified):

```
if (_instance == 0x00)
{
    ... prepare MultiChannelAssociationCmd_Set
    msg->Append(_groupIdx);
    msg->Append(_targetNodeId);
   ...
}
else
{
    ... prepare MultiChannelAssociationCmd_Set
    msg->Append(_groupIdx);
    msg->Append(0x00); // marker
    msg->Append(_targetNodeId);
    msg->Append(_instance);
    ...
}
```

So you see if you try to add Node 1 Endpoint 0 then that variable  _instance is zero and it uses MultiChannel command, but it does not send then "0" to the device. So it is not identical to what Clemen asked you to do... And if you try 1.1 then it uses the second part, but the devices does not accept it (that is in fact the crux of this issue).

That is the part I am discussing with @Fishwaldo and that is also why I asked if you could compile from source, because then I can make edits and you can test ;)

I hate to ask this... But can you remove the association form node 9, then confirm that you no longer get temperature reports? Then add back that "1' and check if it starts reporting temperature? I know it is very inconvenient, if it stops working, you have to take the stick to your windows machine and use the Silabs PC Controller to perform the procedure in 

<https://github.com/OpenZWave/open-zwave/issues/1895#issuecomment-526664420>

On the other hand... If this works I can push that configuration file to OpenZWave as a more permanent fix...

GravityRZ, 2019-09-06:
> I hate to ask this... But can you remove the association form node 9, then confirm that you no longer get temperature reports? Then add back that "1' and check if it starts reporting temperature? I [...]

i can do this but was it not the idea that this should work without the force instances.

so to be clear.
force instances is in the file, multi channel seems enabled (because i have put 1.1 in the device using the zwave controller) i will remove 1 from group 1,2 and 9) see if it stops the temp(removing 9 only will also tell me this
and then add 1 to group 1,2 and 9 and see if temp reading comes back.

petergebruers, 2019-09-06:
> i can do this but was it not the idea that this should work without the force instances.

OpenZWave seems to work out of the box with most devices, but not all, and you have found one of them... That flag exists to fix some devices.

> force instances is in the file, multi channel seems enabled (because i have put 1.1 in the device using the zwave controller)

You say "1.1" but I have re-watched Qubino's instruction and they Add Node ID 1 and End Point 0. Not Endpoint 1... At the end they do a "get" and they display the result which has payload 01 01 00 00 01 00 which means Group 1, Max assoc 1, 0 reports to follow, 00 is the MCA byte marker, node 1 , endpoint 0. So endpoint zero, not one.

If you look at the Domoticz Group tab, after removing all possible associations, do you still see anything? Like 1 or 1.1 or even 1.0? I think (after refreshing that page) you will see all groups are empty.

> see if it stops the temp(removing 9 only will also tell me this
Correct.

if the "groups" tab really showed no associations after removal, you can do the next step:

> and then add 1 to group 1,2 and 9 and see if temp reading comes back.

Yes, that would prove this flag works to fix this issue.

GravityRZ, 2019-09-06:
ok. here are the results

first let me point out i have already tested this before (look at the beginning of the thread) so this flag works and with it you are able to remove and add node 1 to groups.
the discussion then was that this is normal association, not multi channel association

now i have tested it again(but with 1.0 multi channel association zwave procedure in place)

i can add and remove node 1 from group 1, 2 and 9 and add it again(because of the flag)
removing node1 from group 9 still gives me temp readings and kWh readings
removing node1 from group 1 and 2 stops everything, no temp reading, no kWh reading

so the flag works but i wonder what adding 1.0 to multi channel association has done (besides giving me a temp readout without having node 1 in group 9)

petergebruers, 2019-09-06:
Thank you for thoroughly testing that... Yes you've tried it before, nevertheless I think we see slightly different behavior.

> removing node1 from group 9 still gives me temp readings and kWh readings

I kind of expected that, but I think at least 2 people told me that it is necessary to add node 1 to group 9 to get temperature readings. I expect "lifeline group" to report everything the device is capable of.

> removing node1 from group 1 and 2 stops everything, no temp reading, no kWh reading

This tells me, the controller was no longer in any group. Good. Without any association, the device does not know where to send reports.

> so the flag works but i wonder what adding 1.0 to multi channel association has done (besides giving me a temp readout without having node 1 in group 9)

The way I see it... Without that flag, OZW sends "1.1" and the device does not accept that. You were never able to add the controller to any group with OpenZWave 1.6. With that flag, it accepts 1.0 (but OZW sends it as 1, without endpoint). Your module was working previously because you upgraded from 1.4 which always sets "1" and it the config was still there... Then you had a power outage and my guess is the memory got corrupted and the group configuration was lost. When you tried to fix that, nothing worked because you were running OZW 1.6 instead of 1.4.

You were also able to set any other node, because the change of 1 -> 1.1 only applies to the controller node 1... The module would always accept eg node 2 or 2.1 (it does not check if that node actually exists).

GravityRZ, 2019-09-06:
ok so with the forceinstance flag it always works

that leaves me with one question.
was it necessary to add 1.0 to the multi channel association by using the pc controller software.
before it was also working with the flag(but maybe not multi channel

regarding adding node 1 to group 9
from the tests it looks like it is needed for standard association but not anymore for multichannel association.

maybe leave it standard for group 9 as well so it works for both standard and multi channel devices.

will you make this a permanent fix?

petergebruers, 2019-09-09:
Github auto-closed the issue but I'd like to add something ;)

> ok so with the forceinstance flag it always works

Thanks

> will you make this a permanent fix?

Yes. I've merged the config file into OpenZWave master branch.

> second question. i noticed that i had 1.1 on one group of a newly added device. i removed all the 1's from those devices and added them again for devices 11, 12 and 13
> they now appear as 1.1

That happens when OpenZWave 1.6 detects this device supports multi channel association and automatically changes your request to 1.1

> is this the way to go? eg should i do the same for al other devices?

I would argue, if everything works as expected, you don't have to do that. On the other hand... if you don't update them all then revisit that page in 6 months you are probably going to think "why do some nodes have 1 and others have 1.1?"

Maybe it helps when I give a bit of background on this "multi channel" thing.

First I'll talk a bit about Multi Channel, then Multi Channel Association. They are related, but not the same.

The reason “multi channel” exists is because without it, you cannot make a device with 2 relays, or 2 temperature sensors.
This is a while ago (years) but before the introduction of "multi instance" which evolved into "multi channel" A Z-Wave device would have a simple list of supported command classes. The devices can be complex, but it was impossible to define the same CC twice.

An example of such a complex device is the Aeotec MultiSensor 6: it supports multiple sensors because the Multilevel Sensor Command Class itself can report multiple sensors. It has only one instance of that class.

A simple device like a Door/Window sensor can report temperature, because a switch and a sensor use a different command class.

An example of a seemingly simple device that was impossible to make without "multi channel" support is a double relay. Pick any double relay. The Binary Switch Command Class only supports a simple on/off parameter. This means, you can only make it work with one relay.

One way to look at it is this: to uniquely define a property of a device, without "multi channel", you can do it like this:

- Your Z-Wave stick has a homeID and it defines the network.
- The device has a node ID - it defines an address on the wireless network
- The devices supports a list of "Command Classes" and they define what it can send and receive. Command classes vary greatly in complexity and capabilities.
- Per command class, Values get created for each interesting aspect of a command class. In OpenZWave terminology we say "Value", these get transformed into "devices" in Domoticz or "entities" in Home Assistant.

The MultiSensor supports temperature, illuminance and more because the Sensor CC can define multiple types of sensor.
A double really cannot work with this simple scheme, there is no way to define a 2nd relay.
A hypothetical device with TWO temperature sensors is also impossible to define: the Sensor CC can define a whole range of sensors, but only one per type.

To work around that, something is needed to define "multiple of X". Because Z-Wave is very backward and forward compatible, the developers have to take into account what existed at some point and think carefully about how to add functionality. The solution is the Multi Channel Command Class aka COMMAND_CLASS_MULTI_CHANNEL

It belongs to the category of "encapsulation commands" meaning that it wraps around existing data packets. Basically, it adds the "sub addressing" that was missing from the older spec.

So here's what an address in multi channel could look like

- Your Z-Wave stick has a homeID and it defines the network
- The device has a node ID - it defines an address on the wireless network
- If the device has COMMAND_CLASS_MULTI_CHANNEL - Add an extra identifier called an "endpoint". When you send data and it has an endpoint, wrap the command
- The devices supports a list of "Command Classes" per endpoint and they define what it can send and receive. Command classes vary greatly in complexity and capabilities.
- Per command class and per endpint, Values get created for each interesting aspect of a command class.

End points start numbering at 0. The spec gives special meaning to 0 : "the Root Device MUST mirror the application functionality of End Point 1"

In fact, the spec takes a few pages to elaborate on what endpoints should and should not be/do.

Now you can make a double relay... It has 3 endpoints

0 first relay = the same as if the device did not support "multi channel"
1 first relay
2 second relay

You'll often find the "dot notation" when instances (aka endpoints) are involved...

The controller knows, by interviewing, if a device supports Multi Channel. And OpenZWave will do the encapsulation for you. If you see a double relay in Domoticz, and you toggle the 2nd relay, Domoticz will ask endpoint X on node Y to toggle.

But how does the device know if the other party supports endpoints? Specifically when the device reports unsolicited? That is what the groups (associations) are for.
[... 7 lines cut ...]
Let me give an example.

If you buy a “Neo Coolcam Wall Plug” you will find that it does not support COMMAND_CLASS_MULTICHANNEL_ASSOCIATION and also does not have COMMAND_CLASS_MULTI_CHANNEL. At least not the version and firmware I have...

[... cut ...]
Does that matter? All devices should offer minimal support in non-multi-channel mode, you might never notice the difference.

For example, your Wall Plugs and Dimmers (if you only use S1) will work and report state...
The ZW100 Multi Sensor does not need MCA either (although it seems complex and you might mistakenly think it is a multi channel device)

Note: the description of COMMAND_CLASS_MULTICHANNEL_ASSOCIATION and COMMAND_CLASS_MULTI_CHANNEL is several pages long, my explanation is a simplification but I think it is quite useful to understand why sometimes A cannot send data to B...

GravityRZ, 2019-09-09:
thanks for the info.
i did some more testing

after an upgrade module 17 was suddenly displaying empty group 2 and 3
i added 1 and it also appeared as 1(not 1.1 anymore)
i did a refresh node info and after that i noticed both 1 and 1.1 on group 2 and 3.

apparently the module did not function as multi association but did after the refresh.
i could remove the 1 so now modules 18,18 and 19 have 1.1 on all groups

inspected the FGMS005-ZW5 module and that one sticked with 1
inspected the config file and it has force instances=false in it(probably because it did not work multi before)
i changed the config and removed the forceinstances, remoced the cacxhe file and let it rebuild.
after that i could also add 1.1 to group 5 of the module.
so i think this mofdule is also multi association but maybee it is save to let it stick to single.

petergebruers, 2019-09-09:
That is interesting information. I can explain the FGMS001-ZW, I am responsible for adding that flag because without it, auto association did not work (and because of that no reporting).

Discussion:

https://github.com/OpenZWave/open-zwave/issues/1830

In fact, that module should report perfectly with that config file (having force instances=false), no need to add anything to any group (so "1" in lifeline should be enough). I debugged this issue personally. If you are thinking, that issue sounds like my ZMNHADx problem, and has the same fix, I would say: "indeed" :wink:

Fishwaldo, 2019-09-09:
Just a side note - I plan to make “instances” or endpoints a first class citizen in the next “breaking API” change of OZW (1.7? 1.8?). That means that very function that takes/exposes Node ID’s, it would be represented as 1.0, 1.1 etc etc etc. (0 might pose problems due to some restrictions on early versions of the spec), and treat since channel devices as a special case, rather than the way we do it now where multichannel is the special case and single channel is the exception (as the way the market has evolved - most devices released today have multichannel capabilities). 

With 1.6 we took a “baby step” in this way with ForceInstances becoming default, and that change has thrown up a compatibility issue with a few devices and cause some confusion.

GravityRZ, 2019-09-09:
ok. i will do a test with the fgms001-zw5 device

first i will remove 1 from all other groups(only keep it in 1)
you state that because of the flag it will report lux, temp, motion  and tamper to 1 so it should still work.
if this works as expected i will remove the flag, cachefile
the unit should act as a multi assoc device, therefor i would only receive lux, temp, motion and temper only if i put 1(displayed as 1.1) in the appropriate groups

GravityRZ, 2019-09-09:
ok with forceinstances=false
temp, motion,tamper and lux are still coming in, also when i remove 1 from groups 2-5

remove forceinstances from config file
remove device 16 from cachefile
stop/start domoticz
can add 1.1 to any group but 1(lifeline)
motion coming in but no temp or lux
[image]

reversed everything
can add node1 again to group 1(old group 2-5 still at 1.1)
[image]

i think when i remove 1.1 from the group and then add it again it appears as 1
[image]
[image]

so the weird thing is indeed that it can work as a multi assoc device but when it is the controller can not be added to group 1

a bit different then the qubino device because with that one node 1 could not be aded to any group

petergebruers, 2019-09-09:
@Fishwaldo thanks for clarifying that. To end-users , Z-Wave looks as if it is one standard... If you dig into this or start developing you see it is an evolving standard and (sometimes slowly) change. Add to that the occasional bug and misinterpretation of the spec ...

@GravityRZ That sensor got the label "buggy" because of that and that is also why I changed the config file for it. It breaks a "be backwards compatibility rule". In the 1.4 era someone worked around that by adding the controller to a few groups. But there was still an issue with the device always returning one sensor ("Lux" from the top of my head) instead of reporting the other values. IMHO everything is working now.

petergebruers, 2019-10-10:
Not sure what I can do for you... If the module "forgets" its associations after a power cut, I don't see how that is caused by OZW.

I hate to say it, but I don't think I can help you with this.  I don't own the device, I cannot test. There is only so much I can do to help... Sorry about that.

The only thing I can suggest is to work around it by enabling polling at a reasonable pace (eg every 5 minutes). Depends on what you want to do with that temperature reading...

petrovcicklemen, 2019-10-10:
@GravityRZ: I suggest, that you write to our support again. We can connect to you, include the device into PC Controller, set the same association groups like you set them in Domoticz and simulate a power loss. We should be able to replicate the issue, if this is a problem of the device, which means this would also be a valid RMA case.

[4 more replies]

commit 2f2b314f (2019-09-09): Qubino ZMNHADx set ForceInstances to false to fix #1895
closed 2019-11-25 by Fishwaldo