OpenZWave/open-zwave #1896: Notification Type_ValueAdded, exception "Invalid ValueID passed to GetValueLabel"

gizmocuz, 2019-07-27:
OZW Version: Latest (from 21 July)
Node: Philio PMS02

When:
Happens after a include, or when forcing it to send it's Node Information

A Notification of type "Type_ValueAdded" is being send to the application with a Value of:

vID	{m_id=0x044c0010 m_id1=0x00010000 m_homeId=0xc1355dd9 }

commandclass = 0x30 (SENSOR_BINARY)
Instance = 1
Index = 1
ValueType = 0 (boolean)
ValueGenre = 1 (user)

When calling 'GetValueLabel' on this value a exception is thrown "Invalid ValueID passed to GetValueLabel"
The same happens when calling 'GetValueUnits'

Log from OZW:

> 2019-07-27 15:34:29.640 Info, Node004, Sending (Query) message (Callback ID=0x0e, Expected Reply=0x04) - ManufacturerSpecificCmd_Get (Node=4): 0x01, 0x09, 0x00, 0x13, 0x04, 0x02, 0x72, 0x04, 0x25, 0x0e, 0xbe
> 2019-07-27 15:34:29.641 Info, Node004, Encrypted Flag is 0
> 2019-07-27 15:34:29.652 Detail, Node004,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
> 2019-07-27 15:34:29.653 Detail, Node004,   ZW_SEND_DATA delivered to Z-Wave stack
> 2019-07-27 15:34:29.666 Detail, Node004,   Received: 0x01, 0x07, 0x00, 0x13, 0x0e, 0x00, 0x00, 0x02, 0xe7
> 2019-07-27 15:34:29.667 Detail, Node004,   ZW_SEND_DATA Request with callback ID 0x0e received (expected 0x0e)
> 2019-07-27 15:34:29.668 Info, Node004, Request RTT 23 Average Request RTT 23
> 2019-07-27 15:34:29.668 Detail, Node004,   Expected callbackId was received
> 2019-07-27 15:34:29.678 Detail, Node004,   Received: 0x01, 0x0e, 0x00, 0x04, 0x00, 0x04, 0x08, 0x72, 0x05, 0x01, 0x3c, 0x00, 0x02, 0x00, 0x02, 0xb3
> 2019-07-27 15:34:29.679 Detail, 
> 2019-07-27 15:34:29.680 Info, Node004, Response RTT 35 Average Response RTT 35
> 2019-07-27 15:34:29.681 Info, Node004,   Opening config param file Config/philio/psm02.xml
> 2019-07-27 15:34:29.683 Detail, Node004, Initial read of value
> 2019-07-27 15:34:29.683 Info, Node004, Queuing Lookup on 0002.0002.013c.db.openzwave.com for Node 4
> 2019-07-27 15:34:29.684 Detail, Node004, Initial read of value
> 2019-07-27 15:34:29.685 Info, Node004,   Configuration File Revision is 7
> 2019-07-27 15:34:29.686 Info, LookupTxT Checking 0002.0002.013c.db.openzwave.com
> 2019-07-27 15:34:29.686 Info, Node004, (112 - COMMAND_CLASS_CONFIGURATION) - Compatibility Flags:
> 2019-07-27 15:34:29.687 Info, Node004, (112 - COMMAND_CLASS_CONFIGURATION) - State Flags:
> 2019-07-27 15:34:29.688 Info, Lookup for 0002.0002.013c.db.openzwave.com returned 7
> 2019-07-27 15:34:29.689 Info, Node004, (48 - COMMAND_CLASS_SENSOR_BINARY) - Compatibility Flags:
> 2019-07-27 15:34:29.689 Info, Node004, 	 CreateVars: false
> 2019-07-27 15:34:29.690 Info, Node004, (48 - COMMAND_CLASS_SENSOR_BINARY) - State Flags:
> 2019-07-27 15:34:29.691 Info, Node004, (133 - COMMAND_CLASS_ASSOCIATION) - Compatibility Flags:
> 2019-07-27 15:34:29.692 Info, Node004, (133 - COMMAND_CLASS_ASSOCIATION) - State Flags:
> 2019-07-27 15:34:29.693 Info, Node004, 	 StaticRequests: 2
> 2019-07-27 15:34:29.694 Info, Node004, Received manufacturer specific report from node 4: Manufacturer=Philio Technology Corp, Product=PSM02-1 Slim Multi-Sensor
> 2019-07-27 15:34:29.694 Info, Node004, Node Identity Codes: 013c:0002:0002
> 2019-07-27 15:34:29.695 Detail, Node004,   Expected reply and command class was received
> 2019-07-27 15:34:29.696 Detail, Node004,   Message transaction complete
> 2019-07-27 15:34:29.697 Detail, 
> 2019-07-27 15:34:29.697 Detail, Node004, Removing current message
> 2019-07-27 15:34:29.698 Detail, Node004, Notification: ValueChanged
> 2019-07-27 15:34:29.700 Detail, Node004, Notification: ValueChanged
> 2019-07-27 15:34:29.702 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.704 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.705 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.707 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.711 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.713 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.715 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.717 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.719 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.723 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.725 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.727 Detail, Node004, Notification: ValueAdded
> 2019-07-27 15:34:29.728 Warning, Exception: OpenZWave::Manager::GetValueLabel:1512 - 101 - Invalid ValueID passed to GetValueLabel
> 2019-07-27 15:35:38.452 Warning, Exception: OpenZWave::Manager::GetValueUnits:1568 - 101 - Invalid ValueID passed to GetValueUnits
> 

There is no information from this node in the ozwcache file

Stepped a bit into the code:

Manager.cpp, line 1504 (**if (Internal::VC::Value* value = driver->GetValue(_id))**)

Node.cpp, line 2657 (**return GetValueStore()->GetValue(_id.GetValueStoreKey());**)

ValueStore.cpp line 182 (**Value* ValueStore::GetValue(uint32 const& _key) const**)

```
map<uint32, Value*>::const_iterator it = m_values.find(_key);
```

it returns **m_values.end()**

It seems the value is not found internally

Is there anything i could do the help debug this ?

petergebruers, 2019-07-27:
This is a long shot... It think the code that adds to the value store and triggers the Notification is quite OK, so maybe something is removing the value...

I've noticed pst02 and psm02 config files have this set under CC 0x30 (48)

<CreateVars>false</CreateVars>

I'm not familiar with that option but digging into the code it is possible that de value was added (eg during interview) but later was removed (triggering

```
// Make sure previously created values are removed if create_vars=false
if (!m_com.GetFlagBool(COMPAT_FLAG_CREATEVARS))
{
    if (Node* node = GetNodeUnsafe())
    {
        node->GetValueStore()->RemoveCommandClassValues(GetCommandClassId());
    }
}
```

That code triggers a "Notification(Notification::Type_ValueRemoved)" and I see you handle that in Domoticz, so I might be on the wrong track.

But maybe remove that CreateVars from psm02.xml and see what happens...

gizmocuz, 2019-07-28:
@petergebruers , yes ! That does the trick indeed !

Tested the node, everything works...

I don't know why CreateVars is set to 'false', but even if this was set,
it would be best to not 'add' the value first, and later remove it,
maybe it is possible to check if the value should be created,
if not, do not add the value ?

But i think we could remove it there... These vars are for binary sensors (motion/tamper/door), you do want those sensors

Is the default value for CreateVars 'true' ? In this case it could be removed by 20 other configuration files.

The CreateVars 'false' for the commandClass 48 is also done in:
philio/pst02.xml
wenzhou/tsp01.xml

the wenzhou seems the same as the Philio pst02 ... tsp -> pst ... yes.... 

I do have a PST02 at a remote location that i should be able to test in 4 hours.
I will report back shortly, if the createvars could be removed there as well i will create a PR for both

Thanks!

petergebruers, 2019-07-28:
> @petergebruers , yes ! That does the trick indeed !

That is good news!

> I don't know why CreateVars is set to 'false'

We probably can't find out for those older config files, with 1.6 and its "revision" attribute on config files that might get a little easier because when you create a PR you are very likely to enter a reference to this issue.

> it would be best to not 'add' the value first, and later remove it,
> maybe it is possible to check if the value should be created,
> if not, do not add the value ?

That is a reasonable request :) but I don't know if it is easy, it might have to do with the order of the interview, but I am speculating... The "false" is only know after the manufacturer specific data (the device id's) is know, before that OZW doesn't know the device and thus cannot know the compatibility flags. On the other hand, you first have to check what CCs a device supports. "To Be Tested"

> But i think we could remove it there... These vars are for binary sensors (motion/tamper/door), you do want those sensors

That's what I thought too...

> Is the default value for CreateVars 'true' ? In this case it could be removed by 20 other configuration files.

Default is true. Similar cases, remove, yes, but I guess most```<CreateVars>false</CreateVars>``` on other CCs are valid:

```<!-- COMMAND_CLASS_WAKE_UP. This class is in the list reported by the Minimote, but it does not
  respond to requests.  It still needs to be supported so that wake up notifications are handled. -->```

I haven't checked but the explanation makes sense.

> I do have a PST02 at a remote location that i should be able to test in 4 hours.
> I will report back shortly, if the CreateVars could be removed there as well i will create a PR for both

Thanks!

I see this is a Domoticz user with PSM02 and he says the values are not updating

https://github.com/domoticz/domoticz/issues/3544

I have a PST02-B (with door sensor) on my Fibaro HomeCenter and if my memory serves me well, it has to do with setting of a parameter, or multiple parameters, to certain values. There are different models, different firmware revisions and iirc different manuals, all very confusing.

gizmocuz, 2019-07-28:
I can confirm that for the pst02.xml file, the createvars have to be removed.
When it does not create the vars, it will produce the same crash

I create a PR for this:

https://github.com/OpenZWave/open-zwave/pull/1898

gizmocuz, 2019-07-28:
I do have a strange issue with the PST02 !

When i included the node 'secure', and operating the device magnet to simulate a door open/close,
i am receiving strange index values with strange labels, or no labels at all:

[image]

I excluded it, included it in the normal way, and it seems to work much better !!

Excluded it... included it secure again....

Again strange values are received:

[image]

Last test, exclude, include normal again, working without any issue !

I think this is unrelated to this issue, but there is something not correct here...

petergebruers, 2019-07-28:
I saw on Domoticz tracker, config file update fixes this issue so I am going to close it... Moving the PST problem to a separate issue.

closed 2019-07-28 by petergebruers

Fishwaldo, 2019-08-22:
I've commited a better fix just now to address this where we will test if the ValueID is valid before sending a notification of ValueAdded and also exposed a IsValueValid method on the manager class so apps can test ValueID's without raising exceptions etc.

gizmocuz, 2019-08-22:
Thank you for this! Going to integrate this asap