OpenZWave/open-zwave #1903: Add Node does not complain to Z-Wave protocol specification

velocityzen, 2019-08-10:
This implementation has `AddNode` method to activate add mode on a controller. 

From the spec:
* Add Node command is used to activate and **de-activate** add mode. 
* Mode parameter is used to indicate that
  * value 0x01 – ADD_MODE_ANY
  * value 0x05 – ADD_MODE_STOP

How do you de-activate stop add node mode in OZW?

petergebruers, 2019-08-11:
Not sure what you are saying are asking, OpenZWave really does implement that.

If you write software, always use the "Manager" functions... In this case:

bool Manager::CancelControllerCommand	(	uint32 const 	_homeId	)	

Have a look at its implementation in cpp/src/Driver.cpp

bool Driver::CancelControllerCommand

This one calls

void Driver::AddNodeStop

And there is your ADD_NODE_STOP

msg->Append( ADD_NODE_STOP );

velocityzen, 2019-08-11:
I thought that BeginControllerCommand and CancelControllerCommand were deprecated. Thank you for explanation. I couldn't find this in docs!

closed 2019-08-11 by velocityzen

Fishwaldo, 2019-08-11:
BegincontrollerCommand is depreciated. Use the dedicated controller command methods. CancelControllerCommand can be used for them if necessary