OpenZWave/open-zwave #1905: Improve consistency of units eg W versus Watt

gizmocuz, 2019-08-13:
OZW 1.6 currently uses two different notations of 'Watt' (**W** vs **Watt**)

It is an idea to create a PR that will make it consistent throughout the entire library ?

```
<CommandClass id="49" name="COMMAND_CLASS_SENSOR_MULTILEVEL">
   <Value type="decimal" genre="user" instance="1" index="4" label="Power" units="Watt" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="10.4">
</CommandClass>
```

```
<CommandClass id="50" name="COMMAND_CLASS_METER">
   <Value type="decimal" genre="user" instance="1" index="0" label="Electric - kWh" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="10.56">
   <Value type="decimal" genre="user" instance="1" index="2" label="Electric - W" units="W" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="9.7">
</CommandClass>
```

Configuration files, like FGD-212 (Dimmer 2)

```
    <Value genre="config" index="39" instance="1" label="Power limit - OVERLOAD" max="350" min="1" type="short" units="Watt" value="250">
```

Qubino (ZMNHCDx)

```
<Value genre="config" index="76" instance="1" label="Motor operation detection" max="127" min="0" type="byte" units="w" value="10">
```

And if so, what is the preferred notation ?

subtrEM, 2019-08-13:
Not critical for good operation in this case, though not bad to pay attention (units _are_ important [https://spectrum.ieee.org/aerospace/robotic-exploration/why-the-mars-probe-went-off-course](url) )
At least, both W and Watt are right.
Since the used symbol for unit of energy is kWh, usage of W for power (=energy/time) would be consistent.

petergebruers, 2019-08-13:
Well spotted!

I guess over the years, having various sources of information, this has lead to inconsistencies

As an contributor and engineer, I would apply the rules of the SI as written here

https://en.wikipedia.org/wiki/International_System_of_Units

I quote the most important one:

"The value of a quantity is written as a number followed by a space (representing a multiplication sign) and a unit symbol; e.g., 2.21 kg, 7.3×102 m2, 22 K. This rule explicitly includes the percent sign (%)[3]:134 and the symbol for degrees Celsius (°C).[3]:133 Exceptions are the symbols for plane angular degrees, minutes, and seconds (°, ′, and ″), which are placed immediately after the number with no intervening space."

So the unit symbols are A, V, W, lx, ...

I am not sure if this conflicts with the Z-Wave Specifications though.

To make it a bit more exciting, ° and Ω are UTF symbols, eg https://en.wikipedia.org/wiki/Omega

Fishwaldo, 2019-08-19:
so Units specified for the Configuration CC values are supplied by users. There has never really been any checks on consistency there, so its probably all over the place and would take some effort to clean up.

Units on any other CC should be uniform though. If they are not, then by all means, PR's are welcome.