OpenZWave/open-zwave #1907: Excluded node not removed from cache.

janneman001, 2019-08-18:
Using node-openzwave-shared and libopenzwave-1.6.878-86.1.x86_64 on opensuse 42.3

When i add nodes they will be added to cache. However when i exclude a node it will not get removed from the cachefile. This results in exceptions like:

19:26:20.535 Warning, Exception: Manager.cpp:1512 - 101 - Invalid ValueID passed to GetValueLabel

and errors like:
throwing an instance of 'OpenZWave::OZWException'
  what():  Manager.cpp:1512 - InvalidValueIDError (101) Msg: Invalid ValueID passed to GetValueLabel

When restarting the application.

Deleting the cachefile and starting the application or removing the node from the cachefile and starting the application makes it ok again.

Please file a log of inclusion and exclusion of node and the resulting cachefile. (renamed it to txt otherwise no upload possible)
Tried also different nodetypes. All same result.

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3513349/OZW_Log.txt)
[ozwcache_0xea387b58.xml.txt](https://github.com/OpenZWave/open-zwave/files/3513351/ozwcache_0xea387b58.xml.txt)

Fishwaldo, 2019-08-19:
I just pushed a fix for this hopefully. Can you test please?

janneman001, 2019-08-19:
Thanks for your quick return. 
Do you know the frequency of the distribution package builds? It looks like monthly on: http://old.openzwave.com/downloads/
I guess, if i want to test it right away, i will have to do a build of 1.6.881?

closed 2019-08-19 by janneman001