OpenZWave/open-zwave #1908: Edit of node name and/or location not stored remembered

janneman001, 2019-08-19:
When editing name and/or location of a node it is only remembered when the application runs. Restarting results in loss.
When updating name location in cache-file by hand it will remember but any edits will be lost on application restart.

I am not sure if it should be the application's responsibility to remember name/location. I assume the network should. With this the cache-file contains information that, when deleting it , will be lost. All other information can be recovered by queries to the nodes.

Please find a log of editing of node name and location and the resulting cachefile. (renamed it to txt otherwise no upload possible)

Reproduction on:
Using node-openzwave-shared and libopenzwave-1.6.878-86.1.x86_64 on opensuse 42.3
Also on master 2019081915:00 CEST. (just compiled)

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3516210/OZW_Log.txt)
[ozwcache_0xdbf6e587.xml.txt](https://github.com/OpenZWave/open-zwave/files/3516211/ozwcache_0xdbf6e587.xml.txt)

Fishwaldo, 2019-08-19:
This is dependent upon the node supporting then NodeName CommandClass, which less and less devices do now days (unfortunately). 

If the device does not support it, then yes, wiping the ozwcache file will mean loosing the name/location as there is no way to persist that information on the network etc.

closed 2019-08-19 by Fishwaldo

gdombiak, 2019-08-24:
Is there a plan to make it work as when we used to use zwcfg_*.xml? I used to use that file as a way to store location/name and it would survive across restarts or updates. It seems like that it got renamed to ozwcache_*.xml it does behave like a cache. :( The previous behavior was useful for me.  It would be great if we can have the ability to store state outside of the capabilities of devices.

I'm trying to update to latest version of open-zwave and this issue is blocking me. Unless I store my own settings by myself which is going to be my last resort.

Thanks,
Gaston