OpenZWave/open-zwave #1911: Device with two binary sensors (1.6 regression bug)

ratatine, 2019-08-21:
Running ozw 1.4 & 1.6 on Rpi3 buster.

I've found an issue where 1.6 support for the Linear WADWAZ-1 binary sensor is no longer complete. This device works better under 1.4 although the behavior is a little unintuitive in 1.4.

The WADWAZ binary sensor has an internal reed switch as well as external wired contacts for a second switch. Both binary sensors work simultaneously. They both report closed as 0x00 and open as 0xFF. However it reports which sensor was opened by sending a 0x02 for the internal switch and a 0xFE for the external switch. In 1.4 this is reported in the field 113-1-10 (burglar) as either 2 or 254 and the open/closed status is reported in 113-1-1 (alarm level) as 0 or 255. Additionally there is 48-1-0 (sensor) which reports true/false for ONLY the internal switch.

In 1.6 this behavior is different. The sensor values for 113-1-0 (alarm type) and 113-1-1 (alarm level) as well as 113-1-10 (burglar) are gone. You still have 48-1-0 which acts the same with true/false for the internal switch but the only other value is 113-1-7 (home security) which changes between various text declaring an incident when either switch is triggered followed by a "clear" message 5 seconds later. 

The version 1.4 behavior is usable although not intuitive and definitely troublesome if you are using both sensors and the 1.6 behavior is totally unusable for the external sensor. 

I shared quite a few logs when discussing this over at zwave2mqtt (https://github.com/OpenZWave/Zwave2Mqtt/issues/100) and while I'm happy to test or try different things to support finding a fix, I don't know my way around the project well enough to direct the effort. 

Ideally this sensor would work by having two values (true/false or 0/255) that represent the two sensors. Thanks for any help anyone can provide!

Here's the key log from MinOZW 1.6:
2019-08-19 11:20:07.893 Detail, Node033, AdvanceQueries queryPending=0 queryRetries=0 queryStage=None live=1
2019-08-19 11:20:07.893 Detail, Node033, QueryStage_ProtocolInfo
2019-08-19 11:20:07.893 Detail, Node033, Queuing (Query) Get Node Protocol Info (Node=33): 0x01, 0x04, 0x00, 0x41, 0x21, 0x9b
2019-08-19 11:20:07.893 Detail, Node033, Queuing (Query) Query Stage Complete (ProtocolInfo)
2019-08-19 11:20:07.893 Info, Node033, Initializing Node. New Node: false (false)
2019-08-19 11:20:07.902 Detail, Node033, Notification: NodeNew
2019-08-19 11:20:07.902 Detail, Node033, Notification: NodeAdded
2019-08-19 11:20:36.021 Info, Node033, Sending (Query) message (Callback ID=0x00, Expected Reply=0x41) - Get Node Protocol Info (Node=33): 0x01, 0x04, 0x00, 0x41, 0x21, 0x9b
2019-08-19 11:20:36.021 Info, Node033, Encrypted Flag is 0
2019-08-19 11:20:36.025 Detail, Node033, Received: 0x01, 0x09, 0x01, 0x41, 0x53, 0x9c, 0x00, 0x04, 0x20, 0x01, 0x5c
2019-08-19 11:20:36.025 Info, Node033, Received reply to FUNC_ID_ZW_GET_NODE_PROTOCOL_INFO
2019-08-19 11:20:36.025 Info, Node033, Protocol Info for Node 33:
2019-08-19 11:20:36.025 Info, Node033, Listening = false
2019-08-19 11:20:36.026 Info, Node033, Frequent = false
2019-08-19 11:20:36.026 Info, Node033, Beaming = true
2019-08-19 11:20:36.026 Info, Node033, Routing = true
2019-08-19 11:20:36.026 Info, Node033, Max Baud Rate = 40000
2019-08-19 11:20:36.026 Info, Node033, Version = 4
2019-08-19 11:20:36.026 Info, Node033, Security = false
2019-08-19 11:20:36.026 Info, Node033, Basic device class (0x04) - Routing Slave
2019-08-19 11:20:36.027 Info, Node033, Generic device Class (0x20) - Binary Sensor
2019-08-19 11:20:36.027 Info, Node033, Specific device class (0x01) - Routing Binary Sensor
2019-08-19 11:20:36.027 Info, Node033, COMMAND_CLASS_BASIC will be mapped to COMMAND_CLASS_SENSOR_BINARY
2019-08-19 11:20:36.028 Info, Node033, Mandatory Command Classes for Node 33:
2019-08-19 11:20:36.028 Info, Node033, COMMAND_CLASS_SENSOR_BINARY
2019-08-19 11:20:36.028 Info, Node033, COMMAND_CLASS_MANUFACTURER_SPECIFIC
2019-08-19 11:20:36.028 Info, Node033, COMMAND_CLASS_WAKE_UP
2019-08-19 11:20:36.028 Info, Node033, Mandatory Command Classes controlled by Node 33:
2019-08-19 11:20:36.028 Info, Node033, COMMAND_CLASS_BASIC
2019-08-19 11:20:36.028 Detail, Node033, Expected reply was received
2019-08-19 11:20:36.028 Detail, Node033, Message transaction complete
2019-08-19 11:20:36.029 Detail, Node033, Removing current message
2019-08-19 11:20:36.029 Detail, Node033, Notification: NodeProtocolInfo
2019-08-19 11:20:36.029 Detail, Node033, Notification: ValueAdded
2019-08-19 11:20:36.029 Detail, Node033, Notification: ValueAdded
2019-08-19 11:20:36.029 Detail, Node033, Query Stage Complete (ProtocolInfo)
2019-08-19 11:20:36.029 Detail, Node033, AdvanceQueries queryPending=0 queryRetries=0 queryStage=Probe live=1
2019-08-19 11:20:36.030 Detail, Node033, QueryStage_Probe
2019-08-19 11:20:36.030 Info, Node033, NoOperation::Set - Routing=true
2019-08-19 11:20:36.030 Detail, Node033, Queuing (NoOp) NoOperation_Set (Node=33): 0x01, 0x09, 0x00, 0x13, 0x21, 0x02, 0x00, 0x00, 0x25, 0x1c, 0xff
2019-08-19 11:20:36.030 Detail, Node033, Queuing (Query) Query Stage Complete (Probe)
2019-08-19 11:20:36.031 Info, Node033, Sending (NoOp) message (Callback ID=0x1c, Expected Reply=0x13) - NoOperation_Set (Node=33): 0x01, 0x09, 0x00, 0x13, 0x21, 0x02, 0x00, 0x00, 0x25, 0x1c, 0xff
2019-08-19 11:20:36.031 Info, Node033, Encrypted Flag is 0
2019-08-19 11:20:36.038 Detail, Node033, Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-08-19 11:20:36.038 Detail, Node033, ZW_SEND_DATA delivered to Z-Wave stack
2019-08-19 11:20:43.367 Detail, Node033, Received: 0x01, 0x05, 0x00, 0x13, 0x1c, 0x01, 0xf4
2019-08-19 11:20:43.367 Detail, Node033, ZW_SEND_DATA Request with callback ID 0x1c received (expected 0x1c)
2019-08-19 11:20:43.367 Info, Node033, WARNING: ZW_SEND_DATA failed. No ACK received - device may be asleep.
2019-08-19 11:20:43.367 Info, Node033, Node 33 has been marked as asleep
2019-08-19 11:20:43.367 Info, Node033, Node not responding - moving QueryStageComplete command to Wake-Up queue
2019-08-19 11:20:43.367 Detail, Node033, Notification: Notification - NoOperation
2019-08-19 11:20:43.368 Detail, Node033, Notification: Notification - Node Asleep
2019-08-19 11:20:43.371 Info, Node033, Skipping Cache Save for Node 33 as its not past QueryStage_CacheLoad
2019-08-19 11:20:56.790 Info, Node033, Skipping Cache Save for Node 33 as its not past QueryStage_CacheLoad

[... 6 lines cut ...]
2019-08-19 11:21:43.671 Info, Node033, ApplicationCommandHandler - Unhandled Command Class 0x71

**Triggering external contact closed**
2019-08-19 11:21:48.791 Detail, Node033, Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x21, 0x0a, 0x71, 0x05, 0x07, **0x00**, 0x00, 0xff, 0x07, **0xfe**, 0x00, 0x00, 0xb5
2019-08-19 11:21:48.792 Info, Node033, ApplicationCommandHandler - Unhandled Command Class 0x71

**Trigger external contact open**
2019-08-19 11:21:53.317 Detail, Node033, Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x21, 0x0a, 0x71, 0x05, 0x07, **0xff**, 0x00, 0xff, 0x07, **0xfe**, 0x00, 0x00, 0x4a
2019-08-19 11:21:53.320 Info, Node033, ApplicationCommandHandler - Unhandled Command Class 0x71

**Triggering reed switch closed**
2019-08-19 11:21:57.539 Detail, Node033, Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x21, 0x03, 0x20, 0x01, 0x00, 0xf1
2019-08-19 11:21:57.539 Warning, Node033, Routing HandleIncomingMsg to HandleMsg - Please Report: COMMAND_CLASS_BASIC
2019-08-19 11:21:57.539 Info, Node033, Received Basic set from node 33: level=0. Sending event notification.
2019-08-19 11:21:57.540 Detail, Node033, Notification: NodeEvent
2019-08-19 11:21:57.585 Detail, Node033, Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x21, 0x0a, 0x71, 0x05, 0x07, **0x00**, 0x00, 0xff, 0x07, **0x02**, 0x00, 0x00, 0x49
2019-08-19 11:21:57.585 Info, Node033, ApplicationCommandHandler - Unhandled Command Class 0x71

**Triggering reed switch open**
2019-08-19 11:22:05.959 Detail, Node033, Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x21, 0x03, 0x20, 0x01, 0xff, 0x0e
2019-08-19 11:22:05.960 Warning, Node033, Routing HandleIncomingMsg to HandleMsg - Please Report: COMMAND_CLASS_BASIC
2019-08-19 11:22:05.960 Info, Node033, Received Basic set from node 33: level=255. Sending event notification.
2019-08-19 11:22:05.960 Detail, Node033, Notification: NodeEvent
2019-08-19 11:22:06.004 Detail, Node033, Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x21, 0x0a, 0x71, 0x05, 0x07, **0xff**, 0x00, 0xff, 0x07, **0x02**, 0x00, 0x00, 0xb6
2019-08-19 11:22:06.004 Info, Node033, ApplicationCommandHandler - Unhandled Command Class 0x71

Fishwaldo, 2019-08-22:
>2019-08-19 11:21:37.645 Detail, Node033, Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x21, 0x0a, 0x71, 0x05, 0x07, 0x00, 0x00, 0xff, 0x07, 0xfe, 0x00, 0x00, 0xb5
2019-08-19 11:21:37.646 Info, Node033, ApplicationCommandHandler - Unhandled Command Class 0x71

As already mentioned in the other post, you need to wake up the device to let the interview process complete. Before the interview process is complete, not all Values will be exposed etc. 

I also have no idea about the format that node-shared uses. Please reference the Value Label or Index at the least. 

Finally - The Notification CC was rewritten in 1.6 to be in line with the zwave specifications. Things work slightly differently now so a direct comparision of "values" to 1.4 isn't valid. Your code would have to take into account new Values being exposed etc.

ratatine, 2019-08-22:
I don't think you read my response to your message in the other thread. I let the system sit for 4 days and there was no change. During that time I woke up the devices many times. I get the 0x71 message because it hasn't gone through cache load in the short time I'm running MinOZW to get you the log data required, as I understand it.

If you can tell me exactly what you would like me to do I am happy to do it. I do not know what "node-shared" is. To my knowledge I provided all the value labels. I do not know what the index is.

Also you say "Your code would have to take into account new Values being exposed etc.". I haven't written any code. Everything I have provided has come from MinOZW and zwave2mqtt.

Thanks.

Fishwaldo, 2019-08-23:
The logs you have reposted are useless as the device has not completed the interview process. For your background.

When a ZWave device is sleeping, it can only send events to the network, it can not recieve anything. With the logs you posted, OZW has been unable to complete the necessary interview upon restarting to get the latest state of the device. During this cacheload phase, the device is mostly "useless" and only very basic messages can be processed by OZW. 

So, you need to wake up the device after restarting OZW and then check what Values are exposed by the Notification (Alarm) CC and see if additional "new" values are present for your device.

As mentioned, the Notification Class was rewritten in 1.6 to be compliant with the current Z-Wave Specifications. In 1.4 we only supported a very old version of the specifications. With the new class, things are different, so you can not expect the same "values" to be present on 1.6 as it is represented differently (in line with the specifications). One clear example of that is that the Alarm Type and Alarm Level are gone for all but very old devices. New devices should expose a Specific Value for each type of Alarm. 

As you continued to post logs (with messages that indicate that the message is unhandled), I can't deduce anything as OZW is not "ready" for that device.

So - after restarting OZW, wake up the device to allow it to progress to QueryStage_Complete, inspect the ValueID's that have now been exposed, and see what maps to what. 

(How the device maps different events to the Notification Class is purely device dependant - Hence why the interview needs to complete and is going to be different from device to device, so there is not really a "standard" solution. There are best practices but not all vendors have the same idea of what is considered best practices)

ratatine, 2019-08-23:
I believe this is what you want. This is a full log from MinOZW. I've removed nothing so there's a lot of other info from the other sensors tied to this device. I did not include the secure key in this session as the device in question doesn't use security as far as I can tell. The node to look at is still Node 33. During this session I triggered both the internal reed switch and the external contact.

At 10:03:13 you can see me close the internal switch and at 10:03:15 you can see me open it again.
At 10:03:31 you can see me close the externals witch and at 10:03:35 you can see me open it again.

<script src="https://gist.github.com/ratatine/a48efc14102a92350c85dbfb88e41142.js"></script>

ratatine, 2019-08-23:
Quick add - when I say what you want, this is MinOZW _after_ the cacheload has completed.

Fishwaldo, 2019-08-26:
looks like its working as the manufacturer intended:
internal:

>**2019-08-23 10:03:13.939 Info, Node033, Received Notification report (>v1): Type: Home Security (7) Event: Intrusion at Unknown Location (2) Status: true, Param Length: 0

External:
>2019-08-23 10:03:31.529 Info, Node033, Received Notification report (>v1): Type: Home Security (7) Event: Unknown Event (254) Status: true, Param Length: 0

So you Internal Sensor triggers "Intrusion" and External Triggers "Unknown Event"

petergebruers, 2019-08-26:
@Fishwaldo I think I see what is happening with this (specific) device and I also think I understand what @ratatine said in the first post and it probably explains the difference between 1.4 and 1.6

Took a while to see it, I am slow ;)

The key is here, when he says: "The sensor values for 113-1-0 (alarm type) and 113-1-1 (alarm level) as well as 113-1-10 (burglar) are gone."

Indeed, I see no mention of "alarm level" in the OZW_Log and looking at Alarm.cpp is because 1.6 really treats it as > V1 (which imho is absolutely correct).

But, here is "close the internal switch".

10:03:13.938 Detail, Node033,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x21, 0x0a, 0x71, 0x05, 0x07, 0x00, 0x00, 0xff, 0x07, 0x02, 0x00, 0x00, 0x49
2019-08-23 10:03:13.939 Detail, 
2019-08-23 10:03:13.939 Info, Node033, Response RTT 625 Average Response RTT 625
2019-08-23 10:03:13.939 Info, Node033, Received Notification report (>v1): Type: Home Security (7) Event: Intrusion at Unknown Location (2) Status: true, Param Length: 0
2019-08-23 10:03:13.939 Detail, Node033, Initial read of value

And here is "open the internals switch".

10:03:15.045 Detail, Node033,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x21, 0x0a, 0x71, 0x05, 0x07, 0xff, 0x00, 0xff, 0x07, 0x02, 0x00, 0x00, 0xb6
2019-08-23 10:03:15.045 Detail, 
2019-08-23 10:03:15.045 Info, Node033, Response RTT 1733 Average Response RTT 1444
2019-08-23 10:03:15.045 Info, Node033, Received Notification report (>v1): Type: Home Security (7) Event: Intrusion at Unknown Location (2) Status: true, Param Length: 0
2019-08-23 10:03:15.046 Detail, Node033, Refreshed Value: old value=2, new value=2, type=list
2019-08

Both generate the same notification for open and close...

The only way to see the difference between open and close of this sensor is by looking at:

_data[1] = ValueID_Index_Alarm::Type_v1 (if OZW 1.6 would expose it, but it does not do that)
_data[2] = ValueID_Index_Alarm::Level_v1 (if OZW 1.6 would expose it, but it does not do that) - this is the 11th byte on the "received" line.

For close,  _data[2] is 0x00 and for open it is 0xFF

I expected this sensor to send a "clear/idle" notification for one of those events but I see none in the log file.

Is this the first sensor to exhibit such a behavior?

Would it make sense to leave out the "if (GetVersion() == 1)" on line 196 to create those values for any class cersion? The spec says ("4.71 Notification Command Class, version 3-8"):

```
These fields carry the proprietary Alarm Type and Alarm Level fields originally introduced with Alarm Command Class, Version 1. V1 Alarm Type and V1 Alarm Level fields MUST be specified in the product manual.
If the V1 Alarm Type is not supported, these fields MUST be set to 0x00.
```

Fishwaldo, 2019-08-27:
I have no idea what those numbers are. They are a internal representation of the NodeJS wrapper! Please use the Value Labels to describe what you expect. 

This is neither a OZW bug or zwave2mqtt bug, this is how the vendor chose to implement the Notification CC. Both software do not “interpret” anything, we mearly expose what the device sends. 

To differentiate between internal and external you would look at the Home Security ValueID for either “Unknown Location (2)” or “Unknown event(254)”

Fishwaldo, 2019-08-28:
@petergebruers Thats not what I'm seeing for the external sensor (based on the times provided by Ratatine)

When the external sensor is triggered, its sending:
>2019-08-23 10:03:31.410 Detail, Node033,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x21, 0x0a, 0x71, 0x05, 0x07, 0x00, 0x00, 0xff, 0x07, 0xfe, 0x00, 0x00, 0xb5

Which is translating into:
> 2019-08-23 10:03:31.410 Info, Node033, Received Notification report (>v1): Type: Home Security (7) Event: Unknown Event (254) Status: true, Param Length: 0

which is different from the internal sensor:
>2019-08-23 10:03:15.045 Detail, Node033,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x21, 0x0a, 0x71, 0x05, 0x07, 0xff, 0x00, 0xff, 0x07, 0x02, 0x00, 0x00, 0xb6

which results in:
>2019-08-23 10:03:15.045 Info, Node033, Received Notification report (>v1): Type: Home Security (7) Event: Intrusion at Unknown Location (2) Status: true, Param Length: 0

So just looking at the ValueID for Home Security (either a 2 (0x02) - which translates to "Intrusion at Unknown Location" or 254 (0xFE) which is "Unknown Event") you can differentiate between the two sensors.

Unfortunately, this device is a bit of a wack job. Its also sending a Basic Report upon status changes which triggers OZW to refresh the BinarySensor Value, but as its sleeping, we can't get the value. If it could refresh, then I'm sure BinarySensor would allow you to get the "Open/Closed" status.

If the device sends a "Clear" event when its closed, then you can use "Previous Status" to determine what Sensor it was (0x02 or 0xFE). If it doesnt - Well, then its a issue to take up with the manufacturer

Fishwaldo, 2019-08-28:
Ok.... I see that V1 Alarm Type And V1 Alarm Level are not depreciated, but I'm not keen to expose them the same way as 1.4 as individual ValueID's. I may look at exposing them as Event Params instead.

Fishwaldo, 2019-09-03:
Could you please post a OZW Log file with ozwcache removed please? I need to check a flag sent by this device during discovery.

ratatine, 2019-09-03:
Here you go. Took a while to finish so the log is somewhat long. Also towards the end you'll see I actuated both the internal and external switches.

https://gist.github.com/ratatine/2d27302ab2a5dcba2570b17680aa413e

Fishwaldo, 2019-09-05:
So unfortuantly this device in particular isn't 100% compliant with the  Z-Wave Specifications, hence why its probably affecting you:

In the SupportedReport Message we get back from the device during discover:
>2019-09-03 12:12:31.058 Detail, Node016,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x10, 0x04, 0x71, 0x08, 0x01, 0x80, 0x1d

the 0x01 0x80 are the bytes that indicate what alarm types it reports. In this case, it says it "does not" support Version 1 Alarm Types (otherwise it would be something like 0x8). Hence OZW never creates the old OZW Type 1 alarm ValueID's. 

See 4.71.8 in the CommandClass Specification

A work around is to update the config file for the device and enable ["EnableV1AlarmTypes"](https://github.com/OpenZWave/open-zwave/wiki/CommandClass-Compatibility-Flags#notification-commanclass) 

Please test and reopen if its still having a issue.

closed 2019-09-05 by Fishwaldo