OpenZWave/open-zwave #1916: make xmltest fails because of config=""

petergebruers, 2019-08-25:
After merge of `<Product config="" id="0001" name="Alfred DB2 Deadbolt" type="0003"/>`

```
Checking Config Files... Please Wait
config/manufacturer_specific.xml - Updating Database
read error at /System/Library/Perl/Extras/5.18/darwin-thread-multi-2level/XML/Parser/Expat.pm line 470.
XML::Simple called at cpp/build/testconfig.pl line 200.
```

The reason is no config means no metadata which imho goes against the philosophy of OZW 1.6 -there should be a minimum config file for a device. @Fishwaldo please correct me if I am wrong.

Introduced by merge of  https://github.com/OpenZWave/open-zwave/pull/1913

petergebruers, 2019-08-25:
Device is:

https://products.z-wavealliance.org/products/3324

petergebruers, 2019-08-25:
Config files fixed and merged

closed 2019-08-25 by petergebruers

Fishwaldo, 2019-08-26:
added a check to make sure that this doesn't hit us again.