OpenZWave/open-zwave #1919: Zipato Bulb association groups

cincuranet, 2019-08-26:
Looking at the [config file for Zipato Bulb](https://github.com/OpenZWave/open-zwave/blob/master/config/zipato/RGBBulb2.xml#L20) I maximum 5 associations in the first group. Is that correct? Isn't the _Lifeline_ always for (and only for) controller?

petergebruers, 2019-08-26:
Config files are usually contributed by users, the developers here do not own every possible device. So it is a legitimate question.  Weare talking about Z-Wave Plus... Not all manufacturers restrict group 1 to only one associaton. The spec does say the controller should be in group one. It does not say group one should be restrioted. What does the manual say? Do you own this device? Does it work? Do you have problems or are you just curious?

You can help us answer your question!

Try to find your device here:

https://products.z-wavealliance.org/

Often there is a manual

Then follow these instructions to update the file:

https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

cincuranet, 2019-08-26:
Well, the [documentation says](https://products.z-wavealliance.org/products/2712/assoc?noFilename=True) only 1. Reason I'm asking is just curiosity (and maybe somebody has experience how this device behaves).

petergebruers, 2019-09-04:
Thanks. I had a quick look at the manual on the manufacturer site, and it does not mention the group, but it does list parameters that aren't in the config file (yet).

<https://www.zipato.com/wp-content/uploads/2017/04/rgbw2-Zipato-RGBW-gen2-Bulb-User-Manual-Z-Wave-v1.3.pdf>

This can happen when a manufacturer updates the device but does not submit that data to z-wavealliance. But in this case it seems the alliance database also lists a few more parameters.

These things can stay under the radar because people get basic functionality of the device and adding a node to group 1 aka lifeline (if it is Z-Wave Plus) is rarely done imho because a device having some interesting feature (eg a wall plug which can detect overload and trigger another device) then the device will expose a special group for that.

I could fix that... But it is always safer to test it with a real device. I'll leave the issue open.