OpenZWave/open-zwave #1920: Manager->GetValueMin not reporting properly.

kdschlosser, 2019-08-28:
value.label: Acked Frames
value.node.name: GE 45609 On/Off 
value.type: Short
value.help: Number of Messages successfully Acked by the Target Node
value.max: 32767
value.min: 4294934528
value.genre: System
value.index: 9
value.data_items: []
value.is_read_only: True
value.is_write_only: False
value.command_class: COMMAND_CLASS_POWERLEVEL
value.instance: 1
value.instance_label: Instance 1:
value.precision: None

I am not exactly sure how the minimum can be higher then the maximum. it should be a 0 if I am not mistaken a short is can be from 0 to 32767

petergebruers, 2019-09-01:
Can you please check the config file?

It should say:

```
<Value type="short" genre="system" instance="1" index="9" label="Acked Frames" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-32768" max="32767" value="0">
```

I expect your program to print -32768 

OZW initializes the Value object with

m_min = SHRT_MIN;

And that is a compiler constant (on my Mac, limits.h defines it as -SCHAR_MAX - 1).

I think it would be unusual to evaluate that constant to something else than -32768...

kdschlosser, 2019-09-01:
```xml
<Value type="short" genre="system" instance="1" index="9" label="Acked Frames" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-32768" max="32767" value="0">
					<Help>Number of Messages successfully Acked by the Target Node</Help>
				</Value>
```

kdschlosser, 2019-09-01:
here is limits.h from MSVC

``` cpp
//
// limits.h
//
//      Copyright (c) Microsoft Corporation. All rights reserved.
//
// The C Standard Library <limits.h> header.
//
#pragma once
#define _INC_LIMITS

#include <vcruntime.h>

_CRT_BEGIN_C_HEADER

#define CHAR_BIT      8         // number of bits in a char
#define SCHAR_MIN   (-128)      // minimum signed char value
#define SCHAR_MAX     127       // maximum signed char value
#define UCHAR_MAX     0xff      // maximum unsigned char value

#ifndef _CHAR_UNSIGNED
    #define CHAR_MIN    SCHAR_MIN   // mimimum char value
    #define CHAR_MAX    SCHAR_MAX   // maximum char value
#else
    #define CHAR_MIN    0
    #define CHAR_MAX    UCHAR_MAX
#endif

#define MB_LEN_MAX    5             // max. # bytes in multibyte char
#define SHRT_MIN    (-32768)        // minimum (signed) short value
#define SHRT_MAX      32767         // maximum (signed) short value
#define USHRT_MAX     0xffff        // maximum unsigned short value
#define INT_MIN     (-2147483647 - 1) // minimum (signed) int value
#define INT_MAX       2147483647    // maximum (signed) int value
#define UINT_MAX      0xffffffff    // maximum unsigned int value
#define LONG_MIN    (-2147483647L - 1) // minimum (signed) long value
#define LONG_MAX      2147483647L   // maximum (signed) long value
#define ULONG_MAX     0xffffffffUL  // maximum unsigned long value
#define LLONG_MAX     9223372036854775807i64       // maximum signed long long int value
#define LLONG_MIN   (-9223372036854775807i64 - 1)  // minimum signed long long int value
#define ULLONG_MAX    0xffffffffffffffffui64       // maximum unsigned long long int value

#define _I8_MIN     (-127i8 - 1)    // minimum signed 8 bit value
#define _I8_MAX       127i8         // maximum signed 8 bit value
#define _UI8_MAX      0xffui8       // maximum unsigned 8 bit value

#define _I16_MIN    (-32767i16 - 1) // minimum signed 16 bit value
#define _I16_MAX      32767i16      // maximum signed 16 bit value
#define _UI16_MAX     0xffffui16    // maximum unsigned 16 bit value

#define _I32_MIN    (-2147483647i32 - 1) // minimum signed 32 bit value
#define _I32_MAX      2147483647i32 // maximum signed 32 bit value
#define _UI32_MAX     0xffffffffui32 // maximum unsigned 32 bit value

// minimum signed 64 bit value
#define _I64_MIN    (-9223372036854775807i64 - 1)
// maximum signed 64 bit value
#define _I64_MAX      9223372036854775807i64
// maximum unsigned 64 bit value
#define _UI64_MAX     0xffffffffffffffffui64

#if _INTEGRAL_MAX_BITS >= 128
    // minimum signed 128 bit value
    #define _I128_MIN   (-170141183460469231731687303715884105727i128 - 1)
    // maximum signed 128 bit value
    #define _I128_MAX     170141183460469231731687303715884105727i128
    // maximum unsigned 128 bit value
    #define _UI128_MAX    0xffffffffffffffffffffffffffffffffui128
#endif

#ifndef SIZE_MAX
    #ifdef _WIN64
        #define SIZE_MAX _UI64_MAX
    #else
        #define SIZE_MAX UINT_MAX
    #endif
#endif

#if __STDC_WANT_SECURE_LIB__
    #ifndef RSIZE_MAX
        #define RSIZE_MAX (SIZE_MAX >> 1)
    #endif
#endif

_CRT_END_C_HEADER
```

kdschlosser, 2019-09-01:
ok so now the hunt to locate where the issue lies... I am almost willing to bet it is going to be a bug in Cython probably. I have to go poking about 60K lines of code to locate where in the hell it converts that data type.

kdschlosser, 2019-09-01:
and I believe I found the problem!

this is what is set in the cython code for pyozw for the manager method.
uint32_t GetValueMin(ValueID& valueid)

and this is what the method is in the manager in ozw.
int32 GetValueMin(ValueID const& _id);

small problem there.

I do have a question tho. Because there are several different value data types that support the min and max.. "byte", "short", "int" how come this method is not overloaded to return the data type associates with the value data type?

I would image this would be the correct thing to do. 
int32 GetValueMin(ValueID const& _id); // ValueInt
int16 GetValueMin(ValueID const& _id); // ValueShort
uint8 GetValueMin(ValueID const& _id); // ValueByte

same goes for GetValueMax

Another question. The Thermostat Setpoint Command Class does not support min and max? I ask this question because the data type for a thermostat setpoint is a float (decimal) and it would appear as tho there is no min and max support even possible for a float

end result is the problem is actually in python-openzwave. should any changes be made to openzwave???

I did also want to mention that getting the values precision if the value is not a ValueDecimal causes an app crash But getting the Min and Max on a value that does not support it does not cause an app crash. I would think that if it is not supported NULL should be returned and not to cause an app crash. I coded around the problem in the python side of things. But I thought i would mention it since we are discussing manager methods for values.

kdschlosser, 2019-09-01:
It looks like as of version 3 of COMMAND_CLASS_THERMOSTAT_SETPOINT the Thermostat Setpoint Capabilities Report Command is supported and does provide min and max. This may be a good time to review/update the GetValueMin, GetValueMax and GetPrecision methods and the functions called in the values.

Fishwaldo, 2019-09-02:
>value.min: 4294934528

That looks like a uninitialized value. Its close to double a ULONG_MAX though, but not quite.

>this is what is set in the cython code for pyozw for the manager method.
>uint32_t GetValueMin(ValueID& valueid)
>
>and this is what the method is in the manager in ozw.
>int32 GetValueMin(ValueID const& _id);

only difference is the a hint telling the compiler if its signed/unsigned. The storage size is correct and most compilers can natively convert between the two. It doesn't explain the problem above. There is something else going on. I don't see this on the native OZW library (using MinOZW) so something in your transformation between C and Python.

>I do have a question tho. Because there are several different value data types that support the min and max.. "byte", "short", "int" how come this method is not overloaded to return the data type associates with the value data type?

It doesn't need to. (u)int32 is the maximum size the variable can hold. For somelike a byte, the MAX size is 256 <  2,147,483,647

>Another question. The Thermostat Setpoint Command Class does not support min and max? I ask this question because the data type for a thermostat setpoint is a float (decimal) and it would appear as tho there is no min and max support even possible for a float

Correct - as we can't represent floats with Min/Max currently. 

> I did also want to mention that getting the values precision if the value is not a ValueDecimal causes an app crash But getting the Min and Max on a value that does not support it does not cause an [...]

Could you provide a backtrace please?

kdschlosser, 2019-09-02:
The problem with the GetValuePrecision is that GetPrecision is only defined in ValueDecimal and not in Value so when the manager makes the call to a value and the value does not have that method. the application crashes.

I do not know all of the inner workings of Cython. I am thinking that even tho an int32 and a uint32 are the same in terms of object size it might be corrupting the data because of where the min and may points are. Cython basically reads the python/cpp mixed code and create cpp code from it. It needs to specifically told what to do and when to do it otherwise the outcome will be unpredictable if it even runs at all.  I am pretty confident that changing the data type from uint32 to int32 in the Cython code is going to solve the issue.

As far as the min and max for decimal values goes, after this comment I am going to submit a PR that addresses that specific problem. :smile: I am hoping that either the PR it's self or the idea behind it will lead to a solution. I noticed for the thermostat setpoint you added 2 additional values per supported setpoint so the user would be able to get the min and the max from those 2 values. I came up with a way that does not involve creating additional values and sets the min and max into the m_min and m_max attributes for that specific setpoint. I left the mechanics in place that were already written. so it's not API breaking at all. 

I do not have a thermostat to be able to test with. and I did want to ask if there is any evaluation against the min and the max when the user is setting a value. This goes for ValueByte, ValueShort, ValueInt and ValueDecimal. this is the only piece I did not go hunting for to change if needed. 

You guys will know better then I if it is going to work properly or not. Or if it is something that might be worth adding as is or a revised version of. or even something completely different doesn't matter really. 

There are quite a few command classes that can be extended in the manner in which the thermostat setpoint was so a nice solution should be considered now to be ahead of the game. @FishWaldo, I am sure you have some people that work on the ZWave spec that you can chat with and ask if the same type of thing as version 3 of the setpoint is being considered for other command classes. It will help to determine if it is something worth doing. 

I also fixed the GetValuePrecision problem in the PR as well.

Fishwaldo, 2019-09-02:
>The problem with the GetValuePrecision is that GetPrecision is only defined in ValueDecimal and not in Value so when the manager makes the call to a value and the value does not have that method. the application crashes.

It shouldn't even compile, let alone crash. As noted, its only defined on the ValueDecimal class as precision is only applicable to that class/ValueType. If you are getting a crash, its in he python wrapper and not OZW, as no other Value* class has it defined. Somehow, I suspect you have a weird cast in there that is shifting it to ValueDecimal (if its compiling) and then you would have bigger problems as the GetValue* calls would be working with a instance that is *NOT* a ValueDecimal class and in fact something else. 

> I am pretty confident that changing the data type from uint32 to int32 in the Cython code is going to solve the issue.

I'd argue and say that's incorrect. Anyway, as mentioned, in the C++ programs that use the OZW library, it works as expected. Sorry, your gonna have to dig into the wrapper as to why its not working on your side.

>As far as the min and max for decimal values goes, after this comment I am going to submit a PR that addresses that specific problem.

And I wish you had discussed it further before doing the work, as I wont accept that PR. The GetValueMin* methods that you have added are completely redundant as they expose exactly the same info as the existing GetValueMin/Max functions. these are NOT necessary at all.

The only "enhancement" i see if Min/Max for ValueDecimal Classes - but exposing floats via the C++ API is wrought with danger (float versus double) and challenges. Additionally, I'd expect the Min/Max code for reading the XML file should be in the base class and not reimplemented in each class that inherits from Value. 

>I did want to ask if there is any evaluation against the min and the max when the user is setting a value.

We once upon a time did "enforce" checking the values were in range of Min/Max (on Byte/Short/Ints etc) but "vendor defined" behavior meant that it wasn't the same for each device (even with same CommandClasses) so I reverted that as it broke too much for very little benefit at the time.

Unfortunately, even things like the SwitchMultiLevel are inconsistent, with some devices being "decimal 1-100" and others being fractional 1-255 for setting/reporting levels.

petergebruers, 2019-09-02:
Okay, let's take a step back... to that "weird" number which is not weird at all.

hex 4294934528
= 0xFFFF8000

Right... 0x8000 takes 16 bits and represents the smallest (so negative) signed short , ie -32767 consistent with everything said here...

The conversion to 0xFFFF8000 is consistent with what the C spec says: it is a sign extended 32 bit version of the same number. That is:

4294934528 - 2^32
= -32768

So now treated as 32 bits:

signed: 4294934528
unsigned: -32768

Let's combine that with what you wrote about the Cython source:

```
this is what is set in the cython code for pyozw for the manager method.
uint32_t GetValueMin(ValueID& valueid)

and this is what the method is in the manager in ozw.
int32 GetValueMin(ValueID const& _id);
```

There is your problem: the mismatch turns the exact same bit pattern into wildly different decimal interpretations.

In C "casting" bits like this forces a certain interpretation and there is absolutely no check on what you are doing.

OZW is doing nothin weird or wrong here, the Manager.h file says:

`int32 GetValueMin(ValueID const& _id);`

I also agree with @Fishwaldo on this:

> It doesn't need to. (u)int32 is the maximum size the variable can hold. For somelike a byte, the MAX size is 256 < 2,147,483,647

The int32 version can represent the other types, but the reverse is not true. So why would you want a smaller" one like byte? Suppose the value = 300 and you call get min as a byte, what do you expect it to return? An exception? Zero? The lower 8 bits = 44 in decimal?

That code you wrote in that PR will silently convert 300 to 44, I do not like that very much.

I've tried it by compiling a simplified version:

```
#include <cstdint>
#include <cstdio>

#define int32 int32_t
#define uint8 int8_t

uint8 a;

class ValueID
{
} dummy;

int32 real_limit = 300;

template <class T>
T _GetValueMin(ValueID const &_id)
{
    T limit;
    limit = real_limit;
    return limit;
}

uint8 GetValueMinByte(ValueID const &_id)
{
    return _GetValueMin<uint8>(_id);
}

int main(int, char const **)
{
    uint8 minimum_byte = GetValueMinByte(dummy);
    printf("GetValueMinByte of %d is %d \n", real_limit, minimum_byte);
}
```

prints

`GetValueMinByte of 300 is 44`

kdschlosser, 2019-09-05:
Yeah I know the type in the Cython code is wrong. It has been like this since I started using python-openzwave...

@Fishwaldo 
I do not mind doing the work even if the PR does not get accepted, I am still learning C/C++ so getting my feet wet so to speak is not a waste of time for me.. I learn while I do. For example that PR doesn't work correctly. I still created the PR so you can see the idea and not necessarily for it being a functioning thing. And I knew you would tell me the ups and downs to doing something like that. The reason why i set up the methods like that was really because of API. look at the GetValue it is broken into data type I wanted to keep the same API. I guess the methods could have been created. It would still be nice to be able to get the min and max for the Decimal values from a method instead of having to get it from a value But either way I guess it works. I will have to add support for those 2 values.

I would really love to get rid of Cython. it is a headache. I have already written all of the code in C++ that would allow the removal of Cython. I am having a single issue I have not been able to figure out. and that is passing a python callback for the notifications. I know this can be done and I have extension modules that use this mechanism and I have tried following the code. But i still get an application crash. It is really close to being complete.

petergebruers, 2019-09-09:
Checked with @kdschlosser : after fixing the definition in Cython, the GetValueMin prints as expected.

closed 2019-09-09 by petergebruers