OpenZWave/open-zwave #1921: MinOZW segfaults when doing a clean build/run of the Dev branch

dhylands, 2019-08-30:
It I do a new checkout of the open-zwave repository, checkout the Dev branch and build (debug or release) and run MinOZW then I get a segfault.

I'm building/running with Ubuntu 19.04 and using an Aeotec Z-Stick 5.

Commands executed:
```
git clone https://github.com/OpenZWave/open-zwave.git
cd open-zwave
git checkout Dev
BUILD=debug make
./MinOZW gdb
```

Here's the output on the console:
```
2141 >./MinOZW gdb
GNU gdb (Ubuntu 8.2.91.20190405-0ubuntu3) 8.2.91.20190405-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from .lib/MinOZW...
(gdb) set args /dev/ttyACM0
(gdb) run
Starting program: /home/dhylands/xxx/open-zwave/.lib/MinOZW /dev/ttyACM0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Starting MinOZW with OpenZWave Version 1.6-840-g95a8c3c5-dirty
2019-08-30 16:28:55.628 Error, Cannot find a path to the configuration files at ../../../config/, Using config/ instead...
2019-08-30 16:28:55.628 Info, Reading config/options.xml for Options
2019-08-30 16:28:55.628 Warning, Failed to Parse options.xml: Failed to open file
2019-08-30 16:28:55.628 Info, Options:
2019-08-30 16:28:55.628 Info, 	appendlogfile: false
2019-08-30 16:28:55.628 Info, 	associate: true
2019-08-30 16:28:55.628 Info, 	assumeawake: true
2019-08-30 16:28:55.628 Info, 	autoupdateconfigfile: true
2019-08-30 16:28:55.628 Info, 	configpath: config/
2019-08-30 16:28:55.628 Info, 	consoleoutput: true
2019-08-30 16:28:55.628 Info, 	customsecuredcc: 0x62,0x4c,0x63
2019-08-30 16:28:55.628 Info, 	drivermaxattempts: 5
2019-08-30 16:28:55.628 Info, 	dumptrigger: 4
2019-08-30 16:28:55.628 Info, 	dumptriggerlevel: 1
2019-08-30 16:28:55.628 Info, 	enablesis: true
2019-08-30 16:28:55.628 Info, 	enforcesecurereception: true
2019-08-30 16:28:55.628 Info, 	exclude: 
2019-08-30 16:28:55.628 Info, 	include: 
2019-08-30 16:28:55.628 Info, 	includeinstancelabel: true
2019-08-30 16:28:55.628 Info, 	interface: 
2019-08-30 16:28:55.628 Info, 	intervalbetweenpolls: true
2019-08-30 16:28:55.628 Info, 	language: 
2019-08-30 16:28:55.628 Info, 	logfilename: OZW_Log.txt
2019-08-30 16:28:55.628 Info, 	logging: true
2019-08-30 16:28:55.628 Info, 	networkkey: 
2019-08-30 16:28:55.628 Info, 	notifyondriverunload: false
2019-08-30 16:28:55.628 Info, 	notifytransactions: false
2019-08-30 16:28:55.628 Info, 	performreturnroutes: false
2019-08-30 16:28:55.629 Info, 	pollinterval: 500
2019-08-30 16:28:55.629 Info, 	queueloglevel: 9
2019-08-30 16:28:55.629 Info, 	refreshallusercodes: false
2019-08-30 16:28:55.629 Info, 	reloadafterupdate: AWAKE
[... 84 lines cut ...]
#11 0x00007ffff7f12754 in std::_Rb_tree<long long, std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> >, std::_Select1st<std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> > >, std::less<long long>, std::allocator<std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> > > >::_M_erase_aux (this=0x7ffff7fcbe60 <OpenZWave::Internal::ManufacturerSpecificDB::s_productMap>, Python Exception <class 'gdb.error'> No type named std::_Rb_tree_node<struct std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> >>.: 
__position=...) at /usr/include/c++/8/bits/stl_tree.h:2493
#12 0x00007ffff7f11529 in std::_Rb_tree<long long, std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> >, std::_Select1st<std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> > >, std::less<long long>, std::allocator<std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> > > >::erase[abi:cxx11](std::_Rb_tree_iterator<std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> > >) (
    this=0x7ffff7fcbe60 <OpenZWave::Internal::ManufacturerSpecificDB::s_productMap>, Python Exception <class 'gdb.error'> No type named std::_Rb_tree_node<struct std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> >>.: 
__position=...) at /usr/include/c++/8/bits/stl_tree.h:1141
#13 0x00007ffff7f10af5 in std::map<long long, std::shared_ptr<OpenZWave::Internal::ProductDescriptor>, std::less<long long>, std::allocator<std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> > > >::erase[abi:cxx11](std::_Rb_tree_iterator<std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> > >) (
    this=0x7ffff7fcbe60 <OpenZWave::Internal::ManufacturerSpecificDB::s_productMap>, Python Exception <class 'gdb.error'> No type named std::_Rb_tree_node<struct std::pair<long long const, std::shared_ptr<OpenZWave::Internal::ProductDescriptor> >>.: 
__position=...) at /usr/include/c++/8/bits/stl_map.h:1037
#14 0x00007ffff7f0f143 in OpenZWave::Internal::ManufacturerSpecificDB::UnloadProductXML (this=0x555555579160) at /home/dhylands/xxx/open-zwave/cpp/src/ManufacturerSpecificDB.cpp:276
#15 0x00007ffff7f0f9e8 in OpenZWave::Internal::ManufacturerSpecificDB::mfsConfigDownloaded (this=0x555555579160, driver=0x55555559ba90, file="config/manufacturer_specific.xml", 
    success=true) at /home/dhylands/xxx/open-zwave/cpp/src/ManufacturerSpecificDB.cpp:368
#16 0x00007ffff7f09630 in OpenZWave::Driver::processDownload (this=0x55555559ba90, download=0x7ffff0001ea0) at /home/dhylands/xxx/open-zwave/cpp/src/Driver.cpp:6990
#17 0x00007ffff7f09c83 in OpenZWave::Driver::ProcessEventMsg (this=0x55555559ba90) at /home/dhylands/xxx/open-zwave/cpp/src/Driver.cpp:7082
#18 0x00007ffff7ef26f2 in OpenZWave::Driver::DriverThreadProc (this=0x55555559ba90, _exitEvent=0x55555557b730) at /home/dhylands/xxx/open-zwave/cpp/src/Driver.cpp:426
#19 0x00007ffff7ef22ba in OpenZWave::Driver::DriverThreadEntryPoint (_exitEvent=0x55555557b730, _context=0x55555559ba90) at /home/dhylands/xxx/open-zwave/cpp/src/Driver.cpp:322
#20 0x00007ffff7e8d908 in OpenZWave::Internal::Platform::ThreadImpl::Run (this=0x555555578910) at /home/dhylands/xxx/open-zwave/cpp/src/platform/unix/ThreadImpl.cpp:166
#21 0x00007ffff7e8d8cc in OpenZWave::Internal::Platform::ThreadImpl::ThreadProc (_pArg=0x555555578910) at /home/dhylands/xxx/open-zwave/cpp/src/platform/unix/ThreadImpl.cpp:153
#22 0x00007ffff7a0a182 in start_thread (arg=<optimized out>) at pthread_create.c:486
#23 0x00007ffff7933b1f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb) 
```

The master branch doesn't segfault when doing the same steps.

Running `./MinOZW /dev/ttyACM0` a second time works fine, so it the crash appears to be related to downloading config files.

Fishwaldo, 2019-09-02:
I just pushed a fix here. As usual, strange it doesn't crash on Master as it has the same bug. Thanks for the detailed report - Appreciated.

closed 2019-09-02 by Fishwaldo

dhylands, 2019-09-03:
I suspect that it didn't crash on master because it didn't need to download any updated configs.