OpenZWave/open-zwave #1923: SwitchMultilevel SetLevel crashes with Aeotec Dimmer

mooninite, 2019-08-31:
Device: AEON Labs ZW099 Smart Dimmer 6+
OpenZWave library: 1.6.899
Fedora 30 x86_64

In using Domoticz (latest git checkout) to try and turn on/off or set a level on the dimmer the OpenZWave library crashes.

Removing the version 2 code block (lines 430-454) stops the crash. The dimmer works in all functionality - on/off/dims.

Backtrace:

> (gdb) bt
#0  0x00007f65835da607 in OpenZWave::Internal::CC::SwitchMultilevel::SetLevel (this=0x7f652404a560, 
    _instance=<optimized out>, _level=255 '\377')
    at /usr/src/debug/openzwave-1.6.899-1.fc30.x86_64/cpp/src/command_classes/SwitchMultilevel.cpp:432
#1  0x00007f65835db367 in OpenZWave::Internal::CC::SwitchMultilevel::SetValue (this=0x7f652404a560, _value=...)
    at /usr/src/debug/openzwave-1.6.899-1.fc30.x86_64/cpp/src/value_classes/ValueByte.h:67
#2  0x00007f6583634a52 in OpenZWave::Internal::VC::Value::Set (this=this@entry=0x7f656c055de0)
    at /usr/src/debug/openzwave-1.6.899-1.fc30.x86_64/cpp/src/value_classes/Value.cpp:296
#3  0x00007f658363d3c5 in OpenZWave::Internal::VC::ValueByte::Set (this=this@entry=0x7f6524024190, 
    _value=_value@entry=255 '\377') at /usr/src/debug/openzwave-1.6.899-1.fc30.x86_64/cpp/src/value_classes/ValueByte.cpp:124
#4  0x00007f65836577cd in OpenZWave::Manager::SetValue (this=<optimized out>, _id=..., _value=_value@entry=255 '\377')
    at /usr/src/debug/openzwave-1.6.899-1.fc30.x86_64/cpp/src/Manager.cpp:2594
#5  0x0000559adabbca69 in COpenZWave::SwitchLight (this=0x559adc114070, pDevice=0x7f65240267f0, instanceID=1, 
    value=<optimized out>) at /usr/src/debug/domoticz-4.10718-0.1.fc30.x86_64/hardware/OpenZWave.cpp:1257

petergebruers, 2019-09-01:
This will be fixed when OZW dev branch gets merged into master. It is still being worked on by Justin. Afaik the problem only happens right after inclusion. As a workaround, you should be able to run an unmodified version of Domoticz, but do not use the device right after inclusion, instead reboot. BTW you are right: it is related to the version of the command class (and interview order).

mooninite, 2019-09-01:
The issue I'm reporting is occurring with a device that has been included months ago. It is crashing after a fresh start of Domoticz.

Fishwaldo, 2019-09-01:
It’s the same issue. If Ozwcache*.xml is not present or the node isn’t in it, you will have a crash. If it is present with the node in it, then it will work fine.

Fishwaldo, 2020-02-06:
Duplicate of #1981

closed 2020-02-06 by Fishwaldo