OpenZWave/open-zwave #1926: Device database corrupted

kdschlosser, 2019-09-02:
It would appear as tho there is an issue with the device database.

This link is for General Electric.
http://www.openzwave.com/device-database/0020

and this link is for the only device under General Electric
http://www.openzwave.com/device-database/0020:1390:8007
Which is showing a manufacturer of CyberHouse. I do not think it is where it is supposed to be.

There are no other GE devices listed. It should be populated with at the very least 23 devices.

I have not checked every device xml against the product database and what the zwave alliance states. But I did run across this issue. it could be more widespread then just this single occurrence. I am not sure more investigation would need to be done.

This appears to be incorrect in the device xml file. ge/12724-dimmer.xml.
the device xml file states it is a Smart Fan controller and on the zwave alliance product specification it states it is a wall dimmer. 

```xml
 <MetaData>
    <MetaDataItem name="OzwInfoPage">http://www.openzwave.com/device-database/0063:3031:4944</MetaDataItem>
    <MetaDataItem name="ProductPic">images/ge/12724-dimmer.png</MetaDataItem>
    <MetaDataItem id="3031" name="ZWProductPage" type="4944">https://products.z-wavealliance.org/products/1197/</MetaDataItem>
    <MetaDataItem name="Description">REMOVED TO SAVE SPACE</MetaDataItem>
    <MetaDataItem id="3031" name="FrequencyName" type="4944">U.S. / Canada / Mexico</MetaDataItem>
    <MetaDataItem name="Name">In-Wall Smart Fan Control </MetaDataItem>
    <MetaDataItem name="ProductPage">http://www.ezzwave.com</MetaDataItem>
    <MetaDataItem id="3031" name="Identifier" type="4944">12724 / ZW3005</MetaDataItem>
    <ChangeLog>
      <Entry author="Justin Hammond - Justin@dynam.ac" date="08 May 2019" revision="3">Initial Metadata Import from Z-Wave Alliance Database - https://products.z-wavealliance.org/products/1197/xml</Entry>
      <Entry author="Justin Hammond - Justin@dynam.ac" date="08 May 2019" revision="4">Updated Metadata Import from Z-Wave Alliance Database - https://products.z-wavealliance.org/products/1201/xml</Entry>
      <Entry author="Justin Hammond - Justin@dynam.ac" date="08 May 2019" revision="5">Updated Metadata Import from Z-Wave Alliance Database - https://products.z-wavealliance.org/products/1202/xml</Entry>
      <Entry author="Justin Hammond - Justin@dynam.ac" date="08 May 2019" revision="6">Updated Metadata Import from Z-Wave Alliance Database - https://products.z-wavealliance.org/products/1221/xml</Entry>
      <Entry author="Justin Hammond - Justin@dynam.ac" date="02 Jun 2019" revision="7">Updated Metadata Import from Z-Wave Alliance Database - https://products.z-wavealliance.org/products/2103/xml</Entry>
      <Entry author="Justin Hammond - Justin@dynam.ac" date="02 Jun 2019" revision="8">Updated Metadata Import from Z-Wave Alliance Database - https://products.z-wavealliance.org/products/2506/xml</Entry>
    </ChangeLog>
    <MetaDataItem id="3033" name="ZWProductPage" type="4944">https://products.z-wavealliance.org/products/1201/</MetaDataItem>
    <MetaDataItem id="3033" name="FrequencyName" type="4944">U.S. / Canada / Mexico</MetaDataItem>
    <MetaDataItem id="3033" name="Identifier" type="4944">12729 / ZW3004</MetaDataItem>
    <MetaDataItem id="3034" name="ZWProductPage" type="4944">https://products.z-wavealliance.org/products/1221/</MetaDataItem>
    <MetaDataItem id="3034" name="FrequencyName" type="4944">U.S. / Canada / Mexico</MetaDataItem>
    <MetaDataItem id="3034" name="Identifier" type="4944">45743</MetaDataItem>
    <MetaDataItem id="3131" name="ZWProductPage" type="4944">https://products.z-wavealliance.org/products/2506/</MetaDataItem>
    <MetaDataItem id="3131" name="FrequencyName" type="4944">U.S. / Canada / Mexico</MetaDataItem>
    <MetaDataItem name="ProductManual">https://Products.Z-WaveAlliance.org/ProductManual/File?folder=&amp;filename=Manuals/2506/Binder2.pdf</MetaDataItem>
    <MetaDataItem id="3131" name="Identifier" type="4944">14287/ZW4002</MetaDataItem>
    <MetaDataItem name="InclusionDescription">REMOVED TO SAVE SPACE</MetaDataItem>
    <MetaDataItem name="ExclusionDescription">REMOVED TO SAVE SPACE</MetaDataItem>
    <MetaDataItem name="ResetDescription">REMOVED TO SAVE SPACE</MetaDataItem>
  </MetaData>
```

this is a partial of the xml from https://products.z-wavealliance.org/products/1197

```xml
<Id>1197</Id>
  <Name>GE Z-Wave In-Wall Smart Dimmer</Name>
  <Description>"REMOVED TO SAVE SPACE"
</Description>
  <Brand>GE</Brand>
  <Identifier>12724 / ZW3005</Identifier>
  <CertificationNumber>ZC08-14110010</CertificationNumber>
  <OemVersion>3.0e</OemVersion>
  <HardwarePlatform>ZM3102</HardwarePlatform>
  <ZWaveVersion>4.55.00</ZWaveVersion>
  <LibraryType>Routing Slave</LibraryType>
  <SpecificDeviceClass>Multilevel Power Switch</SpecificDeviceClass>
  <GenericDeviceClass>Switch Multilevel</GenericDeviceClass>
  <DeviceType>Central Controller</DeviceType>
  <ManufacturerId>0x0063</ManufacturerId>
  <ProductTypeId>0x4944</ProductTypeId>
  <ProductId>0x3031</ProductId>
  <FrequencyName>U.S. / Canada / Mexico</FrequencyName>
  <ProductUrl>http://www.ezzwave.com</ProductUrl>
```

http://www.openzwave.com/device-database/0063:3031:4944
Manufacturer ID: 0063
Manufacturer: GE
Product: 12724/ZW3005
Product ID: 3031
Product Type: 4944
Description: GE Z-Wave In-Wall Smart Dimmer

I do know that when I start openzwave when there is no user config file that I get a mess of product collisions reported by openzwave.

petergebruers, 2019-09-02:
Suspicious to say the least. I cannot tell much about that online database, but I do have some info about /config/manufacturer_specific.xml

The GE /JASCO stuff is in there, under `<Manufacturer id="0063" name="GE (Jasco Products)">`

Git blame tells me:

Commit fb82b14e
by eiddor, 11/18/2016 12:35 AM

Introduces:

<Product type="4944" id="3034" name="12730 Fan Control Switch" config="ge/12724-dimmer.xml"/>

That is a while ago in the OZW 1.4 era
 
So that is the first time someone deemed the 12724-dimmer.xml is suitable for type="4944" id="3034". That is the oldest reference I can find that has "Fan Control" in its name and "dimmer" in the config file name.

It looks odd but I cannot judge that without more information. I can say it was not caused by some automatic process, the data was contributed.

So now in 2019 we have these four, all using that 12724-dimmer.xml

```
		<Product type="4944" id="3031" name="12724 3-Way Dimmer Switch" config="ge/12724-dimmer.xml"/>
		<Product type="4944" id="3033" name="12729 3-Way Dimmer Switch" config="ge/12724-dimmer.xml"/>
		<Product type="4944" id="3034" name="12730 Fan Control Switch" config="ge/12724-dimmer.xml"/>
		<Product type="4944" id="3131" name="14287 Fan Control Switch" config="ge/12724-dimmer.xml"/>
```

Do you own a "Fan Control"? Can you make a config file for it?

petergebruers, 2019-09-02:
Forgot to mention... Even with that suspicious looking information, users mind find all devices are "usable" because these probably all expose a multilevel switch, which gives the user a slider in their gateway software to adjust dimming level or speed...

Fishwaldo, 2019-09-03:
Fixed. One of the build scripts was erroring out.

closed 2019-09-03 by Fishwaldo

petergebruers, 2019-09-03:
Thanks for quickly fixing <http://www.openzwave.com/device-database> - that looks more like it :)

kdschlosser, 2019-09-03:
The thing is with a fan controller from GE it would not be a slider type of control. it would be a choice. Off, Low, Medium and High.

petergebruers, 2019-09-03:
I am not debating this...

You own one? Then you can fix the config file and contribute?

kdschlosser, 2019-09-05:
I can do that... once I install it in the wall. LOL