OpenZWave/open-zwave #1943: options.xml documentation could be better

electrofloat, 2019-09-18:
Searching through documentation I only found this on how to enable/disable/configure logging of OpenZWave: https://github.com/OpenZWave/open-zwave/wiki/Config-Options
I have 2 issues with this:
- Missing default values
- no mention of what these options do: SaveLogLevel, QueueLogLevel, DumpTriggerLevel

Also a logrotate would be nice, but it is still under development:
https://github.com/OpenZWave/open-zwave/issues/465

electrofloat, 2019-09-19:
So to answer my own questions after checking the code:
- SaveLogLevel (https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/Options.cpp#L105):
  - default value: LogLevel_Detail
  - description: Save (to file) log messages equal to or above LogLevel_Detail
- QueueLogLevel (https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/Options.cpp#L106):
  - default value: LogLevel_Debug
  - description: Save (in RAM) log messages equal to or above LogLevel_Debug
- DumpTriggerLevel (https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/Options.cpp#L107):
  - default_value: LogLevel_None
  - description: Default is to never dump RAM-stored log messages. (checking the code dumping means writing the log from memory to file https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/platform/unix/LogImpl.cpp#L218) 

Also the Enum containing the values for loglevels does not really match the docs. According to the source LogLevel_Invalid is 0, LogLevel_None is 1, etc. https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/platform/Log.h#L52

Fishwaldo, 2019-09-30:
Patches always welcome - for Options.xml and the wiki.