OpenZWave/open-zwave #1948: Zipato NE-NAS-AB02Z.EU Indoor Siren config

kervern, 2019-09-21:
image et JSON sont à placer dans : plugins/openzwave/core/config/devices/zipato_305
Le fichier XML est à placer dans : plugins/openzwave/resources/openzwaved/config/zipato

Pour finir, il faut modifier le fichier : plugins/openzwave/resources/openzwaved/config/manufacturer_specific.xml
Dans ce fichier, il faut rechercher le bloc commençant par : <Manufacturer id="0131" name="Zipato">
A la fin de ce bloc, il faut ajouter la ligne suivante : <Product type="0003" id="1083" name="Siren AB02Z" config="zipato/Zipato-AB02Z.xml"/>
(juste au dessus de : </Manufacturer>)

petergebruers, 2019-09-21:
I can read French but I am not sure what you are trying to say...

petergebruers, 2019-09-23:
Closed because of lack of feedback

closed 2019-09-23 by petergebruers