OpenZWave/open-zwave #1949: No idea where to start with this error that causes my Z-wave network to hang

x99percent, 2019-09-22:
I have reset my controller and paired only two other nodes: a Yale YRD240 and YRD220. Both locks have been added using "add secure node" and are within 4m of the controller. Every node is a neighbor of every other node.

Any time I remotely operate the locks, the lock will immediately perform the lock/unlock action, then the Z-wave network will hang for anywhere from a few seconds to a minute (or more?) until an error is thrown. This can be seen in the log at 13:41:30 and 13:41:51.

All I've done here is...

- start the Z-wave network
- wait for initial queries complete at 13:32:57
- send an unlock command at 13:41:20 (unlocks immediately)
- wait for the error at 13:41:30
- send a lock command at 13:41:41 (locks immediately)
- wait for the error at 13:41:51

OZW_Log.txt: https://pastebin.com/vbcJ5Paj
zwcfg_0xf34bf775.xml: https://pastebin.com/VR56T55Q

x99percent, 2019-09-22:
I believe this has something to do with ```CommandClass: DOOR LOCK LOGGING```

Fishwaldo, 2019-09-28:
I can't deduce anything as you have filtered the log file, but based on what i see, it appears to be a device bug..

x99percent, 2019-09-28:
The log isn't filtered.

Fishwaldo, 2019-09-30:
Could you please turn up logging to Level 9 (Debugging) and repost the log file.

The Device is behaving weirdly with regards to encrypted packets. I can't tell if its a OZW bug, or device bug at the moment.

x99percent, 2019-10-02:
I've re-added the rest of my devices to the Z-wave network. Every node throws an error as the network is coming up. Then I lock and unlock Node003, which causes the delay before the error is thrown.

https://pastebin.com/8H5Lj56d

x99percent, 2019-10-07:
BUT... both Node 2 and 3 (the locks) have been "secure added" with a network key set up. If that were not the case, I shouldn't be able to operate the locks and change user codes.... right?

Fishwaldo, 2019-10-07:
There is something even worse going on then. If the Lock is working, then its vulnerable to a downgrade attack - as the DoorLock_Set command is unencrypted:
>2019-10-02 15:14:02.334 Info, Node002, Sending (Send) message (Callback ID=0x5a, Expected Reply=0x04) - DoorLockCmd_Set (Node=2): 0x01, 0x0a, 0x00, 0x13, 0x02, 0x03, 0x62, 0x01, 0xff, 0x25, 0x5a, 0x04

(the 0x62 is the DoorLock CommandClass, it should actually be 0x98 (The Security CC) and a longer string of bytes afterwards which is the encrypted payload. 

The rest of the Log File, there is most definitely something screwed up with these devices or your OZW installation. Honestly, I can't tell if its just these devices that are messed up, or if you have multiple other issues. 

To confirm, I'd wipe you installation and do everything from scratch, that will rule out anything OZW, but before you do that, I'd confirm the above issue with regards to the DoorLock accepting clear text commands. If thats really the case, I'd be advising to remove them right now (or at least find a way to disable the Z-Wave comms on them) as it means basically anybody with a SDR could open your locks. 

(goes without mentioning - I'd be alerting Yale to this issue as well. If it is unlocking, its a MAJOR security issue)

Fishwaldo, 2019-10-07:
(btw, not sure how old these locks are, or if you are on recent firmware, but a researcher did contact me about a downgrade attack against Z-Wave around June/July 2015.)

x99percent, 2019-10-07:
I've 100% reset the controller and network many times while testing. Even tried pairing only the locks to see if the problem was repeatable. Also getting the same error using the OZW Control Panel that uses OZW 1.6. Do you want to see a log?

Not sure how I'd go about checking/upgrading the locks' firmware, but I'm all ears.

x99percent, 2019-10-07:
Also, in the log that you're addressing, I only operated Node 3, not Node 2. Does that make a difference?

Fishwaldo, 2019-10-08:
> 
> 
> Also, in the log that you're addressing, I only operated Node 3, not Node 2. Does that make a difference?

Thats not what the Logs say:
>2019-10-02 15:14:01.067 Info, Node002, Value::Set - COMMAND_CLASS_DOOR_LOCK - Locked - 0 - 1 - True
2019-10-02 15:14:01.067 Info, Node002, Value_Lock::Set - Requesting lock to be Locked

Fishwaldo, 2019-10-08:
Honestly - The issue you are seeing is something unique to your enviroment. When I say rebuild, I'm even talking about blowing away your existing OZW install and start from scratch.

It could be the Locks you have, it could be something funky on your install. All I can say, is that its something about your environment, and the logs dont make sense in several cases.

x99percent, 2019-10-08:
Seriously, I've done exactly that. I didn't know what else to do, so I made a post.

Here's my plan:
- reset the controller
- set up a new instance of the OZW Control Panel
- securely pair one lock
- wait for query completion
- stop the network
- wipe the cache file and log
- start the network
- wait for query completion
- operate the lock
- look for errors
- post the OZW_Log.txt and options.xml in its entirety, including a throwaway security key

x99percent, 2019-10-08:
- started the network
- waited for all queries to complete and everything to settle
- sent a lock command
- waited for the error and everything to settle
- sent an unlock command
- waited for the error and everything to settle
- stopped the network

options.xml: https://pastebin.com/ry1TkWNc
OZW_Log.txt: https://pastebin.com/tn8uq7m1
ozwcache_0xd104e56d.xml: https://pastebin.com/0LBxtuuV

CookieMonster87, 2019-10-08:
I have the exact same problem for the two Danalock V3 Z-wave i have. The locks are responding in a few seconds but the network hangs for 15-60 seconds after. 
Think this is a open Zwave problem. Tried the locks against smartthings hub without this problem. I have 45 nodes and the locks are the only one that make my network hang. No errors on other nodes.

CookieMonster87, 2019-10-08:
`2019-10-08 14:26:11.671 Error, Node044, ERROR: Dropping command, expected response not received after 1 attempt(s)
2019-10-08 14:26:11.676 Error, Node044, ERROR: ZW_SEND_DATA could not be delivered to Z-Wave stack
2019-10-08 14:26:21.675 Error, Node044, ERROR: Dropping command, expected response not received after 1 attempt(s)
2019-10-08 14:28:19.178 Error, Node042, ERROR: Dropping command, expected response not received after 1 attempt(s)
2019-10-08 14:28:36.450 Error, Node044, ERROR: Dropping command, expected response not received after 1 attempt(s)`

errors I get when the network hangs. Only get errors in the log when using the locks. Both locks are added secure and tried to add them 2-3 times just to be sure the secure include was 100%.

Matchlighter, 2019-12-05:
I think I'm hitting this too. I've got a Schlage BE469ZP. Responds very quickly when I send it a command via ZWave. But after that, my whole network freezes for several seconds - I can't control the lock or any other devices. I've noticed this when I physically lock/unlock the lock too. It's so annoying that my light switches stop working when someone opens the door...

I'm running Home Assistant on a Pi4. Let me know if some additional logs would be helpful. From what I've researched, other people that have had this problem haven't had it with commercial hubs or even OpenHAB, so to me it looks like something in OZW. I'm a Dev, so if anyone can offer some plausible theories, I could start looking at how OpenHAB and OZW vary in their ZWave implementations.

stiansoevik, 2019-12-05:
I have experienced similar issues with Popp Keypad + Danalock V3. This is my theory about what is happening:

- I have quite many Z-wave devices, most with default configuration. They send power measurements and various other reports all the time.
- When unlocking, especially with keypad, several encrypted messages are sent back and forth, both between keypad and lock, but also to the controller. These take more time to send than other messages.
- Since I typically unlock the door when I come home, many other things are naturally triggered in the same time frame - motion detection in the entrance, light automations triggering when I come home, door sensor changes state, and so on.
- During all this, all the nodes usually "spamming" power reports etc are blocked, so they enqueue their messages.

All this causes a "network storm", which either can be a sufficient cause in itself of the following freeze, or maybe even amplify it due to back off algorithms or similar.

In my case, the problem was reduced significantly by finding chatty devices by inspecting the logs, and configuring them to be quieter.

x99percent, 2019-12-05:
As stated above, I have tested this with a minimal configuration... just the controller and one lock. This isn't from devices "spamming", as there weren't other devices to send spam.

@Matchlighter , be prepared for the OZW dev to say that we're all making this up.

stiansoevik, 2019-12-05:
I replied in general to all people experiencing such problems.

So, you have no other Z-Wave devices in your house, or did you turn all off, or isolate them by other means? AFAIK, Z-Wave only communicates on a single frequency. I am not sure whether nodes will give up reporting if the controller does not acknowledge, but if they won't, I think the situation could be the same as I described above.

x99percent, 2019-12-05:
Scroll back to my post from Oct 8, along with the post just before that lays out the test plan. 

Even with all of my devices paired and operational (only 6 + controller at this time), my Z-Wave network is far from "noisy".

Matchlighter, 2019-12-06:
I think to say that it's too noisy just doesn't make sense. Users have reported that they have tested with a single lock on the network. In my case, I have 3 switches and a lock - super quiet: tailing the log doesn't show any packets until I actually do something.

There's also plenty of reports of users trying a commercial hub or OpenHAB and not having the same issues (I intend to try this as well this weekend).

Fishwaldo, 2019-12-11:
So here is the issue guys:
1) I have older Locks (Pre ZWave+ devices) that work correctly. 
2) Other devices that use S0 work correctly
3) the Logs that @x99percent posted are just so weird I can't even begin to figure out whats going on as it just doesn't make sense.
4) In reality - if you want this fixed, I need one of these problematic devices. Challenge - A lot of these devices are around ~$1000 where I live. (Shipping etc). Its just way over my budget.

So - Unless someone else can determine exactly whats going on, its going to continue to be unaddressed.

x99percent, 2019-12-12:
I stated very clearly on Oct 7 what steps I took to produce the logs I posted on Oct 8. What's weird about those logs?
  
I'll gladly follow any procedure you'd like to help diagnose the issue, but I thought that the logs from Oct 8 would be useful.

Matchlighter, 2020-01-14:
So I've been playing with a few things, and at the behest of a friend, tried re-including the locks at close range. So I excluded the lock, moved my Hass Setup to within 5' of the lock, and did a Secure Include. After moving my setup back to it's normal location and testing a few times, I have not been able to repro the network lockup issues.
It's not a very explanatory solution (and it seems _super_ odd), but it may help. I'm going to be keeping an eye on things, as well as monitor it on my second lock once I get it installed.