OpenZWave/open-zwave #1950: Wrong thermostat Horstmann HRT4-ZW setpoint type

sucesor, 2019-09-23:
I use ioBriker smart home system. The thermostat Horstmann HRT4-ZW has setpoint type Cooling_1_1 instead Heating_1_1 from version 1.6.0 to 1.6.1

Fishwaldo, 2019-09-30:
Eh? could you please explain as I don't follow?

AlCalzone, 2019-10-02:
I'll try to chime in. ioBroker has changed the installation routine for OZW so it uses the downloads from old.openzwave.com as you suggested. This probably has lead to an update of OZW for @sucesor.

It seems that the mentioned thermostat now has a `Cooling` setpoint type when it should be `Heating`

sucesor, 2019-10-06:
After the last update of the adapter ioBroker.zwave the setpoint type became Heating.
This issue resolved.
Thanks.

closed 2019-10-06 by sucesor