OpenZWave/open-zwave #1951: inconsistant order of data returned from GetValueListItems

kdschlosser, 2019-09-23:
I am not 100% sure if the issue is in OpenZWave or yet again another possible problem with python-openzwave. The returned information from both GetValueListItems and GetValueListValues on the python side of things use almost identical code except for the decleration of the vector passed to either of the manager functions. This is why I am thinking it is not on the python end of things. If you notice the order of the items returned from GetValueListItems changes between runs. so there is no solid way to map the value to the label, Now the values do stay consistent between runs. First thing is the fact that the array indexes of the returned list items changing means that an application is not going to be able to only store the index number instead of the string value it's self. which I know is one of the larger objectives in 1.6 was to shift away from doing that.

This mapping from the value to the list item is needed because there are occasions where there may be a node that is not 100% supported. and in this case the data returned when getting the data from the value is going to be the underlying byte(s) passed by the node. If an application and it's users use that information as a constant then when support is added then things will break. So I am providing a wrapper around the data where the app is able to reference either the index or the underlying byte value instead of using a string. In the case of a node being supported I am only able to get the data returned from the values as a string. so the index order of the list items and the list values need to match up 100% of the time. 

Here are some examples of what I am seeing. it changes every single time i run OZW this is just 2 runs to show what I am talking about.
```
RUN 1
VALUE_ID: 22799473140563988
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 81
LIST SELECTIONS: ['Enable', 'Disable']     <---- incorrect
LIST VALUES: [0, 1]

RUN 2
VALUE_ID: 22799473140563988
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 81
LIST SELECTIONS: ['Enable', 'Disable']     <---- incorrect
LIST VALUES: [0, 1]   <------ correct

RUN 1
VALUE_ID: 61924494903345172
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 220
LIST SELECTIONS: ['3', '8', '6', '10', '7', '4', '1', '9', '5', '2']    <------ incorrect
LIST VALUES: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10]   <------ correct

RUN 2
VALUE_ID: 61924494903345172
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 220
LIST SELECTIONS: ['7', '3', '2', '5', '4', '10', '8', '9', '6', '1']    <------ incorrect & does not match run 1
LIST VALUES: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10]

------------------------------
RUN 1
VALUE_ID: 68116944390979604
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 242
LIST SELECTIONS: ['Enable', 'Disable']   <---- incorrect
LIST VALUES: [0, 1]  <---- correct

RUN 2
VALUE_ID: 68116944390979604
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 242
LIST SELECTIONS: ['Enable', 'Disable']  <---- incorrect
LIST VALUES: [0, 1]  <---- correct
------------------------------

------------------------------
RUN 1
VALUE_ID: 70931694158086164
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 252
LIST SELECTIONS: ['Unlock', 'Lock']   <---- correct
LIST VALUES: [0, 1]  <---- correct

RUN 2
VALUE_ID: 70931694158086164
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 252
LIST SELECTIONS: ['Lock', 'Unlock']   <---- incorrect & does not match run 1
LIST VALUES: [0, 1]  <---- correct
------------------------------

------------------------------
RUN 1
VALUE_ID: 97107988
VALUE CC: COMMAND_CLASS_SWITCH_ALL
VALUE INDEX: 0
LIST SELECTIONS: ['Disabled', 'On Enabled', 'On and Off Enabled', 'Off Enabled']   <---- incorrect
LIST VALUES: [0, 1, 2, 255]   <---- correct

RUN 2
VALUE_ID: 97107988
VALUE CC: COMMAND_CLASS_SWITCH_ALL
VALUE INDEX: 0
LIST SELECTIONS: ['Disabled', 'On Enabled', 'Off Enabled', 'On and Off Enabled']   <---- correct & does not match run 1
LIST VALUES: [0, 1, 2, 255]  <---- correct
------------------------------

------------------------------
RUN 1
VALUE_ID: 844425024241684
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 3
LIST SELECTIONS: ['No', 'Yes']   <---- correct
LIST VALUES: [0, 1]   <---- correct

RUN 2
VALUE_ID: 844425024241684
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 3
LIST SELECTIONS: ['No', 'Yes']   <---- correct
LIST VALUES: [0, 1]   <---- correct
------------------------------

------------------------------
RUN 1
VALUE_ID: 1125900000952340
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 4
LIST SELECTIONS: ['No', 'Yes']   <---- correct
LIST VALUES: [0, 1]   <---- correct

RUN 2
VALUE_ID: 1125900000952340
VALUE CC: COMMAND_CLASS_CONFIGURATION
VALUE INDEX: 4
LIST SELECTIONS: ['No', 'Yes']   <---- correct
LIST VALUES: [0, 1]   <---- correct
------------------------------

------------------------------
RUN 1
VALUE_ID: 147439636
VALUE CC: COMMAND_CLASS_SWITCH_ALL
VALUE INDEX: 0
LIST SELECTIONS: ['Disabled', 'On Enabled', 'On and Off Enabled', 'Off Enabled']   <---- incorrect (I think)
LIST VALUES: [0, 1, 2, 255]

RUN 2
VALUE_ID: 147439636
VALUE CC: COMMAND_CLASS_SWITCH_ALL
VALUE INDEX: 0
LIST SELECTIONS: ['Disabled', 'On Enabled', 'Off Enabled', 'On and Off Enabled']  <---- correct (I think) & does not match run 1
LIST VALUES: [0, 1, 2, 255]  <---- correct
------------------------------

------------------------------
RUN 1
VALUE_ID: 148684820
VALUE CC: COMMAND_CLASS_POWERLEVEL
VALUE INDEX: 0
LIST SELECTIONS: ['-4dB', '-1dB', '-2dB', '-8dB', '-6dB', 'Normal', '-3dB', '-5dB', '-9dB', '-7dB']  <---- incorrect
LIST VALUES: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9]  <---- correct

RUN 2
VALUE_ID: 148684820
VALUE CC: COMMAND_CLASS_POWERLEVEL
VALUE INDEX: 0
LIST SELECTIONS: ['-6dB', 'Normal', '-7dB', '-1dB', '-9dB', '-4dB', '-3dB', '-2dB', '-5dB', '-8dB'] <---- incorrect & does not match run 1
LIST VALUES: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9] <---- correct
------------------------------
```

petergebruers, 2019-09-23:
I am not going to claim this 100% settles this case, but I had a quick look at my test setup using ozwcp and all "list" objects seem fine. Maybe I did not try hard enough to make it fail?

ozwcp uses this code to build the data for the lists:

```cpp
		if (id.GetType() == ValueID::ValueType_List)
		{
			vector<string> strs;

			Manager::Get()->GetValueListItems(id, &strs);
			valueElement->SetAttribute("count", strs.size());
			string str;
			Manager::Get()->GetValueListSelection(id, &str);
			valueElement->SetAttribute("current", str.c_str());
			for (vector<string>::iterator it = strs.begin(); it != strs.end(); it++)
			{
				TiXmlElement *itemElement = new TiXmlElement("item");
				valueElement->LinkEndChild(itemElement);
				TiXmlText *textElement = new TiXmlText((*it).c_str());
				itemElement->LinkEndChild(textElement);
			}
		}
```

I've sampled "Powerlevel" and it lists (top to bottom) Normal, -1dB, ... -9dB as expected.

I've also checked a "Switch All" list. It lists, as expected, 'Disabled', 'Off Enabled', 'On Enabled', 'On and Off Enabled'. That is hard coded in SwitchAll.cpp, array c_switchAllStateName.

I've restarted ozwcp three times.

Are you sure it is not python or messing up the order?

kdschlosser, 2019-09-23:
here is the cpp interface code for GetValueListItems and GetValueListValues. as you can see they are identical. and GetValuesListValues does not change at all. it remains constant. 

```python
    def getValueListItems(self, value_id):
        cdef vector[string] vect
        ret = set()
        if values_map.find(value_id) != values_map.end(): <---- this is a mapping of integer value id to ValueID objects
            if self.manager.GetValueListItems(values_map.at(value_id), &vect): < ---- passing the ValueID object and a pointer to the vector to GetValueListItems
                while not vect.empty() : < ---- iterate over the vector to get the strings
                    temp = vect.back()
                    ret.add(temp.c_str()) < --- add the vector to our python array
                    vect.pop_back()
        return list(item for item in ret) < ---- return the results contained within the set array

    def getValueListValues(self, value_id):
        cdef vector[int32_t] vect
        ret = set()
        if values_map.find(value_id) != values_map.end():
            if self.manager.GetValueListValues(values_map.at(value_id), &vect):
                while not vect.empty():
                    temp = vect.back()
                    ret.add(temp)
                    vect.pop_back()
        return list(item for item in ret)
```

what set() is it's an array that will only allow a single entry of whatever it is you add. so if i did this code
```python
temp = set()
temp.add(1)
temp.add(1)

for item in temp:
    print(item)
```
it is only going to output `1` a single time. 

```cpp
        if (id.GetType() == ValueID::ValueType_List)
		{
			vector<string> strs;

			Manager::Get()->GetValueListItems(id, &strs);
                        while (!strs.empty()) 
                        {
                                string temp = vect.back();
                                strs.pop_back();
                                cout << temp;
			}
		}
```

what would something like this do? this is pseudo code and is probably not correct (not a C guy) but it gives the basic idea. I am going to try changing up the code in the cpp connector to iterate using the index instead maybe that will yield a different result. 
 if you can try something similiar to what you see above or maybe you already know it is wrong and can explain as to why that would be most helpful.,

kdschlosser, 2019-09-23:
and because the code is almost identical between the 2 methods getValueListItems and getValueListValues the same issue would also appear for getValueListValues. and it does not. It always returns the proper thing. another funny thing is that GetValueListValues fills the vector in reverse order. so the last element becomes the first item in our returned results. I am going to remove the use of `set()` from the code. It might be mucking up the data somehow. one other thing I noticed is there is no temporary storage declared for the data that is gotten from pop_back. I still find it strange that the 2 methods behave in completely different manners.

kdschlosser, 2019-09-23:
ok so here is an update. While I am not exactly sure as to why the code that has been in place does not work properly  (It looks as tho it should), and I am not exactly sure as to why the use of back is returning the first item in the vector and pop_back is removing the first item in the vector as well. there is something flawed with Cython in respect to that. I am not sure why I am also getting 2 different results between GetValueListItems and GetValueListValues, the latter being stable and returns the proper data. even tho both use almost identical code except for the data contained in the vector. Again this must be some kind of an issue with Cython is what I am thinking. 

I have changed the code to do a simply count from 0 to vector.size() - 1 and reference the vector using index numbers seems to have fixed the problem.

``` python
def getValueListItems(self, value_id):
        """
        Gets the list of items from a list value

        :param value_id: The ID of a value.
        :type value_id: int

        :return: The list of possible values
        :rtype: list
        """
        cdef vector[string] vect
        cdef string temp
        cdef size_t size
        ret = []

        if values_map.find(value_id) != values_map.end():
            if self.manager.GetValueListItems(values_map.at(value_id), &vect):
                size = vect.size()

                for i in range(size):
                    temp = vect[i]
                    ret += [_str(temp.c_str())]

        return ret

    def getValueListValues(self, value_id):
        """
        Gets the list of values from a list value.

        :param value_id: The ID of a value.
        :type value_id: int

        :return: The list of values
        :rtype: list
        """
        cdef vector[int32_t] vect
        cdef int32_t temp
        cdef size_t size
        ret = []

        if values_map.find(value_id) != values_map.end():
            if self.manager.GetValueListItems(values_map.at(value_id), &vect):
                size = vect.size()

                for i in range(size):
                    temp = vect[i]
                    ret += [temp]

        return ret
```

really strange behavior with the previous code. I am going to check for other code that uses that same kind of a code block to get the contents of a vector and I am going to change them to the above.

closed 2019-09-23 by kdschlosser