OpenZWave/open-zwave #1959: Strange behavior from polling settings.

kdschlosser, 2019-10-01:
OK this is the test setup.

I have a GE ZWave Dimmer. it is not ZWave Plus I want to poll the value. This is the only value that is being polled on my whole network.

The dimmer ramps up and down so it changes level over time. the length of time is adjustable. so If i set the length of time to say 30 seconds from 0-100 that is how long it is going to take.

There are 2 issues I am seeing first issue is i set the value polling to 1 and I can watch the stream of value changed events come in. If I then terminate the program and restart it even tho the polling is still set to a 1 I do not get the stream anymore. Not until i set the polling back to a 0 and then to 1 again does the notification stream start up again. It appears as tho the polling is not persistent between restarts of OpenZWave.

The second issue is this when i set the level to 100 and the starting point is 0 with the above polling set i get just about every single percentage change in the notifications. If I then set options SuppressValueRefresh to True  and IntervalBetweenPolls to False I would imagine that it would suppress the notifications if the value data is the same. so I should see the same notifications when it is turned off minus the duplicates.. This is not the case. I will see only a single notification when it reaches the end of it's transition.

I set the options PollInterval to 5 and also tried using the Manager::SetPollInterval to 5000ms  and False. I am assuming that the Manager::SetPollInterval and the options poll interval set the same thing the options simply provide a mechanism to be able to set the polling on startup.

I am not 100% sure as to why there is this strange behavior from setting the polling count on a value and also from the options SuppressValueRefresh. The options and the value polling count are set directly to the manager so there really isn't much in the way of python code involved in that. and they are simple data conversion from python to c++ so there shouldn't be much there either.

when i have the value poll count set to 1 and close and then reopen the application the value is still set to 1 so openzwave is saving the data it is just not using it when everything starts up.

kdschlosser, 2019-10-05:
apparently it takes a long while before the polling starts. that is why I was getting this behavior. I was not waiting long enough after the AllNodesQueried* notification to see the poling take effect.

closed 2019-10-05 by kdschlosser

chilicheech, 2020-08-12:
Ah, this info was very helpful. I was having an issue with nodes not being polled and it was because I wasn't getting a `AwakeNodesQueried` notification. I removed stale nodes that were somehow still listed and got the notification and polling resumed. Cheers,