OpenZWave/open-zwave #1960: Confusion about timezones in TimeParametersCC

AlCalzone, 2019-10-02:
This is not really an issue, but I have a question about the TimeParametersCC class and how it handles timezones. I've been playing around a bit with that and my MCOHome MH7 thermostat. 

The Z-Wave specs clearly state that this CC is used to set the current time in **UTC**. As far as I can tell, OZW is using local time instead: https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/TimeParameters.cpp#L162

So this seems non-compliant, but when I send a TimeParametersSet command using UTC, my thermostat seems to interpret that as local time and is off by two hours. Do you know if using local time in OZW was a concious decision or just because you didnt't know better at the time?
@Fishwaldo

petergebruers, 2019-10-03:
I can only confirm that there is a long discussion on the Fibaro forum regarding sending time in UTC. I don't know the internals of the Fibaro HC, it is proprietary, but from the problem description I deduce that a HomeCenter implements the spec, meaning "send UTC". This causes the followin problem... I don't own this device but it seems after power on reset, all MH7 will get time from the controller and display UTC, which is not what most people want.

https://forum.fibaro.com/topic/29618-mcohome-mh7-thermostat-wrong-time-after-reboot/

I don't know wether this "was a conscious decision or just because you didn't know better at the time?" because I only started working on the OpenZWave source code quite recently...

If I understand you correctly, you are not actually complaining because for your MH7, OZW does the right thing, though that is not compliant? So if we would make it compliant, we would have to add a flag for the MH7 to revert that?

AlCalzone, 2019-10-03:
No, I'm not complaining. I was just curious. 

As I see it, that device has no other ways of retrieving the correct (local) time or timezone other than interpreting the TimeParametersSet commands it gets sent. So if that is UTC, the device will show UTC.
TimeCC V1 is supported but not controlled, so it won't request the local time from any devices
TimeCC V2 is not supported, so one cannot send it the timezone offset

Thanks for your input!

closed 2019-10-03 by AlCalzone

petergebruers, 2019-10-03:
Thank you for reporting back! That seems to explain the MH7's behaviour...

Fishwaldo, 2019-10-03:
Yeah. It was based on “reverse engineering” at the time and most devices seemed to default to local time rather than UTC. If devices are out there now defaulting to UTC we can add a config switch to handle them. (And update to specify the offsets etc).

mrAceT, 2020-05-25:
@AlCalzone my apologies for going a bit off topic.
I have bought the MH7H-WH-WT15 an I discovered a "small black joystick" on the bottom (to the left) of the device, do you have any idea where it is for?

AlCalzone, 2020-05-25:
Thats the internal temperature sensor :)

mrAceT, 2020-05-25:
Oh! Wierd wiggly thing, thought it was some secret joystick / control thingy..