OpenZWave/open-zwave #1961: Outdated info in readme.md and on website?

electrofloat, 2019-10-03:
in readme.md the "Contributing to OZW" section 4. says:
"OZW is developed by reverse engineering and consulting publically available information."
in the "Final Words" section it says:
"As of Sept, 2016, Sigma has released a large portion (not all) of the Z-Wave Protocol Specifications into the public domain. "

Also the website says (http://openzwave.com/dev/):
"There is one part of the protocol that has not been released into the Public Domain, namely, the "SerialAPI" protocol that specifies how to talk to a USB Stick via a Serial COM Port. This portion of OpenZWave will continue to be reverse engineered if needed in the future. "

But I found a SerialAPI docs here:
https://www.silabs.com/documents/login/user-guides/INS12350-Serial-API-Host-Appl.-Prg.-Guide.pdf

So my question is: Are these info still relevant, or outdated?

Thanks.

petergebruers, 2019-10-03:
AFAIK indeed the protocol was reverse engineered (that happened before I joined the OZW project), and you are correct, INS12350 describes the protocol. However, Silabs has deemed the future is Z/IP and so that document has not been updated for the "Series 700" chips. I have read that document several times in the past and I think nothing is needed right know. There is a slight issue with setting controller timings but I don't think it matters much.

Does this answer your question?

electrofloat, 2019-10-03:
I see, it does. I was just reading through the readme.md and website and noticed these discrepancies, and thought asking about it.

About the "Series 700". I know that aeotec's next Z-Stick will use Z/IP, but I didn't know that this could also be used with the SerialAPI. 

Maybe it is worth asking silabs on an update of that document?

petergebruers, 2019-10-03:
> About the "Series 700". I know that aeotec's next Z-Stick will use Z/IP, but I didn't know that this could also be used with the SerialAPI.

Well, I did not exactly say the Serial API was "gone". The UZB-7 stick manufactured by Silabs (700 series) is clearly running gateway firmware and implements the serial API. I know that because I use this stick for testing (a fork of OZW, it is not merged with upstream)... But Silabs mentions, on every document, the preferred way to communicate is Z/IP.

Fishwaldo, 2019-10-03:
The Serial API will remain in one form or another as long as the z-wave chips are based on serial protocol underneath, and that’s not going to change much given USB etc interfaces remain. 

The Z/IP is a middleware that Silabs are putting between the USB port and application and it still talks SerialAPI to the sticks, encapsulates the packets to/from the nodes in IPv6. Applications still need a “framework” to talk Zwave over IPv6 and technically OZW could be updated to replace the serial port driver with a network driver, but I personally feel that’s a unnecessary additional layer that complicates things without much benefit. We can probably continue to reverse engineer the SerialAPI if needed. (Also add on that the Z/IP gateway software is still proprietary and not OSS) 

The SerialAPI has never been “released” as part of the public specs of Z-Wave (at least not officially). It just happens to be that the document can be downloaded by registering now. Much the same as the $2500 SDK development release can be downloaded, but your not necessarily entitled to develop a product with it. 

Regarding the Readme, agree that it could use some updating.

petergebruers, 2019-10-03:
I'll update the Readme and the DoxygenMain.h

electrofloat, 2019-10-07:
@petergebruers Thanks!