OpenZWave/open-zwave #1966: SWITCH_BINARY and THERMOSTAT_MODE aren't supported of thermostat Horstmann HRT4-ZW

sucesor, 2019-10-06:
I have the thermostat Horstmann HRT4-ZW, Horstmann ASR-ZW Boiler Relay and AEON Labs ZW090 Z-Stick Gen5 EU controller.
I use the ioBrocker smart home system.
When the thermostat and the boiler relay changed their own state to heating or vice versa, there are no expected changes in the system. I mean the THERMOSTAT_SETPOINT was changed to the desired state however SWITCH_BINARY and THERMOSTAT_MODE of both devices weren't changed.

The log from OZWCP:

2019-10-06 11:32:37.762 Info, Notification: Notification home fd1ba840 node 2 Sleep
2019-10-06 11:33:36.052 Detail, Node002,   Received: 0x01, 0x09, 0x00, 0x04, 0x08, 0x02, 0x03, 0x40, 0x01, 0x01, 0xbb
2019-10-06 11:33:36.053 Detail, 
2019-10-06 11:33:36.053 Warning, Node002, CommandClass COMMAND_CLASS_THERMOSTAT_MODE HandlerMsg Returned False
2019-10-06 11:33:36.066 Detail, Node002,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x02, 0x03, 0x40, 0x01, 0x01, 0xb3
2019-10-06 11:33:36.066 Detail, 
2019-10-06 11:33:36.067 Warning, Node002, CommandClass COMMAND_CLASS_THERMOSTAT_MODE HandlerMsg Returned False
2019-10-06 11:33:36.180 Detail, Node002,   Received: 0x01, 0x09, 0x00, 0x04, 0x08, 0x02, 0x03, 0x25, 0x01, 0xff, 0x20
2019-10-06 11:33:36.181 Detail, 
2019-10-06 11:33:36.181 Warning, Node002, Routing HandleIncomingMsg to HandleMsg - Please Report: COMMAND_CLASS_SWITCH_BINARY 
2019-10-06 11:33:36.181 Warning, Node002, CommandClass COMMAND_CLASS_SWITCH_BINARY HandleIncomingMsg Returned False
2019-10-06 11:33:36.194 Detail, Node002,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x02, 0x03, 0x25, 0x01, 0xff, 0x28
2019-10-06 11:33:36.194 Detail, 
2019-10-06 11:33:36.194 Warning, Node002, Routing HandleIncomingMsg to HandleMsg - Please Report: COMMAND_CLASS_SWITCH_BINARY 
2019-10-06 11:33:36.194 Warning, Node002, CommandClass COMMAND_CLASS_SWITCH_BINARY HandleIncomingMsg Returned False
2019-10-06 11:33:36.278 Detail, Node002,   Received: 0x01, 0x0c, 0x00, 0x04, 0x08, 0x02, 0x06, 0x43, 0x03, 0x01, 0x22, 0x00, 0xfa, 0x62
2019-10-06 11:33:36.278 Detail, 
2019-10-06 11:33:36.279 Detail, Node002, Refreshed Value: old value=19.0, new value=25.0, type=decimal
2019-10-06 11:33:36.279 Detail, Node002, Changes to this value are not verified
2019-10-06 11:33:36.279 Info, Node002, Received thermostat setpoint report: Setpoint Heating 1 = 25.0C
2019-10-06 11:33:36.279 Detail, Node002, Notification: ValueChanged
2019-10-06 11:33:36.279 Info, Notification: Value Changed Home 0xfd1ba840 Node 2 Genre user Class THERMOSTAT SETPOINT Instance 1 Index 1 Type decimal IsSet True ValueAsString 25.0
2019-10-06 11:33:36.291 Detail, Node002,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x02, 0x06, 0x43, 0x03, 0x01, 0x22, 0x00, 0xfa, 0x6a
2019-10-06 11:33:36.292 Detail, 
2019-10-06 11:33:36.292 Detail, Node002, Refreshed Value: old value=25.0, new value=25.0, type=decimal
2019-10-06 11:33:36.292 Detail, Node002, Changes to this value are not verified
2019-10-06 11:33:36.292 Info, Node002, Received thermostat setpoint report: Setpoint Heating 1 = 25.0C
2019-10-06 11:33:36.292 Detail, Node002, Notification: ValueChanged
2019-10-06 11:33:36.293 Info, Notification: Value Changed Home 0xfd1ba840 Node 2 Genre user Class THERMOSTAT SETPOINT Instance 1 Index 1 Type decimal IsSet True ValueAsString 25.0

Full log file attached to the issue
[ozwcp.log.log](https://github.com/OpenZWave/open-zwave/files/3694135/ozwcp.log.log)

Fishwaldo, 2019-10-07:
For some reason the device is sending "SET" messages instead of "REPORT" messages. As OZW is not a "Node" it doesn't respond/support SET messages. Check to see if there is anything in the config for this device to change its behavior otherwise you need to report this to the manufacturer

>2019-10-06 11:14:15.759 Detail, Node002,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x02, 0x03, 0x40, 0x01, 0x01, 0xb3
2019-10-06 11:14:15.759 Detail, 
2019-10-06 11:14:15.760 Warning, Node002, CommandClass COMMAND_CLASS_THERMOSTAT_MODE HandlerMsg Returned False
2019-10-06 11:14:15.873 Detail, Node002,   Received: 0x01, 0x09, 0x00, 0x04, 0x08, 0x02, 0x03, 0x25, 0x01, 0xff, 0x20
2019-10-06 11:14:15.874 Detail, 
2019-10-06 11:14:15.874 Warning, Node002, Routing HandleIncomingMsg to HandleMsg - Please Report: COMMAND_CLASS_SWITCH_BINARY 
2019-10-06 11:14:15.874 Warning, Node002, CommandClass COMMAND_CLASS_SWITCH_BINARY HandleIncomingMsg Returned False

closed 2019-10-07 by Fishwaldo

sucesor, 2019-10-07:
I used the OpenHab smart home system and the thermostat worked as expected.
However, OpenHab doesn't use openzwave project https://github.com/openhab/org.openhab.binding.zwave/blob/master/doc/horstmann/srt323_0_0.mPerhaps this information will be helpful https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/989#

Fishwaldo, 2019-10-08:
OpenZWave != OpenHab

a SET message is used for "Controlling" another device. OZW Is not another device but a controller. When updating a controller/gateway, a device should use a REPORT message. Unless this device is used for controlling another thermostat, its not operating according to the specifications.