OpenZWave/open-zwave #1972: Qubino ZMNHADX needs node refresh after power failure

GravityRZ, 2019-10-10:
running domoticz V4.11354

The qubino node had problems with lost grouping after power failure. this was solved in #1895
i however discovered that when this unit is disconnected from power and the power comes back the temp sensor still does not work.
it needs a node refresh to get the temperature reading working again
is there a way to solve this that this unit registers temperature again automatically after a power loss/failure?

petergebruers, 2019-10-10:
Please continue the discussion in #1895 because it is still the same issue but another manifestation.

petergebruers, 2019-10-10:
I've reopened that issue for you and Github references this issue so no need to repeat what you have said here.

closed 2019-10-10 by petergebruers