OpenZWave/open-zwave #1977: thermofloor/heatit033.xml missing?

kjetilpp, 2019-10-15:
i can't find the config/thermofloor/heatit033.xml - Is it removed by an error?

petergebruers, 2019-10-15:
Hi! It is not in the 1.6 aka master branch as you say, but it is still in the 1.4 branch. What software are you using, what version of OZW does it use? You cannot mix config files from different versions.

kpine, 2019-11-12:
The `heatit033.xml` file was removed and replaced with `heatit056.xml` (TRM2fx). You can see the pull request [here](https://github.com/OpenZWave/open-zwave/pull/1714). According to that (and other places) the TRM2 was replaced by the TRM2fx. You can supposedly get it replaced by Heatit. The devices share the same product IDs, so there's not much that can be done.

closed 2019-11-25 by Fishwaldo