OpenZWave/open-zwave #1978: Feature request: method to get "End Point" from Value

petergebruers, 2019-10-15:
OpenZWave maps Instances (an index assigned by OpenZWave) to End Points (what is in the data packet) internally, hiding the details from the user. Config files can alter this map.

I can think of 2 reasons wanting to know the End Point:
- For diagnostic purposes
- To identify what to use in MultiChannel Association. This CC expects "End Points".

I’ll make a proof of concept add that to ozwcp and make a proposal
Something like this might be a start:

`uint8 Manager::GetValueEndPoint(ValueID const &_id)`

Or ... GetEndPointFromValue, or GetEndPointOfValue?

Shoot!

kdschlosser, 2019-10-22:
This would be a most helpful thing to have.

petergebruers, 2019-11-27:
PR #2013