OpenZWave/open-zwave #1980: Ability to disable confirmation get for devices with reliable automatic state reporting

matejdro, 2019-10-17:
Check out the following log. It contains simple action of me toggling (dimming) an Aeotec LED Bulb 6 Multi-Color (E27).

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3737382/OZW_Log.txt)

From what I understand, this happened:

* Single multilevel set is sent to the bulb
* Because OZW apparently assumes bulb does not properly report its own state, it also sends another get.
* When bulb starts actually dimming, it sends automatic value update
* On first value update, OZW sends THREE ADDITIONAL GET REQUESTS. It looks like device cannot handle those at once, so OZW of course receives timeout, which starts cascading and bogging down the network with retries

Could an option be added in the future to disable confirmation gets for specific devices (that behave well)?

This is from Home Assistant, which is supposedly still based on OZW 1.4 (my apologies if I should not report HA OZW issues here).

matejdro, 2019-10-18:
This might be related: https://github.com/OpenZWave/python-openzwave/blob/8be4c070294348f3fc268bc1d7ad2c535f352f5a/src-api/openzwave/command.py#L611

matejdro, 2019-10-18:
Not sure why, but setting the color seems to be even worse, sending 5 get requests for every set request:

[OZW_Log_Colors.txt](https://github.com/OpenZWave/open-zwave/files/3742254/OZW_Log_Colors.txt)

Fishwaldo, 2019-10-18:
I have not had a chance to look at the issue yet, but multiple requests are required to get each Color Channel (Eg, Red, Green, Blue, Warm White, Cool White etc). Hence the multiple requests. So this is normal.

matejdro, 2019-10-18:
Yes, that is true. But my point was that many devices (including this bulb) self-report their state on change. So there is no need to get all the data after setting it.

PR #2008 (2019-11-25): Make set after get optional
Resolves #1980 

Patch adds additional parameter to every Z-wave value in config xml: `refresh_after_set`. By default it is turned on (no changed behavior by default), but user can turn it off manually for devices that do not need constant verification.

Smart lighting in my Z-wave network consists solely of recent Aeotec and Fibaro devices that all send SwitchMultiLevel report on their own whenever you update them. By disabling redundant get commands, my network's responsiveness has increased significantly, especially when changing multiple lights at once or when c [... cut]
closed 2019-11-25 by Fishwaldo