OpenZWave/open-zwave #1983: SIGSEGV with Aeotec MultiSensor 6

boolangery, 2019-10-18:
Hi,
When trying to include the MultiSensor 6 in my network, my app just got a SIGSEGV sometimes:

```
2019/10/17 11:36:14 received notification: <ValueRemoved, HomeID: 0xc4398172, NodeID: 8, ValueID: { (Short):  [0 -> 0], genre: System}>
2019/10/17 11:36:14 received notification: <ValueRemoved, HomeID: 0xc4398172, NodeID: 8, ValueID: { (Button):  [0 -> 0], genre: System}>
2019/10/17 11:36:14 received notification: <ValueRemoved, HomeID: 0xc4398172, NodeID: 8, ValueID: { (Button):  [0 -> 0], genre: System}>
2019/10/17 11:36:14 received notification: <ValueRemoved, HomeID: 0xc4398172, NodeID: 8, ValueID: { (List):  [0 -> 0], genre: System}>
2019/10/17 11:36:14 received notification: <ValueRemoved, HomeID: 0xc4398172, NodeID: 8, ValueID: { (Short):  [0 -> 0], genre: System}>
2019/10/17 11:36:14 received notification: <NodeRemoved, HomeID: 0xc4398172, NodeID: 8>
2019/10/17 11:36:14 received notification: <ControllerCommand, HomeID: 0xc4398172, NodeID: 8, Event: 7, Notification: MsgComplete>

Thread 11 "OZW-driver" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd37fe700 (LWP 17773)]
tcache_get (tc_idx=1) at malloc.c:2943
2943	malloc.c: No such file or directory.
(gdb) bt
#0  tcache_get (tc_idx=1) at malloc.c:2943
OpenZWave/open-zwave-control-panel#1  __GI___libc_malloc (bytes=32) at malloc.c:3050
OpenZWave/open-zwave-control-panel#2  0x00007ffff7503258 in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
OpenZWave/open-zwave-control-panel#3  0x00007ffff7b0c64d in OpenZWave::Internal::Platform::WaitImpl::AddWatcher(void (*)(void*), void*) ()
   from /usr/local/lib64/libopenzwave.so.1.6
OpenZWave/open-zwave-control-panel#4  0x00007ffff7b0179a in OpenZWave::Internal::Platform::Wait::Multiple(OpenZWave::Internal::Platform::Wait**, unsigned int, int) ()
   from /usr/local/lib64/libopenzwave.so.1.6
OpenZWave/open-zwave-control-panel#5  0x00007ffff7b6f3df in OpenZWave::Driver::DriverThreadProc(OpenZWave::Internal::Platform::Event*) ()
   from /usr/local/lib64/libopenzwave.so.1.6
OpenZWave/open-zwave-control-panel#6  0x00007ffff7b0cb53 in OpenZWave::Internal::Platform::ThreadImpl::Run() () from /usr/local/lib64/libopenzwave.so.1.6
OpenZWave/open-zwave-control-panel#7  0x00007ffff7b0cb69 in OpenZWave::Internal::Platform::ThreadImpl::ThreadProc(void*) () from /usr/local/lib64/libopenzwave.so.1.6
OpenZWave/open-zwave-control-panel#8  0x00007ffff78006db in start_thread (arg=0x7fffd37fe700) at pthread_create.c:463
OpenZWave/open-zwave-control-panel#9  0x00007ffff6f8888f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

My app is basically the MinOzw example with some modifications.

petergebruers, 2019-10-18:
Voila

boolangery, 2019-10-30:
After deleting cache files (ozwcache_*.xml and others), I could not reproduce the bug. So maybe just close this one for now

petergebruers, 2019-10-30:
Okay. Let's investigate if it happens again.

closed 2019-10-30 by petergebruers