OpenZWave/open-zwave #1987: Qubino ZMNHDD1

chrisns, 2019-10-21:
I've got a bunch of ZMNHDD1 dimmers on my network, recently (past week-ish) while foolishly keeping current on the open-zwave:master branch I've lost the ability to read the current Level, and it's apparently also gained binary switch command classes, so I've gained a `user>Switch` and the `user>Level` always reads as `0` though I can change it, it just doesn't update what the device reports.
The `Switch` however does reflect truth.

I can't see anywhere obvious (to me) in the git history that would explain this
I've got other brand dimmers in my network that work fine
[log](https://github.com/OpenZWave/open-zwave/files/3752804/log.txt)+[cache](https://github.com/OpenZWave/open-zwave/files/3752803/cache.txt) see node id 39 or 56 for an example.

If no one has any immediate suggestions to what's gone awry I can try git-bisect till I find the cause. I'm only hesitating on that route due to the amount of time it takes me to recompile+redeploy each iteration, then delete and rebuild cache etc, I'd probably be better off buying a seperate controller and just including a single dimmer, but still painful
Thanks in advance

chrisns, 2019-10-21:
update: powercycled the dimmers, node 56 now behaves as I'd expect it to, node39 is still misbehaving and misreporting its Level as 0

chrisns, 2019-10-30:
will figure out remove and reinclude the device, if it doesn't work, i'll swap out the device, but its probably not a problem with ozw.
will add notes here if i end up doing anything in ozw config to fix it

closed 2019-10-30 by chrisns

petergebruers, 2019-10-30:
Thank you for reporting back!