OpenZWave/open-zwave #1992: Keyless Connected Smart Lock errors on Domoticz beta

petergebruers, 2019-11-03:
I see this in Domoticz with OpenZWave: Version: 1.6-809-g97ebe02e-dirty - I have an Assa Abloy Yale Keyless Connected Smart Lock (YSL), ID 0x0000, Type 0x0007:

Status: OpenZWave: Unhandled class: 0x4C (DOOR LOCK LOGGING), NodeID: 33 (0x21), Index: 1, Instance: 1
OpenZWave: Value_Added: Node: 33 (0x21), CommandClass: DOOR LOCK LOGGING, Label: Log Record, Instance: 1, Index: 2
Status: OpenZWave: Unhandled class: 0x4C (DOOR LOCK LOGGING), NodeID: 33 (0x21), Index: 2, Instance: 1
OpenZWave: Value_Added: Node: 33 (0x21), CommandClass: DOOR LOCK, Label: Locked, Instance: 1, Index: 0
OpenZWave: Value_Added: Node: 33 (0x21), CommandClass: DOOR LOCK, Label: Locked (Advanced), Instance: 1, Index: 1
Error: OpenZWave: Unhandled value type: 4, OpenZWave.cpp:2176

At some stage this was working fine but I upgraded / rebuilt my Domoticz setup and moved to OpenZWave 1.6 which looks to have caused the issue.

In addition the device has no configuration options now, only "3001 Protection" - it used to have a lot more options which I assume is related to the errors reported?

_Originally posted by @russdan in https://github.com/OpenZWave/open-zwave/issues/1886#issuecomment-549113089_

petergebruers, 2019-11-05:
Let's double check manufacturer_specific.xml

`<Product config="assa_abloy/KeyfreeConnected.xml" id="0000" name="Yale Keyless Connected Smart Lock (YSL)" type="0007"/>`

That's a match.

Then the interesting part of the config file:

```
  <CommandClass id="113">
    <!-- These Door Locks don't send a DoorLockReport when the
		Lock Status is Changed, but instead send a Alarm Message -
		So we trigger a Refresh of the DoorLock Command Class when
		we recieve a Alarm Message Instead -->
    <TriggerRefreshValue Genre="user" Index="0" Instance="1">
      <RefreshClassValue CommandClass="98" Index="1" Instance="1" RequestFlags="0"/>
    </TriggerRefreshValue>
  </CommandClass>
```

This matches your observation `In addition the device has no configuration options now, only "3001 Protection"`

I'll have to check what that config file looked like in OZW 1.4...

Can you please check in Domoticz, on the hardware page of OpenZWave, there is an option to start the (old) ozwcp (open-zwave-control-panel). Can you please try that one to find out if you can lock/unlock? This helps determine if it is a OZW or Domoticz issue.

russdan, 2019-11-05:
Hi, here's what I see from the ozwp pages:

https://1drv.ms/w/s!Apng_hydWq21ljdmhYWn62D0wG2N

russdan, 2019-11-05:
What's weird is checking v1.4 code there isn't an manufacturer_specific entry for the YSL smartlock, and same in v1.5 code - I have a local copy of a manufacturer_specific.xml which does have the entry in it, dated 12/11/2017 (12th November 2017) as I recall looking at updating it to better handle some events - I'm not sure what release that would correspond with, but it was working better (~90%) with this release.  I also have the OpenZWave.cpp file which corresponds to that xml file if that helps? I think it may include the changes I was trying at the time which were around the Access Control events from memory.  Let me know if this would help?

petergebruers, 2019-11-05:
Cannot open that picture on my phone, cannot zoom in. The main question is: can you control the lock?

I have the repo's of Domoticz and OpenZWave on my laptop so I can look at any version of any file. Looking at OpenZWave.cpp from the 1.4 era won't help, Domoticz used many "shortcuts" so to speak that won't work with OZW 1.6.

I don't own a lock, so it is hard to debug... But I'll try.

I'll post some instructions to get debug logging.

russdan, 2019-11-05:
Hi, there is no control of the lock using OZW or Domoticz.

My main reference for the November 2017 files was that it worked then, so I was hoping that might give some clues as to what regressed or was dropped in later releases.  I though domoticz was just a passthru for OZW, but obviously its more complicated than that!

Thanks for your help - appreciate that is hard to do without the device :)

petergebruers, 2019-11-06:
Sorry for the delay, on my laptop I saw your screenshots and indeed you have that pulldown menu after the "Locked" label. It should expose only an "On" and "Off" button...

That "Value" as we call it should map to a device in Domoticz as well. Maybe you have multiple devices 

Now I would like to respond to something you have mentioned but is a "red herring"

You also have a pull down menu called "Locked (Advance)" which has more options.

That property is called "Lock_Mode" , nit "Lock" and that one indeed refers to the issue you posted in previously. As I posted there... Domoticz does not use it and this explains why Domoticz logged this:

`Error: OpenZWave: Unhandled value type: 4, OpenZWave.cpp:2176`

Domoticz explicitly does not handle that "pulldown menu" aka "Lock_Mode"

```
else if (vType == OpenZWave::ValueID::ValueType_List)
			{
				return; //Not in use at the moment
```

So you might think... There's the bug! But unfortunately... No. Because even when Domoticz would add code to handle that we would probably hit `Door Lock: handling of ValueID_Index_DoorLock::Lock_Mode probably incorrect #1886`

So that is a "red herring"... The lines above that in Domoticz do handle that "On"/"Off toggle. That Value is called "ValueID_Index_DoorLock::Lock"

Can you double check all the devices created in Domoticz? One of them should be a on/off switch corresponding to that... If you have one but it does not work, please post on Domoticz forum and put a link to this issue. @gizmocuz knows who I am.

Unfortunately, for personal reasons, I won't be able to work on this until the second half of November.

> I though domoticz was just a passthru for OZW, but obviously its more complicated than that!

It mainly has to adapt what we call "Values" in OZW to "Devices" in Domoticz and back, it is a bit more complicated but you were certainly on the right track when you mentioned "OpenZWave.cpp" in a previous post... That is where the magic happens.

russdan, 2019-11-06:
Hi, thanks again for your response.  I do have a Domoticz Door Lock device, and it works as expected in unlocking the door via Domoticz, which is something I hadn't previously spotted (I had considered this as a read-only device to see when the door had been unlocked).

It looks as if those unhandled errors in the log are effectively working as designed if I understand you correctly?  I'm happy the Issue is closed on that basis and thank you for your support :)

petergebruers, 2019-11-08:
> It looks as if those unhandled errors in the log are effectively working as designed if I understand you correctly?

At the moment, yes indeed...

closed 2019-11-08 by petergebruers