OpenZWave/open-zwave #1995: qubino wrong config parameter 68 in ZMNHDDX.xml

jtonk, 2019-11-10:
I believer there is a mistake in the qubino flush dimmer ZMNHDDx.xml

config parameter hold a list while the qubino is expecting a byte. I've modified the parameter below, however I still can't get parameter 68 to work :(

any suggestion?

`<Value type="byte" genre="config" instance="1" index="68" label="Dimming duration" min="0" max="127" value="1">
  <Help>This parameter is used with association group 3. The Duration field MUST specify the time that the transition should take from the current value to the new target value. A supporting device SHOULD respect the specified Duration value. Default value 0.</Help>
</Value>`

petergebruers, 2019-11-11:
Is your software based on OpenZWave 1.4 or 1.6? It says so at the start of OZW_log.txt. If it is 1.4 we do not support that anymore. If it is 1.6 please investigate and tell us what works for you. I do not own this device, I cannot test and debug. But I can help you add your solution to the repository so everyone benefits from your investigation.

jtonk, 2019-11-11:
I’m using home assistant, so it’s still 1.4. So I guess I have to wait until HA updates.
But regardsless of the 1.6 or 1.4 branch, the parameter 68 is wrong, the qubino manual expects a Value between 0 and 127. The parameter 68 entry in ZMNHDDx.xml is copied from parameter 67 which is presented as a list (0 and 1)

petergebruers, 2019-11-15:
I'll have a look at the 1.6 version. Your explanation matches with what's recorded in the OpnHAB database https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/210

Then you can apply mods to the HASS fork https://github.com/home-assistant/open-zwave

petergebruers, 2019-11-17:
Hi! I've opened a PR for OZW 1.6 and if the build system is happy I'll merge it.

I don't own the device but I've partially simulated it and now in ozwcp I get an entry box instead of the pull-down we had with the previous version. Looks good to me.

You're on 1.4 so you'll have to fix your config file manually.

`<Value genre="config" index="68" instance="1" label="Dimming duration" max="127" min="0" size="1" type="byte" value="0">`

You've told me you changed type to byte, that was the right thing to do. If that did not work maybe forgot max/man? Also it is possible your OZW does not read the config file where you've put it, that depends on how HASS was installed.And I think you need to do "Refresh Node" (blue button below node on "Z-Wave Network Management" configuration page.

closed 2019-12-01 by no-response[bot]