OpenZWave/open-zwave #1996: Potential leak

jveber, 2019-11-11:
Hello I'm using OpenZW library in my project and after deploying address/leak sanitizer I've detected following leak:

```
=================================================================
==28869==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 64 byte(s) in 2 object(s) allocated from:
    #0 0x76a78623 in operator new(unsigned int) /media/tmp_portage/portage/sys-devel/gcc-8.3.0-r1/work/gcc-8.3.0/libsanitizer/asan/asan_new_delete.cc:90
    #1 0x93f24f in OpenZWave::Node::ReadDeviceClasses() /home/jaromir/mq_system/open-zwave/cpp/src/Node.cpp:3285
    #2 0x93fd1f in OpenZWave::Node::SetDeviceClasses(unsigned char, unsigned char, unsigned char) /home/jaromir/mq_system/open-zwave/cpp/src/Node.cpp:2925
    #3 0x94010b in OpenZWave::Node::UpdateProtocolInfo(unsigned char const*) /home/jaromir/mq_system/open-zwave/cpp/src/Node.cpp:1460
    #4 0x9ba08b in OpenZWave::Driver::ProcessMsg(unsigned char*, unsigned char) /home/jaromir/mq_system/open-zwave/cpp/src/Driver.cpp:1974
    #5 0x9baca3 in OpenZWave::Driver::ReadMsg() /home/jaromir/mq_system/open-zwave/cpp/src/Driver.cpp:1691
    #6 0x9bc3bb in OpenZWave::Driver::DriverThreadProc(OpenZWave::Internal::Platform::Event*) /home/jaromir/mq_system/open-zwave/cpp/src/Driver.cpp:438
    #7 0x9fc8e7 in OpenZWave::Internal::Platform::ThreadImpl::Run() /home/jaromir/mq_system/open-zwave/cpp/src/platform/unix/ThreadImpl.cpp:166
    #8 0x9fc90b in OpenZWave::Internal::Platform::ThreadImpl::ThreadProc(void*) /home/jaromir/mq_system/open-zwave/cpp/src/platform/unix/ThreadImpl.cpp:153
    #9 0x7680faff in start_thread (/lib/libpthread.so.0+0x6aff)
    #10 0x76082a67  (/lib/libc.so.6+0xe7a67)

Indirect leak of 62 byte(s) in 2 object(s) allocated from:
    #0 0x76a78623 in operator new(unsigned int) /media/tmp_portage/portage/sys-devel/gcc-8.3.0-r1/work/gcc-8.3.0/libsanitizer/asan/asan_new_delete.cc:90
    #1 0x76765573 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_mutate(unsigned int, unsigned int, char const*, unsigned int) /media/tmp_portage/portage/sys-devel/gcc-8.3.0-r1/work/build/armv7a-unknown-linux-gnueabihf/libstdc++-v3/include/bits/basic_string.tcc:317
    #2 0x93f25b in OpenZWave::Node::ReadDeviceClasses() /home/jaromir/mq_system/open-zwave/cpp/src/Node.cpp:3285
    #3 0x93fd1f in OpenZWave::Node::SetDeviceClasses(unsigned char, unsigned char, unsigned char) /home/jaromir/mq_system/open-zwave/cpp/src/Node.cpp:2925
    #4 0x94010b in OpenZWave::Node::UpdateProtocolInfo(unsigned char const*) /home/jaromir/mq_system/open-zwave/cpp/src/Node.cpp:1460
    #5 0x9ba08b in OpenZWave::Driver::ProcessMsg(unsigned char*, unsigned char) /home/jaromir/mq_system/open-zwave/cpp/src/Driver.cpp:1974
    #6 0x9baca3 in OpenZWave::Driver::ReadMsg() /home/jaromir/mq_system/open-zwave/cpp/src/Driver.cpp:1691
`
    #7 0x9bc3bb in OpenZWave::Driver::DriverThreadProc(OpenZWave::Internal::Platform::Event*) /home/jaromir/mq_system/open-zwave/cpp/src/Driver.cpp:438
    #8 0x9fc8e7 in OpenZWave::Internal::Platform::ThreadImpl::Run() /home/jaromir/mq_system/open-zwave/cpp/src/platform/unix/ThreadImpl.cpp:166
    #9 0x9fc90b in OpenZWave::Internal::Platform::ThreadImpl::ThreadProc(void*) /home/jaromir/mq_system/open-zwave/cpp/src/platform/unix/ThreadImpl.cpp:153
    #10 0x7680faff in start_thread (/lib/libpthread.so.0+0x6aff)
    #11 0x76082a67  (/lib/libc.so.6+0xe7a67)

Indirect leak of 19 byte(s) in 2 object(s) allocated from:
    #0 0x76a787cf in operator new[](unsigned int) /media/tmp_portage/portage/sys-devel/gcc-8.3.0-r1/work/gcc-8.3.0/libsanitizer/asan/asan_new_delete.cc:93
    #1 0x938b7b in OpenZWave::Node::DeviceClass::DeviceClass(TiXmlElement const*) /home/jaromir/mq_system/open-zwave/cpp/src/Node.cpp:3385
    #2 0x93f25b in OpenZWave::Node::ReadDeviceClasses() /home/jaromir/mq_system/open-zwave/cpp/src/Node.cpp:3285
    #3 0x93fd1f in OpenZWave::Node::SetDeviceClasses(unsigned char, unsigned char, unsigned char) /home/jaromir/mq_system/open-zwave/cpp/src/Node.cpp:2925
    #4 0x94010b in OpenZWave::Node::UpdateProtocolInfo(unsigned char const*) /home/jaromir/mq_system/open-zwave/cpp/src/Node.cpp:1460
    #5 0x9ba08b in OpenZWave::Driver::ProcessMsg(unsigned char*, unsigned char) /home/jaromir/mq_system/open-zwave/cpp/src/Driver.cpp:1974
    #6 0x9baca3 in OpenZWave::Driver::ReadMsg() /home/jaromir/mq_system/open-zwave/cpp/src/Driver.cpp:1691
    #7 0x9bc3bb in OpenZWave::Driver::DriverThreadProc(OpenZWave::Internal::Platform::Event*) /home/jaromir/mq_system/open-zwave/cpp/src/Driver.cpp:438
    #8 0x9fc8e7 in OpenZWave::Internal::Platform::ThreadImpl::Run() /home/jaromir/mq_system/open-zwave/cpp/src/platform/unix/ThreadImpl.cpp:166
    #9 0x9fc90b in OpenZWave::Internal::Platform::ThreadImpl::ThreadProc(void*) /home/jaromir/mq_system/open-zwave/cpp/src/platform/unix/ThreadImpl.cpp:153
    #10 0x7680faff in start_thread (/lib/libpthread.so.0+0x6aff)
    #11 0x76082a67  (/lib/libc.so.6+0xe7a67)
```
I'was trying to find the leak reason but T was not able to because the code od Manager.cpp should cleanup all the objects allocated in ReadDeviceClasses.

Thus it may not be a leak at all just an error of Adress sanitizer but I guess it's not. I guess there may be an overwrite of value in s_deviceTypeClasses[key] causing the leak but this overwrite is definitely not in ReadDeviceClasses (I was checking that one).

Ozw log is following but that does not imply anything I guess (it's in reverse order):

```
[2019-11-11 09:40:39.239] [mq_zwave_daemon] [info] Terminating

[2019-11-11 09:40:34.441] [mq_zwave_daemon] [info] ZWLog(Always):0: ***************************************************************************

[2019-11-11 09:40:34.421] [mq_zwave_daemon] [info] ZWLog(Always):0: Messages dropped and not delivered: . . . . . . . . . . . 3

[2019-11-11 09:40:34.398] [mq_zwave_daemon] [info] ZWLog(Always):0: Messages retransmitted: . . . . . . . . . . . . . . . . . 0

[2019-11-11 09:40:34.377] [mq_zwave_daemon] [info] ZWLog(Always):0: Out of frame data flow errors: . . . . . . . . . . . . . 0

[2019-11-11 09:40:34.354] [mq_zwave_daemon] [info] ZWLog(Always):0: NAKs received from controller: . . . . . . . . . . . . . 0

[2019-11-11 09:40:34.332] [mq_zwave_daemon] [info] ZWLog(Always):0: CANs received from controller: . . . . . . . . . . . . . 0

[2019-11-11 09:40:34.310] [mq_zwave_daemon] [info] ZWLog(Always):0: Bad checksum errors: . . . . . . . . . . . . . . . . . . 0
[... 180 lines cut ...]

[2019-11-11 08:52:42.764] [mq_zwave_daemon] [info] ZWLog(Warning):0: Localization::ReadXMLVIDHelp: Error in /etc/mq_system/ozwcache_0xfa641bc0.xml at line 201 - No Help Entry for CommandClass 50, ValueID: 256 (-1): (null) (Lang: )

[2019-11-11 08:52:42.718] [mq_zwave_daemon] [info] ZWLog(Warning):0: Localization::ReadXMLVIDHelp: Error in /etc/mq_system/ozwcache_0xfa641bc0.xml at line 198 - No Help Entry for CommandClass 50, ValueID: 2 (-1): (null) (Lang: )

[2019-11-11 08:52:42.672] [mq_zwave_daemon] [info] ZWLog(Warning):0: Localization::ReadXMLVIDHelp: Error in /etc/mq_system/ozwcache_0xfa641bc0.xml at line 195 - No Help Entry for CommandClass 50, ValueID: 0 (-1): (null) (Lang: )

[2019-11-11 08:52:42.619] [mq_zwave_daemon] [info] ZWLog(Warning):0: Localization::ReadXMLVIDHelp: Error in /etc/mq_system/ozwcache_0xfa641bc0.xml at line 172 - No Help Entry for CommandClass 43, ValueID: 1 (-1): (null) (Lang: )

[2019-11-11 08:52:42.567] [mq_zwave_daemon] [info] ZWLog(Warning):0: Localization::ReadXMLVIDHelp: Error in /etc/mq_system/ozwcache_0xfa641bc0.xml at line 169 - No Help Entry for CommandClass 43, ValueID: 0 (-1): (null) (Lang: )

[2019-11-11 08:52:25.990] [mq_zwave_daemon] [info] ZWLog(Warning):0: Failed - Network Key Not Set

[2019-11-11 08:52:25.833] [mq_zwave_daemon] [info] ZWLog(Always):0: Using Language Localization

[2019-11-11 08:52:25.548] [mq_zwave_daemon] [info] ZWLog(Always):0: OpenZwave Version 1.6-949-g8a6972db-dirty Starting Up

[2019-11-11 08:52:25.471] [mq_zwave_daemon] [info] Entered main

[2019-11-11 08:52:25.338] [mq_zwave_daemon] [info] Demon initialization finished 
```

Ask anything you need to investigate the issue. 

I'm using the latest master 1.6 github version (8a6972db).

petergebruers, 2019-11-11:
Hi! I've wanted to do a similar thing myself (running OZW with sanitizers I mean). Thanks for reporting this.

To be honest... I bet there are more minor leaks. I bet as soon as you see the keyword "new" in any old source code, there is going to be. I am not a seasoned programmer but a 50 year old IT engineer so I see the results of leaks. As you point out it might not harm in real life. Uptime of products based on OZW does not seem to be an issue, we would get more reports here if it was.

I speak for myself but I think OZW contributors are aware of this and @Fishwaldo and I briefly discussed move to smart pointers but someone has to do that (newer parts already have that).

Not sure where to go from here though. Rewrite OZW in a memory safe language perhaps? :smile:

Fishwaldo, 2019-11-11:
There were duplicate entries in device_classes.xml, so the previous entry would get lost. Fixed that and put some tests in to warn if we have duplicate entries in the future

closed 2019-11-11 by Fishwaldo