OpenZWave/open-zwave #1998: POPP 012501 (>= V1.4) Strike Lock device does not work correct

jolichter, 2019-11-13:
Hi, Please could you update the POPP 012501 Strike Lock?
The Strike Lock version 1.4 or higher expects only the timeout over the command class (0x62) 98. The switch off time and status is sent via Z-Wave command.

I need a switch off time "0", that doesn't work with new version and I got this information from an Aeotec Field Application Engineer (Aeotec took over electronics maker POPP):

e.g. Z-Way command code with switch off time set "0":
GATEWAYIP:8083/ZWave.zway/Run/devices[DeviceID].instances[0].commandClasses[98].Set(0)

D.26 Command Class Door Lock (0x62/98)
Version 2, Controlled Allows to operate an electronic door lock. This Command Class is ALWAYS encapsulated in Security.

Door lock modes are the following:

- 0x00 Door Unsecured (Open)
- 0x01 Door Unsecured with timeout
- 0x10 Door Unsecured for inside Door Handles
- 0x11 Door Unsecured for inside Door Handles with timeout
- 0x20 Door Unsecured for outside Door Handles
- 0x21 Door Unsecured for outside Door Handles with timeout
- 0xFE Door/Lock Mode Unknown (bolt not fully retracted/engaged)
- 0xFF Door Secured (closed)

I hope that this information will be sufficient?

Many Thanks

Fishwaldo, 2019-11-14:
It should work. Please post log files of it not working so we can investigate

jolichter, 2019-11-14:
Thanks for the reply.
I'm new in terms of ZWave and hope I am not wrong. I use ioBroker 3.6.12 with the openzwave adapter 1.7.0. I will investigate the log files next weekend.

jolichter, 2019-11-14:
Hi, I checked it this evening. Please could you investigate it?
The only thing I changed at the default settings, is switch from bool to number.

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3848921/OZW_Log.txt)
[object_raw.txt](https://github.com/OpenZWave/open-zwave/files/3848923/object_raw.txt)
[image]

ScratchedFace, 2019-11-18:
Hi, have the same issue, logs:
```

2019-11-18 20:18:24.670 Info, Node041, Value::Set - COMMAND_CLASS_DOOR_LOCK - Locked - 0 - 1 - True
2019-11-18 20:18:24.671 Info, Node041, Value_Lock::Set - Requesting lock to be Locked
2019-11-18 20:18:24.671 Detail, Node041, Queuing (Send) DoorLockCmd_Set (Node=41): 0x01, 0x0a, 0x00, 0x13, 0x29, 0x03, 0x62, 0x01, 0xff, 0x25, 0x37, 0x42
2019-11-18 20:18:24.671 Detail, Node041, Queuing (Send) DoorLockCmd_Get (Node=41): 0x01, 0x09, 0x00, 0x13, 0x29, 0x02, 0x62, 0x02, 0x25, 0x38, 0xb3
2019-11-18 20:18:24.672 Detail,
2019-11-18 20:18:24.672 Info, Node041, Sending (Send) message (Callback ID=0x37, Expected Reply=0x04) - DoorLockCmd_Set (Node=41): 0x01, 0x0a, 0x00, 0x13, 0x29, 0x03, 0x62, 0x01, 0xff, 0x25, 0x37, 0x42
2019-11-18 20:18:24.681 Detail, Node041,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-11-18 20:18:24.681 Detail, Node041,   ZW_SEND_DATA delivered to Z-Wave stack
2019-11-18 20:18:25.935 Detail, Node041,   Received: 0x01, 0x18, 0x00, 0x13, 0x37, 0x00, 0x00, 0x7d, 0x00, 0xca, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x01, 0x03, 0x00, 0x00, 0x00, 0x00, 0x42, 0x04, 0x00, 0x00, 0x30
2019-11-18 20:18:25.935 Detail, Node041,   ZW_SEND_DATA Request with callback ID 0x37 received (expected 0x37)
2019-11-18 20:18:25.936 Info, Node041, Request RTT 1263 Average Request RTT 1271
2019-11-18 20:18:25.936 Detail,   Expected callbackId was received
2019-11-18 20:18:34.673 Error, Node041, ERROR: Dropping command, expected response not received after 1 attempt(s)
2019-11-18 20:18:34.673 Detail, Node041, Removing current message
2019-11-18 20:18:34.674 Detail, Node041, Notification: Notification - TimeOut
2019-11-18 20:18:34.704 Detail,
2019-11-18 20:18:34.704 Info, Node041, Sending (Send) message (Callback ID=0x38, Expected Reply=0x04) - DoorLockCmd_Get (Node=41): 0x01, 0x09, 0x00, 0x13, 0x29, 0x02, 0x62, 0x02, 0x25, 0x38, 0xb3
2019-11-18 20:18:34.713 Detail, Node041,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-11-18 20:18:34.714 Detail, Node041,   ZW_SEND_DATA delivered to Z-Wave stack
2019-11-18 20:18:37.006 Detail, Node041,   Received: 0x01, 0x18, 0x00, 0x13, 0x38, 0x00, 0x00, 0xe5, 0x00, 0xca, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x01, 0x03, 0x00, 0x00, 0x00, 0x00, 0x42, 0x04, 0x00, 0x00, 0xa7
2019-11-18 20:18:37.006 Detail, Node041,   ZW_SEND_DATA Request with callback ID 0x38 received (expected 0x38)
2019-11-18 20:18:37.007 Info, Node041, Request RTT 2302 Average Request RTT 1786
2019-11-18 20:18:37.007 Detail,   Expected callbackId was received
2019-11-18 20:18:44.704 Error, Node041, ERROR: Dropping command, expected response not received after 1 attempt(s)
2019-11-18 20:18:44.704 Detail, Node041, Removing current message
2019-11-18 20:18:44.705 Detail, Node041, Notification: Notification - TimeOut
```

Using the lib at Hass.io 101.3.

Fishwaldo, 2019-11-19:
@ScratchedFace you didn’t include the lock securely. It won’t respond to non-secure messages. So your issue is different.

Fishwaldo, 2019-11-25:
@jolichter 
>The only thing I changed at the default settings, is switch from bool to number.

Changed what?

Fishwaldo, 2019-11-25:
@jolichter 
>2019-11-14 23:04:08.229 Info, Node040, Sending (Send) message (Callback ID=0xaa, Expected Reply=0x04) - DoorLockCmd_Set (Node=40): 0x01, 0x0a, 0x00, 0x13, 0x28, 0x03, 0x62, 0x01, 0x00, 0x25, 0xaa, 0x21

There is nothing wrong with that command. Its requesting the Lock to be "unlocked". Does any command work? (locking or unlocking). 

It responds later that the door was unlocked. Did it actually work or was it always unlocked.

jolichter, 2019-11-25:
Yes, the basic function is running with "DOOR_LOCK.Locked_1" false or 0.
After the Strike Lock 'unlocked', it 'locked' again after 3 seconds and I can not change this time setting.

For example with "DOOR_LOCK.Locked_1" 5 or disable the time set with 0.

The old Strike Lock predecessor version <1.4 expects the timeout over the command "Automatically_switch_off_after".

That's a pity. I wanted to use this power-saving relay in the weekend-house for the lights on a 12 volt system.

Sorry for the effort, I send the Strike Lock back and use a rgbw-controller as an alternative.

closed 2019-11-25 by jolichter