OpenZWave/open-zwave #2002: Openzwave 1.6 - Latest stable version

robertsLando, 2019-11-20:
Where do I find latest stable 1.6 version? I'm installing Ozw inside a docker container and actually I'm doing it like: 

```Dockerfile
RUN git clone https://github.com/OpenZWave/open-zwave.git \
    && cd open-zwave && make && make install \
```
But master isn't always stable.

For 1.4 I'm doing:

```Dockerfile
RUN git clone -b 1.4 https://github.com/OpenZWave/open-zwave.git \
    && cd open-zwave && make && make install \
```

petergebruers, 2019-11-20:
That's correct. To be pedantic, if you do not start from a clean system maybe add "make clean" or "git clean -d -f -x" but be warned that last one will remove your edits. Try git clean -d -f -x -n before doing that.

What do you mean by "master is not always stable"? It should be. It is "in flux" because of bug fixes and device file updates, but not "unstable". The unstable branch is "Dev"

Fishwaldo, 2019-11-20:
Build tested releases are available on the OZW website. They are “release” ready. Id recommend you use them instead of checking out from Github as master branch might have some issues occasionally. Either that, or pick a commit on github and test/use that, and update occasionally. 

I don’t recommend using HEAD unless your willing to deal with random issues.

petergebruers, 2019-11-20:
> I don’t recommend using HEAD unless your willing to deal with random issues.

Thanks, I forgot to mention that, I looked at it from a tinkerer's perspective and not with the eyes of somone like @robertsLando who builds a system on top of that.

I know @gizmocuz (Domoticz) integrates certain commits manually, like you suggested

Others use the releas builds

http://old.openzwave.com/downloads/

Some have been confused by the "old" in that url but that's "for historical reasons" if I am correct...

Fishwaldo, 2019-11-25:
Closing this.

closed 2019-11-25 by Fishwaldo

robertsLando, 2019-11-25:
Sorry I missed to reply to this. I have done by downloading from http://old.openzwave.com/downloads/.

Thanks both for the clarification :)

petergebruers, 2019-11-25:
> Sorry I missed to reply to this. I have done by downloading from

Missed to reply, or did you miss email notifications (replies)? Today I noticed someone replied to another Github issue I was participating in, did not get a mail... Nothing in my spam folder. I kind of rely on my "inbox" for notifications. Maybe I did something wrong. But I do get updates on some topics/issues.