OpenZWave/open-zwave #2007: Fibaro FGS223 only first output is controllable

thucar, 2019-11-24:
When including Fibaro FGS223 Double Relay, it does get recognized as such, with all the correct parameters available and working. However, only one relay output is available. There is no way to control the second relay output.

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3883468/OZW_Log.txt)

petergebruers, 2019-11-24:
It looks like during the interview the device did not respond to a an important part of the interview.

```
2019-11-24 13:04:57.624 Info, Node030, Sending (Send) message (Callback ID=0x4b, Expected Reply=0x04) - MultiChannelAssociationCmd_Get (Node=30): 0x01, 0x0a, 0x00, 0x13, 0x1e, 0x03, 0x8e, 0x02, 0x01, 0x25, 0x4b, 0x18
2019-11-24 13:04:57.625 Info, Node030, Encrypted Flag is 0
2019-11-24 13:04:57.633 Detail, Node030,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-11-24 13:04:57.633 Detail, Node030,   ZW_SEND_DATA delivered to Z-Wave stack
2019-11-24 13:04:57.782 Detail, Node030,   Received: 0x01, 0x07, 0x00, 0x13, 0x4b, 0x00, 0x00, 0x10, 0xb0
2019-11-24 13:04:57.782 Detail, Node030,   ZW_SEND_DATA Request with callback ID 0x4b received (expected 0x4b)
2019-11-24 13:04:57.782 Info, Node030, Request RTT 157 Average Request RTT 157
2019-11-24 13:04:57.782 Detail, Node030,   Expected callbackId was received
2019-11-24 13:05:07.626 Error, Node030, ERROR: Dropping command, expected response not received after 1 attempt(s)
2019-11-24 13:05:07.626 Detail, Node030, Removing current message
2019-11-24 13:05:07.626 Detail, Node030, Notification: Notification - TimeOut
```

Your node might have network issues. Bring it closer, exclude and include again... See what happens.

thucar, 2019-11-24:
For the original inclusion I used my laptop to move from device to device and add them no more than 2 meters from the controller.

I have two more relays, FGS211 and FGS221 right next to this one. Both of those work perfectly.

The log actually shows a "Refresh Node Information" process. I have ran it multiple times by now and these errors about dropping commands always occur. Only for this one device and they seem to end in the exact same result - second output is missing.

If it were a network issue, I would expect to see variable results with different parts of the interview getting lost each time.

petergebruers, 2019-11-24:
I agree that it would be weird to get lost every time. But it is equally weird that it gives that timeout. And equally weird that I have not seen anyone post this issue before.

I don't own this device, I cannot test. Also I am not at home.

OZW should be able to detect that this device is "multi instance". If it doesn't then indeed you only get to control the first relay...

What gateway software do you use? What do you find about this issue on their forum?

thucar, 2019-11-25:
I'm using Zwave2MQTT for my gateway software. I'll try to do a re-inclusion of the relay later today to see if there is a difference between that and just refreshing node information.

Fishwaldo, 2019-11-25:
Its Failing to respond to the MultiChannel_EndpointGet which would find out what other channels it supports.

>2019-11-24 13:04:45.730 Info, Node030, Sending (Query) message (Callback ID=0x42, Expected Reply=0x04) - MultiChannelCmd_EndPointGet (Node=30): 0x01, 0x09, 0x00, 0x13, 0x1e, 0x02, 0x60, 0x07, 0x25, 0x42, 0xf9
2019-11-24 13:04:45.730 Info, Node030, Encrypted Flag is 0
2019-11-24 13:04:45.737 Detail, Node030,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-11-24 13:04:45.738 Detail, Node030,   ZW_SEND_DATA delivered to Z-Wave stack
2019-11-24 13:04:45.821 Detail, Node030,   Received: 0x01, 0x07, 0x00, 0x13, 0x42, 0x00, 0x00, 0x09, 0xa0
2019-11-24 13:04:45.822 Detail, Node030,   ZW_SEND_DATA Request with callback ID 0x42 received (expected 0x42)
2019-11-24 13:04:45.822 Info, Node030, Request RTT 92 Average Request RTT 139
2019-11-24 13:04:45.822 Detail, Node030,   Expected callbackId was received
2019-11-24 13:04:55.729 Error, Node030, ERROR: Dropping command, expected response not received after 1 attempt(s)

Can you try shutting down, removing any ozwcache_*.xml (1.6) or zwcfg_*.xml (1.4)  and then restarting and post the log file?

Based on what I see though, it appears to be a issue with the device and not OZW.

T-Grave, 2019-11-26:
Just reporting in since I have the same issue with the same device. 
Re-including fixed it for me the first time. Recently restarted my network and it only reported the first switch again, so I did a Node Refresh command and it showed everything correctly again after a few minutes.

closed 2019-12-09 by no-response[bot]