OpenZWave/open-zwave #2012: Confusion about m_deviceType aka devicetype versus UserIcon

petergebruers, 2019-11-25:
Justin and I have discussed this a while ago, this is a quick summary...

We are both convinced at some point the Z-Wave spec talked about a "devicetype". We vaguely remember a document, possibly an excel file, describing device types and properties.

But when you look at recent public documents, this is "DKD13867 Z-Wave Specification release 2019D" the property we seem to remember (and what it is called in OpenZWave) is now "User Icon" instead of "device type".

That property starts to live in OpenZWave as ZWavePlusInfo, doing a Get of The Z-Wave Plus info.

SDS13782-12 Z-Wave Management Command Class Specification 2019-10-10
4.45 Z-Wave Plus Info Command Class, version 2
4.45.4 Z-Wave Plus Info Report Command

It defines the payload of ZWAVEPLUS_INFO_REPORT as:

Z-Wave Plus Version
Role Type
Node Type
Installer Icon Type MSB
Installer Icon Type LSB
User Icon Type MSB
User Icon Type LSB

Currenct code in OZW, that is  commit b88f216fbff of Nov 2019:

cpp/src/command_classes/ZWavePlusInfo.cpp

```cpp
...
uint8 version = _data[1];
uint8 role = _data[2];
uint8 nodeType = _data[3];
uint16 installerIcon = (_data[4] << 8) | _data[5];
uint16 deviceType = (_data[6] << 8) | _data[7];
...
```

So imho a better name for "deviceType" would be "userIcon"

After decoding those bits, this gets called:

node->SetPlusDeviceClasses(role, nodeType, deviceType);

In node.cpp that "deviceType" is extensively used as "m_deviceType = _deviceType;"

That m_deviceType is persisted as devicetype when Node::WriteXML gets called:

```
		snprintf(str, 32, "%d", m_deviceType);
		nodeElement->SetAttribute("devicetype", str);
```

Finally deviceType is also exposed as a Value (in ZwavePlusInfo.cpp, but this time with (imho) correct name UserIcon

```
if ((svalue = static_cast<Internal::VC::ValueShort*>(GetValue(_instance, ValueID_Index_ZWavePlusInfo::UserIcon))))
{
	svalue->OnValueRefreshed(deviceType);
	svalue->Release();
}
```

The Icon Types are defined by the standard, in the DKD under Registries, "SDS13738 Z-Wave Plus Assigned Icon Types.xlsx"

The spec explains the difference between Installer and User Icon as (the gist) "The Installer Icon Type field indicates the icon to use in Graphical User Interfaces for network management, e.g. in a floor plan. Installer Icons provide a finer granularity than user icons. (...) The User Icon Type field indicates the icon to use in Graphical User Interfaces for end users. User Icons provide a basic granularity. (...)"

Fishwaldo, 2019-11-25:
Ok - So I have to check how DeviceType in the Node Class is used/handled. Versus Basic/Generic/Specific and NodeType (zw+)

This is where I think I went wrong, assuming DeviceType == userIcon.

petergebruers, 2019-11-25:
I think so but I would not say you were wrong, maybe a few years ago it might have been different, but I did not keep the old documents.

I saw a recent example of a somewhat related change in the spec...

If you look at:

SDS11847-30 Z-Wave Plus Device Type Specification 2019-10-10

In the revision log at the start:

Contributions 2019B:
...
  Removed the suggested interview processes
...

For example in the older SDS11847-28 Z-Wave Plus Device Type Specification 2019-04-09 they would still list a recommended interview order for a door lock - keypad:

[image]

But now they say "Refer to [13] interview process for each individual command class." and 13 means SDS14223, Z-Wave Command Class Control Specification

[image]

This suggest to me they kind of move away from interview based on a certain type identification to purely "checking whatever the device is capable of by looking at the CCs".

I haven't looked into this yet, a while ago I noticed that remark "Removed the suggested interview processes" and was intrigued.