OpenZWave/open-zwave #2015: Fibaro Door Window Sensor 2 not working correctly

T-Grave, 2019-11-26:
I have two Fibaro Door Window Sensor 2's in my ZWave network. Whilst I was using Home Assistant integrated zwave functionaly I did not experience any issues with them. Recently I switched over to zwave2mqtt and since then I can't seem to get my Door Sensor's to work properly.

Both sensors only report temperature, battery level and a (binary) "Sensor" property.  The temperature and battery level I assume are working correctly. The binary sensor doesn't respond to me opening the door, tampering with the device, or anything really, the value stays on "false".

Back when I was using HA, the sensor's binary sensor never worked correctly for me, but there was an Access Level property which updated from 22 to 23 when the door opened. This was as described in the manual of the sensor as well.

I've checked my current OZW Logs and it seems the controller does actually receive a message when the door is opening or closing, but these logs are always accompanied with the following message:

`Info, Node015, ApplicationCommandHandler - Received a MultiInstance Message, but QueryStage Isn't Complete yet`

I see multiple instances of parts of the Query Stage being reported as completed in the logs, but since there seem to be multiple stages, maybe part of it isn't correctly handled?

A friend of mine has the same sensor and also uses zwave2mqtt, he's experiencing the same issue.

Part of the logs for Node015: https://pastebin.com/nv66wVyx

Thanks in advance

petergebruers, 2019-11-26:
A few things:
- The device has to get past QueryStage_Complete which is what we (the OZW-guys) call the "interview process" which can be an issue with battery devices because they sleep most of the time. Try to wake it up... This "interview happens because you started a new "cache file", OZW doesn't know what the device capabilities are and needs to query it. Hence the name... You'll see log lines like "Node not responding - moving message to Wake-Up queue: AlarmCmd_SupportedGet" because the device is asleep or because of the "Out of frame flow" errors...
- If you see a lot of "WARNING: Out of frame flow! (0x1f).  Sending NAK." in your log you probably have something else connection to your serial port. It causes OZW to miss date. It is know to happen on some systems and was discussed on the HASS Discord channel. IIRC it happens when a package called "ModemManager" is installed on Ubuntu. You can have a few out those "out of frame" errors at startup but there should be none during normal operation.
- Do yourself a favor and exclude the device then include in non-secure mode. Secure mode severely slows down your network and is only recommend for things you want to control securely eg open/close a door. This might change with S2 security but right now that is not supported.

T-Grave, 2019-11-27:
After re-adding my devices in non-secure mode all of the properties are being displayed correctly. Thank you for your help!

closed 2019-11-27 by T-Grave

poudenes, 2020-05-03:
Have same issue, I removed the node also, add it again, but same problem.