OpenZWave/open-zwave #2017: Segfault caused by nullptr usage (missing ManufacturerSpecific serial number value)

kpine, 2019-11-28:
I'm using OZWCP via docker. The OZW rpm version is 1.6.878-86.1, so a bit out of date (~4 months). I can try to reproduce on master if necessary.

I added a [CT32 thermostat](https://products.z-wavealliance.org/products/1046), and it is detected properly, but shortly after OZWCP will crash with a segfault. It's repeatable every time I start the network.

Here are the [OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3900405/OZW_Log.txt) and [ozwcache_0xf4e76156.xml](https://github.com/OpenZWave/open-zwave/files/3900409/ozwcache_0xf4e76156.xml.txt) files. The cache file was incomplete as it crashes before the file could be saved. I can also upload a core dump file somewhere if wanted.

Here is the stack trace:
```
(gdb) where
#0  OpenZWave::Internal::VC::Value::VerifyRefreshedValue (this=this@entry=0x0, _originalValue=_originalValue@entry=0x50, _checkValue=_checkValue@entry=0x58, _newValue=_newValue@entry=0x7f5a44948610, _type=_type@entry=OpenZWave::ValueID::ValueType_String,
    _originalValueLength=_originalValueLength@entry=0, _checkValueLength=_checkValueLength@entry=0, _newValueLength=_newValueLength@entry=0) at /usr/src/debug/openzwave-1.6.878/cpp/src/value_classes/Value.cpp:482
#1  0x00007f5a4992e069 in OpenZWave::Internal::VC::ValueString::OnValueRefreshed (this=0x0, _value="AA000063812") at /usr/src/debug/openzwave-1.6.878/cpp/src/value_classes/ValueString.cpp:104
#2  0x00007f5a498d0ce0 in OpenZWave::Internal::CC::ManufacturerSpecific::HandleMsg (this=0x7f5a2c00c970, _data=<optimized out>, _length=<optimized out>, _instance=2) at /usr/src/debug/openzwave-1.6.878/cpp/src/command_classes/ManufacturerSpecific.cpp:259
#3  0x00007f5a498d5818 in OpenZWave::Internal::CC::MultiInstance::HandleMultiChannelEncap (this=this@entry=0x7f5a2c02aec0, _data=_data@entry=0x7f5a44948968 "\r\001\001r\a\001\vAA000063812\327", _length=_length@entry=19)
    at /usr/src/debug/openzwave-1.6.878/cpp/src/command_classes/MultiInstance.cpp:609
#4  0x00007f5a498d6a2e in OpenZWave::Internal::CC::MultiInstance::HandleMsg (this=0x7f5a2c02aec0, _data=0x7f5a44948968 "\r\001\001r\a\001\vAA000063812\327", _length=19, _instance=<optimized out>)
    at /usr/src/debug/openzwave-1.6.878/cpp/src/command_classes/MultiInstance.cpp:186
#5  0x00007f5a49996527 in OpenZWave::Node::ApplicationCommandHandler (this=0x7f5a2c00b5a0, _data=0x7f5a44948962 "", encrypted=<optimized out>) at /usr/src/debug/openzwave-1.6.878/cpp/src/Node.cpp:1983
#6  0x00007f5a49985ca2 in OpenZWave::Driver::ProcessMsg (this=this@entry=0x7f5a38000e70, _data=_data@entry=0x7f5a44948962 "", _length=<optimized out>) at /usr/src/debug/openzwave-1.6.878/cpp/src/Driver.cpp:2177
#7  0x00007f5a49986d35 in OpenZWave::Driver::ReadMsg (this=this@entry=0x7f5a38000e70) at /usr/src/debug/openzwave-1.6.878/cpp/src/Driver.cpp:1691
#8  0x00007f5a49988c88 in OpenZWave::Driver::DriverThreadProc (this=0x7f5a38000e70, _exitEvent=<optimized out>) at /usr/src/debug/openzwave-1.6.878/cpp/src/Driver.cpp:438
#9  0x00007f5a4993c3f3 in OpenZWave::Internal::Platform::ThreadImpl::Run (this=0x7f5a38001ac0) at /usr/src/debug/openzwave-1.6.878/cpp/src/platform/unix/ThreadImpl.cpp:166
#10 0x00007f5a4993c409 in OpenZWave::Internal::Platform::ThreadImpl::ThreadProc (_pArg=<optimized out>) at /usr/src/debug/openzwave-1.6.878/cpp/src/platform/unix/ThreadImpl.cpp:153
#11 0x00007f5a485c6e65 in start_thread () from /lib64/libpthread.so.0
#12 0x00007f5a482ef88d in clone () from /lib64/libc.so.6
```
Seems to be trying to lookup a serial number value id for instance 2 (ValueID_Index_ManufacturerSpecific), but the value id is null, and so accessing the null ptr crashes it.

kpine, 2019-11-28:
FYI, I just tried the latest ozw-mqtt docker image, and it has the same crash. I presume that's a pretty recent build of OZW.

Fishwaldo, 2019-11-29:
Fixed this! thanks

I'll look into updating the RPM's (and thus OZWCP docker image) soon.

closed 2019-11-29 by Fishwaldo

Fishwaldo, 2019-11-29:
FYI - RPM's are building again, and docker image should rebuild soon.