OpenZWave/open-zwave #2018: Qubino Flush 1 Relay older models have different roup associations

StyxyDog, 2019-11-28:
### Issue Guidelines

Before opening a new issue, please review the following:
* The Github Issues is for OZW Bugs only. Please do not use this for general questions or howto type conversations. If you have a question, please post to our mailing list at https://groups.google.com/forum/#!forum/openzwave
* Please use the search feature to see if your issues have been raised or addressed first.
* Try the OpenZWave Log Analyzer at http://www.openzwave.com/log-analyzer if you are experiencing problems with a device on your network.
* Often we will require Log Files to help diagnose the issue. Please stop any application you are using, remove zwcfg_*.xml/ozwcache_*.xml files, and restart your application. After experiencing your issue, stop your application and attach the OZW_Log.txt file to this issue (please do not filter the log file, we often need the full log file to diagnose issues)
* If you want to add a new device to the OZW Database, please consult https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

StyxyDog, 2019-11-28:
Looking here
https://github.com/OpenZWave/open-zwave/blob/e89d2a1677f5afe74e836d8c75d7f33d784ea790/config/qubino/ZMNHADx.xml#L162

I notice that `auto="true"` is not on some of the other qubino device xml files. I suspect that removing this would still leave the group 9 available if needed but it wouldn't force the association for  devices that do not support it. I'm not sure I understand enough to justify the change though, perhaps someone could see if that would make more issues than it solves? Thanks

Fishwaldo, 2019-11-28:
Can you please upload a log file after restarting your app without a ozwcache* file. I suspect I know why, but a log would confirm it.

petergebruers, 2019-11-28:
What @Fishwaldo said, plus:
Not sure why it is flooding, log can help, but iirc "auto" is convenient for those having a temperature sensor connected. Without association in group 9 the device does not report temperature.

StyxyDog, 2019-11-28:
Hi Thank you for helping.
Here is a log file of the node being added then the flooding of AssociationCmd_Get and AssociationCmd_Set that follows. If this log file does not show what you want I'll undo my temp fix and 
upload a full log from restart.
I am running ozw trough home assistant.
I have got around the problem for now by editing the zwcfg_*.xml to remove that auto setting. 
[addnode.txt](https://github.com/OpenZWave/open-zwave/files/3902610/addnode.txt)

Fishwaldo, 2019-11-29:
Its returning Group 0 when we ask for Group 9... I suspect that Group 0 means "does not exist" and unfortuantly OZW keeps trying as its looking for Group 9. 

@petrovcicklemen Can you comment and confirm. Returning Group 0 means its a invalid group? I can add a test to skip it then.

Fishwaldo, 2019-11-29:
>I am running ozw trough home assistant.

Just a FYI, Home assistant is still using OZW 1.4. A fix will be applied to OZW 1.6.

petrovcicklemen, 2019-11-29:
Hi all. This is a 5 years old version of the product, which was distributed well before my time on the company, so I'm not familiar with it. On our latest versions of the products, we return the data for the group1 (Lifeline), if a GET for a non-existing association group is requested.

@Fishwaldo : I think you are correct in assuming that 0 is an invalid group. We can try to look for the manual of that version, but we'd need to confirm how many association groups the root device advertises (ASSOCIATION_GROUPINGS_REPORT).

Apologies for not having any more information on this.

RicardP, 2019-11-29:
Hi, 
running latest Domoticz 4.11535 / OZW 1.6.954.gaa29b66e.dirty on Win10...

Same device so I provide some experiences, FYI

I have two newly purchased and installed ZMNHAD1, have some issues with especially one of them that has a temp sensor connected (dont know if the tempsensor is the caue). Both Qubino Relays jumps via AEON Labs ZW117 Range Extender 6 to the Aeon Labs Gen5 Controller. I Activated polling to help get Power W reported into OZW, if switching it on (load is 2KW) I anyway wont get power W updated immediately, as it should. If I go into OZW cpl and press Refresh it updates W but causes a lot of timeout notifications to the log, also sometimes I get negative bogus W reported... even that my network looks almost perfect with only like 5 of 31 nodes are jumping its data, all other nodes have direct contact with the controller. It seems ZWave Mesh protocol does not work very well at all, or why Time outs from a Node when it jumps?

Here is how different the ZMNHAD1 looks (Node 100 and 101) with and without temp sensor... any comments? :)

[image]

[image]

[image]

StyxyDog, 2019-11-29:
> > I am running ozw trough home assistant.
> 
> Just a FYI, Home assistant is still using OZW 1.4. A fix will be applied to OZW 1.6.
@Fishwaldo Thats great thank you. I have a workaround in place so I can continue using this device with home assistant and ozw 1.4, and should that ever get upgraded to 1.6 I'll be sorted. 
Thanks very much. :)

@RicardP I'm not sure that is the same issue. The Flush 1 relay has come in different versions over the last few years. The documentation on the quibino doesn't seem to recognise this change in hardware. This issue seems to relate to the one that is labelled ZMNHAD1 **H1S1P1**. I have other Flush 1 and Flush 2 relays that do not exhibit this behaviour, I can't check how they are labelled as they are all in the wall, but I know they are later versions of the same product.

Fishwaldo, 2019-12-11:
@StyxyDog Thanks for the feedback. Closing this issue

@RicardP your issue is different and not related.

closed 2019-12-11 by Fishwaldo