OpenZWave/open-zwave #2019: Proxy support

jano42, 2019-11-29:
I'm wondering if you plan to support proxies ? (for downloading config files)

As this might be complex, one other solution would be for calling code to provide an httpClient (there's already a method in Driver class ````bool Driver::setHttpClient(Internal::i_HttpClient *client)````
but parameter is internal (and not accessible from Manager public api).
For example in my app, I already have my http layer which handle proxies, http redirections,.... (quite huge work).

But for that the i_HttpClient should be exposed...

I think giving ability of external code to provide an HttpClient could be a good solution; which allow to handle any custom behavior (proxy w/o authentication, redirections, ,...)

Would you like I make a PR about that ?

Fishwaldo, 2019-12-01:
SSL and Proxy are on the todo.... 

SSL - I don't like pulling in external dependencies though... there is support for SSL in the platform/HttpClient.cpp/h files for PolarSSL.

Proxies - Support would have to be added to HttpSocket::SendRequest

You can call the Driver::setHttpClient directly, but it should be exposed... I'm just not sure if via the Manager Class is the right place for it. (we do a similar implementation for the Log class and thats not exposed via the Manager class either).

>But for that the i_HttpClient should be exposed...

We could move it from the OpenZWave::Internal to the OpenZWave namespace I guess... But.... that breaks the API. So maybe future major revision?

Fishwaldo, 2019-12-01:
>We could move it from the OpenZWave::Internal to the OpenZWave namespace I guess... But.... that breaks the API. So maybe future major revision?

Actually, given you are the first to ask about this... there might not be any users of that setHttpClient (and thus i_HttpClient) so maybe we can do it... let me think about it and where to put the setHttpClient method...

jano42, 2019-12-02:
If you want, I've done a small quick PR for that.

_Just another words of the integration of the PolarSSL library. You should have a look at (PocoProject)[https://pocoproject.org]. This is a complete framework which manage lots of things, and one is Poco::Net (associate with Poco::NetSSL, based on OpenSSL) mange everything around HTTP (request, webserver, ssl, proxies,...). This is a quite good library, stable, opensource, which build on any platform. This is just an example. I use this library for 5 years and this allow me to concentrate my developement on my own job, withtout losting my time on writing something which is already validated by others..._