OpenZWave/open-zwave #2020: Problems with ZME_WALLC-S after Update from 1.4 to 1.6 - states missing

fbatsb17, 2019-11-29:
Hi, 
I am using Zwave in IOBroker. Everything's fine until updating iobroker.zwave from 1.4 to 1.6. So I would say thank you for your work. By running the update, also the openzwave Version changes from 1.4 to 1.6.
After the update, the Group "CLASSES91" was missing and only two states (Scene_Count_1 and Scene_Reset_Timeout) were found in "CENTRAL_SCENE". Many different tries of installing the Zwave Modul leading to the same Problem. Especially state Scene_Number_1 is still missing - which I was using to read the Commands, coming from the wall plug.
In the Logfile I found:
```
2019-11-07 19:30:54.529 Info, Node005, Received Central Scene set from node 5: scene id=4 with key Attribute 0. Sending event notification.
2019-11-07 19:30:54.529 Warning, Node005, No ValueID created for Scene 4
2019-11-07 19:30:54.529 Warning, Node005, CommandClass COMMAND_CLASS_CENTRAL_SCENE HandlerMsg Returned False

``` 
Discussions in the Github and in the forum of iobroker indicates on same Problems on "Devolo wall switch" and "heatit z-push button 8" (take a look at https://github.com/ioBroker/ioBroker.zwave/issues/99 and https://forum.iobroker.net/topic/26440/probleme-nach-zwave-update-mit-zme_wallc-s/28).

Hope, anyone get's a solution for this Problem.
Thanks in advance

petergebruers, 2019-11-29:
I own a ZME_WALLC-S but I am not using this device... This means, I don't know it very well but I was able to make a test.

But first... I am a volunteer and you have to be very patient with me, I am (very) slow...

I can tell you what is going on, when you see this: "No ValueID created for Scene 4" the "interview process is not finished yet and OpenZWave does no know yet what that wall switch can "do".

This is a battery (CR2032) device and it does not stay awake very long... And there is something which I have to investigate, OZW is querying certain data and the device does not respond which makes OZW time out, then the device goes back to sleep. In other words, you have to keep waking it (push 4 buttons for 5 seconds, led turns green, then press key 2). This is in OZW_Log:

```

2019-11-29 20:00:07.656 Info, Node027, MultiInstanceCmd_Get for COMMAND_CLASS_BASIC
2019-11-29 20:00:07.656 Detail, Node027, Queuing (Query) MultiInstanceCmd_Get (Node=27): 0x01, 0x0a, 0x00, 0x13, 0x1b, 0x03, 0x60, 0x04, 0x20, 0x25, 0x1a, 0x85
2019-11-29 20:00:07.656 Info, Node027, MultiInstanceCmd_Get for COMMAND_CLASS_SWITCH_MULTILEVEL
2019-11-29 20:00:07.656 Detail, Node027, Queuing (Query) MultiInstanceCmd_Get (Node=27): 0x01, 0x0a, 0x00, 0x13, 0x1b, 0x03, 0x60, 0x04, 0x26, 0x25, 0x1b, 0x82
2019-11-29 20:00:07.656 Info, Node027, MultiInstanceCmd_Get for COMMAND_CLASS_SWITCH_ALL
2019-11-29 20:00:07.656 Detail, Node027, Queuing (Query) MultiInstanceCmd_Get (Node=27): 0x01, 0x0a, 0x00, 0x13, 0x1b, 0x03, 0x60, 0x04, 0x27, 0x25, 0x1c, 0x84
2019-11-29 20:00:07.656 Info, Node027, MultiInstanceCmd_Get for COMMAND_CLASS_SCENE_ACTIVATION
2019-11-29 20:00:07.656 Detail, Node027, Queuing (Query) MultiInstanceCmd_Get (Node=27): 0x01, 0x0a, 0x00, 0x13, 0x1b, 0x03, 0x60, 0x04, 0x2b, 0x25, 0x1d, 0x89
2019-11-29 20:00:07.656 Info, Node027, MultiInstanceCmd_Get for COMMAND_CLASS_DEVICE_RESET_LOCALLY
2019-11-29 20:00:07.656 Detail, Node027, Queuing (Query) MultiInstanceCmd_Get (Node=27): 0x01, 0x0a, 0x00, 0x13, 0x1b, 0x03, 0x60, 0x04, 0x5a, 0x25, 0x1e, 0xfb
2019-11-29 20:00:07.656 Info, Node027, MultiInstanceCmd_Get for COMMAND_CLASS_CENTRAL_SCENE
2019-11-29 20:00:07.656 Detail, Node027, Queuing (Query) MultiInstanceCmd_Get (Node=27): 0x01, 0x0a, 0x00, 0x13, 0x1b, 0x03, 0x60, 0x04, 0x5b, 0x25, 0x1f, 0xfb
2019-11-29 20:00:07.656 Info, Node027, MultiInstanceCmd_Get for COMMAND_CLASS_ZWAVEPLUS_INFO

(...)

2019-11-29 20:00:07.657 Detail, Node027, Queuing (Query) MultiInstanceCmd_Get (Node=27): 0x01, 0x0a, 0x00, 0x13, 0x1b, 0x03, 0x60, 0x04, 0x8f, 0x25, 0x2a, 0x1a
2019-11-29 20:00:07.657 Detail, Node027, Queuing (Query) Query Stage Complete (Instances)
```

I counted 17 times MultiInstanceCmd_Get and they are all failing on my device. That might be firmware version dependent.

If  you wake up the device you'll see this many times:

```
2019-11-29 20:00:07.686 Detail, Node027,   Expected callbackId was received
2019-11-29 20:00:12.658 Error, Node027, ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "MultiInstanceCmd_Get (Node=27): 0x01, 0x0a, 0x00, 0x13, 0x1b, 0x03, 0x60, 0x04, 0x20, 0x25, 0x1a, 0x85"
2019-11-29 20:00:12.658 Detail, Node027, Removing current message
2019-11-29 20:00:12.658 Detail, Node027, Notification: Notification - TimeOut
```

At a certain point you'll see all those request will get moved to the wake up queue:

`Node027, Node not responding - moving message to Wake-Up queue (many times)`

So you'll have to do that wake-up sequence again (very tedious!) - each time you see in the log "Node027, Notification: Notification - Node Asleep".

If you look at the timestamps (20:00:07 -> 20:00:12) you can see I have reduced the timeout from 10 to 5 seconds in my test environment. It is in options.xml

If you want to try that, I'd recommend to reduce the timeout maybe even more, to 3 seconds.

Anyway, I did not count them, but after 10-15 times waking up the device... Progress...

Once you get past these "Get"s it will accelerate, and get to the point where those (missing) ValueIDs are created:

```
Notification: Value Added Home 0xcb4ff894 Node 27 Genre user Class CENTRAL SCENE Instance 1 Index 1 Type list IsSet False ValueAsString Inactive
(...)
Notification: Value Added Home 0xcb4ff894 Node 27 Genre user Class CENTRAL SCENE Instance 1 Index 8 Type list IsSet False ValueAsString Inactive
```

After waking up the device one more time I get this important message:

```
Node027, QueryStage_Complete
```

OZW will write its config to ozwcache.

Then I get 4 functioning buttons... Here are some examples from OZW_Log, they are self-explanatory:

```
Node 27 Genre user Class CENTRAL SCENE Instance 1 Index 1 Type list IsSet True ValueAsString Pressed 1 Time
Node 27 Genre user Class CENTRAL SCENE Instance 1 Index 6 Type list IsSet True ValueAsString Pressed 1 Time
Node 27 Genre user Class CENTRAL SCENE Instance 1 Index 2 Type list IsSet True ValueAsString Key Held down
```

I am not claiming this will solve all your issues with this device... It is the first time I use it with OZW and previously it was in my box with "test devices" because I did not particularly like the device on my Fibaro HC2, that's why I cannot tell you much about it. But it will at least transfer key presses to IOBroker.

To summarize:

- OZW has trouble interviewing the device. Or the device has issues passing the interview process, it depends on how you look at it.
- I'll mark this issue as "investigate by devs". The devs are already working on a few things, please be patient.
- You can try what I said, reduce timeout in options.xml, keep waking up the device while watching your log, until you get QueryStage_Complete on that node.

fbatsb17, 2019-11-30:
Hi and thank you for your extensive answer.
I found the `options.xml` in `/usr/local/etc/openzwave`, with the 
```
<Option name="RetryTimeout" value="40000" />
```
in it. I was deleting the comment and try to generate the missing states by updating the node information. Pressing all four buttons for 5 sec doesn't work for me; no green LED appears. I have to press all four, wait until the LED is flashing an then press button 2 to send out a NIF/waking the node up.
But - nothings happened.
I switched RetryTimeout down to 3000 (ms), and now it works. 
Happy as I was, I was trying the same procedure with the second (of four) ZME_WALLC-S.
The end result in this attempt: nothing turns better, no state appears. I was trying 20 times, I was counting by a list.

Summary:

- workaround works for one but not for the other button
- issue still open for me
- I can't understand, why the Information appears in IOBroker until the last Update, but not now

petergebruers, 2019-11-30:
Thank you for trying that and reporting back.

> Pressing all four buttons for 5 sec doesn't work for me; no green LED appears. I have to press all four, wait until the LED is flashing an then press button 2 to send out a NIF/waking the node up.

I am not sure what you are saying here. I said  (and I quote) "(push 4 buttons for 5 seconds, led turns green, then press key 2)" - isn't your LED green? Should I rephrase that?

> Happy as I was, I was trying the same procedure with the second (of four) ZME_WALLC-S.
> The end result in this attempt: nothing turns better, no state appears. I was trying 20 times, I was counting by a list.

That is annoying. I am not sure about the difference (firmware? distance?) and I understand it is very hard to try this (more than) 20 times...

> * can't understand, why the Information appears in IOBroker until the last Update, but not now

I don't know, I only know about Home Assistant and Domoticz and going from OZW 1.4 to OZW 1.6 causes some issues. We have to upgrade because Z-Wave is not "set in stone", it is an evolving standard. So OZW has to evolve too.

Yes, issue still open....

petergebruers, 2019-11-30:
@fbatsb17 forget that question about the wake up procedure. We agree. What happened is this. You received an email when I posted, but a bit later I edited that part, but you don't get an email when that happens...

fbatsb17, 2019-11-30:
everything is fine, you did/do a good job and we are thankful for it
developing is a good thing and so it's logic that not every working thing can be set in stone
since 5th of Nov, I was discussing with the developers of the zwave Adapter in IOBroker about this issue. In their opinion, iobroker is only able to work with the information, which is sent from OZW. And you investigation from today shows (in my opinion) that this is right.
so I am hopefully, that you can find the difference in handling battery devices between 1.4 and 1.6 and fix this issue

gaudes, 2019-11-30:
Hi,

same problem here with a Devolo wall switch:

`2019-11-22 12:09:34.139 Detail, Node012,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x0c, 0x05, 0x5b, 0x03, 0xd9, 0x00, 0x01, 0xbf, 0x00, 0xc0
2019-11-22 12:09:34.139 Info, Node012, Received Central Scene set from node 12: scene id=1 with key Attribute 0. Sending event notification.
2019-11-22 12:09:34.139 Warning, Node012, No ValueID created for Scene 1
2019-11-22 12:09:34.139 Warning, Node012, CommandClass COMMAND_CLASS_CENTRAL_SCENE HandlerMsg Returned False
2019-11-22 12:11:23.798 Detail, Node012,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x0c, 0x05, 0x5b, 0x03, 0xda, 0x00, 0x02, 0xbc, 0x00, 0xc3
2019-11-22 12:11:23.798 Info, Node012, Received Central Scene set from node 12: scene id=2 with key Attribute 0. Sending event notification.
2019-11-22 12:11:23.798 Warning, Node012, No ValueID created for Scene 2
2019-11-22 12:11:23.798 Warning, Node012, CommandClass COMMAND_CLASS_CENTRAL_SCENE HandlerMsg Returned False
`

Thanks and regards,
Ralf

petergebruers, 2019-11-30:
@gaudes I googled "Devolo wall switch" and it looks like a rebranded ZME_WALLC-S switch. If you want a workaround, try to apply what I've said a few posts back...

Fishwaldo, 2019-12-01:
>so I am hopefully, that you can find the difference in handling battery devices between 1.4 and 1.6 and fix this issue

Central Scene never worked in 1.4 unless you hacked zwcfg or the config files, so I'm not so sure about that statement. 

In 1.6, Central Scene was properly implemented, supporting auto discovery etc

You need to make sure the device reaches QueryStage_Complete - Before then, it will be unusable. (even in 1.4 series). That means you need to wake it up

I recall there was a problem with these devices where they didn't "wake up" for long, hence you had to do the wakeup dance (as peter mentioned) several times.

To confirm, please shutdown your app, remove any ozwcache*.xml file and start up. Then try doing the wakeup dance several times and send us the OZW_Log.txt

>If you want to try that, I'd recommend to reduce the timeout maybe even more, to 3 seconds.

To Clarify -  You want to reduce this for testing, but I wouldn't recommend it for production

fbatsb17, 2019-12-01:
> Central Scene never worked in 1.4 unless you hacked zwcfg or the config files, so I'm not so sure about that statement.

I never claimed, Central Scene has worked in 1.4. In 1.4 I was using CLASSES91 an there the value Scene_Number (as you find in my opening text; perhaps my english is a part of the problem  ;-) )
[image]
to get the information, which button was pressed. 

in the last hour I was removing the ZME_WALLC-S (my second) from the ZWave network and including it again. You are right, after the including only a very rudimentary number of information was found for the node and I had to wake it up. After only one wakeup a large number varied properties but not the Scene_1 to _8
[image]
(screenshot from the first ZME_WALLC-D) And the next 10 steps in the wakeup dance won't make a difference:
[image]

removing the cache file on another day - too little time today

Fishwaldo, 2019-12-02:
I don't know IOBroker so what you are showing doesn't mean anything to me... hence why I asked for Log Files... but
[image]

Those values where *never* in the official OpenZWave. 

CentralScene in 1.6 works very differently. You can not compare what you had (which wasn't even official) with what is in 1.6. Please ensure you read https://github.com/OpenZWave/open-zwave/wiki/OpenZWave-1.6-Release-Notes#central-scene-commandclass to get a glimpse of how it now works.

petergebruers, 2019-12-02:
@fbatsb17 like @Fishwaldo I have never used IOBroker so please understand I cannot debug that part, but as @Fishwaldo points out, it might be an implementation issue why - even after many wake-up cycles - you don't get to see anything in your app. I understand it might be difficult for them to test but OZW 1.6 now has a standard way of doing "Central Scene CC" so the IOBroker guys can test with other devices, they do not specifically need the ZME_WALLC-S.

I have some interesting news. We can rule out a (fundamental) ZME_WALLC-S issue because I was able to add the device to my Fibaro HC2 with only one wake-up button press.

I have captured the HC interview with Zniffer, so I will be able to compare it with the OZW interview.

[Zniffer 026 - HC2 ZME_WALLC-S 2019-12-02.zlf.zip](https://github.com/OpenZWave/open-zwave/files/3910866/Zniffer.026.-.HC2.ZME_WALLC-S.2019-12-02.zlf.zip)

If someone wants to have a peek, and you are on Windows, you can decode that file by downloading the free Zniffer tool from the Silabs website. Unfortunately it is impossible to get a good plain-text version of such a capture (you can, but it will lack detail)

If you want to know more about Zniffer you can register a free account on the Fibaro forum and read this:

https://forum.fibaro.com/topic/29923-tutorial-z-wave-diagnostics-with-pc-controller-and-zniffer/

cburghardt, 2019-12-03:
iobroker basically creates states based upon information from openzwave, nothing more, nothing less. Central scenes also work for example with Fibaro Keyfob. However that devices seems to be awake long enough to complete the query stage and all information is present.

petergebruers, 2019-12-03:
Thanks. I'll probably be able to investigate this next Thursday.

petergebruers, 2019-12-05:
The main issue is, OZW before this PR sees "Multi Channel CC" and queries the endpoints. But the device is not MC itself, it only claims to be "controlling" that CC.

The second interesting thing I've noticed in the Zniffer capture is that Fibaro HC2 uses Multi Command Encapsulation, often bundling 6 commands in one packet and getting 2 responses in one packet, during interview. This is something OZW cannot do at the moment.

Fishwaldo, 2019-12-06:
Shower Thoughts: it should be straight forward to implement MultiCommand in Driver::WriteMsg or Driver::WriteNextMsg by just scanning the next message(s) in the send queues. 

Semi random question: what’s the max payload size of a zwave message. I swore there was a define and check in OZW but it’s eluding me. I have a inkling it was 95 bytes?

Fishwaldo, 2019-12-07:
So I dug into the MultiCommand... should be simple:
https://github.com/OpenZWave/open-zwave/blob/9526554bdffd9e2049b3e9ad1ab0d119fdb9595f/cpp/src/Driver.cpp#L996

Here, we can just check the next message in the queue with a Iterator - If its a item.m_nodeId == currentMsg.m_nodeId and item.m_command == MsgQueueCmd_SendMsg then just start encapsulating the "next iterator" message into a MultiCommand... and keep going till either of those conditions are not true.

Regarding packet sizes:
MultiCast (which we dont support):
>The maximum size of a packet is 64 bytes. The protocol header, multicast addresses and checksum takes 39 bytes in a multicast frame leaving 25 bytesfor the payload.The payload must be minimum one byte

For the rest:
[image]

Fishwaldo, 2019-12-07:
[image]

petergebruers, 2019-12-07:
Yes, that's the same data I based my assumptions on, while fixing PST02 MultiCmd. If you want to be on the safe side (when you don't know the encryption level) then the payload size indeed drops to 26 bytes. The PST02 indeed uses that to split its MultiCmd into 2 secure messages. I haven't checked the sizes of the ZME_WALLC-S packets in normal mode on the HC2, I'll verify that.

Fishwaldo, 2019-12-07:
If we can sequence frames working on S0 then the size should be ok. 

I now have a device that can generate packets bigger than the frame size (the new Aeotec recessed door sensor with Config Command Class version 4) so I should be able to fix that up as well (at least receiving side)

fbatsb17, 2019-12-07:
I'm sorry that posting the pictures out of iobroker has led to confusion. I was searching for a short way to display the known values in iobroker. Anyway. A long hard working week is over...

iobroker is stopped and I removed the ozwcache*.xml and the OZW_Log.txt file.
After Problems restarting the zwave Modul in iobroker, I was starting doing the wake up dance at 12:55.
The first 10 times I was doing at my desk; after 13:09 I was shorten the distance to the zwave USB dongle under 2m. After five more times requesting new information from iobroker for the node, I'm only awaking the node three times (in 13:17 - 13:20) - absolutely nothing s happen. I tried five more times: iobroker is requesting, node is sending NIF, but central scene was only filled up with Scene_Count_1 and Scene_Reset_Timeout.
Last but not least: the [OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3935066/OZW_Log.txt)

petergebruers, 2019-12-07:
Not much confusion here...
What's the version of OZW? A PR was accepted very recently, do you build OZW from source?

cburghardt, 2019-12-07:
We use 1.6.6xx (around 619 if I remember rightly) . And yes, build from source

fbatsb17, 2019-12-07:
I think it's 1.6
no, I am using OZW as a part of the Zwave Adapter in iobroker

petergebruers, 2019-12-07:
> I think it's 1.6
> no, I am using OZW as a part of the Zwave Adapter in iobroker
> I think it's 1.6
> no, I am using OZW as a part of the Zwave Adapter in iobroker

The fix is in v1.6-984-g6c3939ae and today we've bumped OZW to v1.6-989-g9526554b

OZW prints the version at the start of the log.

fbatsb17, 2019-12-07:
2019-12-07 12:45:45.753 Always, OpenZwave Version 1.6-953-g79663e1e-dirty Starting Up

petergebruers, 2019-12-07:
So please try this with v1.6-984 or higher: properly exclude the device,  restart your gateway software. Include the device, wake it up at least 1 time. If you still do not get all scene events, zip the log and drag and drop it here. Please do not edit it and please do not restart your gateway software because that will erase your log.

fbatsb17, 2019-12-07:
I am afraid, I have no idea, how to change to v1.6-984

petergebruers, 2019-12-07:
> I am afraid, I have no idea, how to change to v1.6-984

That's something you have to ask to the ioBroker guys (forum, wiki, chat, ..) ... I do not run ioBroker and a quick search does not bring up instructions.

cburghardt, 2019-12-07:
I will post instructions how you you can upgrade the ozw version on the 
corresponding iobroker issue 
(https://github.com/ioBroker/ioBroker.zwave/issues/99)

fbatsb17, 2019-12-08:
Hi guys - this is awesome, grateful teamwork. Updating to OpenZwave Version 1.6-989-g9526554b, how described in [ioBroker/ioBroker.zwave/issues/99](https://github.com/ioBroker/ioBroker.zwave/issues/99) works for me. States are present after only one wake up. No exclusion necessary. 
Thanks for your work.

closed 2019-12-08 by fbatsb17