OpenZWave/open-zwave #2029: HELP! Danfoss Air config file

paulhartlieb, 2019-12-07:
Hi!
Can someone help me to create a config file for Danfoss air unit? The entry in the manufacturer_specific.xml looked like this for me:
```
<Manufacturer id = "0002" name = "Danfoss">
    <Product config = "danfoss / air.xml" id = "0200" name = "Air Unit" type = "8007" />
```
I already created the file air.xml, but it does not work:
```
<Product Revision = "1" xmlns = "https://github.com/OpenZWave/open-zwave">
  <MetaData>
    <MetaDataItem name = "OzwInfoPage"> http://www.openzwave.com/device-database/0002:8007:0200 </ MetaDataItem>
    <MetaDataItem name = "ProductPic"> images / danfoss / z.png </ MetaDataItem>
    <MetaDataItem id = "0200" name = "ZWProductPage" type = "8007"> https://products.z-wavealliance.org/products/173/ </ MetaDataItem>
    <MetaDataItem id = "0200" name = "Identifier" type = "8007"> 014G0013 </ MetaDataItem>
    <MetaDataItem id = "0200" name = "FrequencyName" type = "8007"> CEPT (Europe) </ MetaDataItem>
    <MetaDataItem name = "Name"> Danfoss Air Unit </ MetaDataItem>
    <MetaDataItem name = "Description"> Ventilation </ MetaDataItem>
  <! - COMMAND_CLASS_HRV_STATUS ->
<CommandClass id = "55">
    <Value genre = "config" index = "0" label = "Outdoor Air temperature" max = "100" min = "1" type = "short" value = "100">
      <Help>
       Outdoor Air temperature
      </ Help>
    </ Value> <Value genre = "config" index = "1" label = "Supply Air (to room) temperature" max = "100" min = "1" type = "short" value = "100">
      <Help>
       Supply Air (to room) temperature
      </ Help>
    </ Value>
    <Value genre = "config" index = "2" label = "Exhaust Air (from room) temperature" max = "100" min = "1" type = "short" value = "5">
      <Help>
        Exhaust Air (from room) temperature
      </ Help>
    </ Value>
    <Value genre = "config" index = "3" label = "Discharge Air temperature" max = "40" min = "0" type = "short" value = "12">
      <Help>
       Discharge Air temperature
      </ Help>
    </ Value>
    <Value genre = "config" index = "4" label = "Room temperature" max = "40" min = "0" type = "short" value = "28">
      <Help>
       Room temperature
      </ Help>
    </ Value>
    <Value genre = "config" index = "5" label = "Relative Humidity in room" size = "1" type = "short" value = "1">
      <Help>
       Relative Humidity in the room
      </ Help>
    </ Value>
    <Value genre = "config" index = "6" label = "Remaining filter life" size = "1" type = "short" value = "1">
      <Help>
       Remaining filter life
      </ Help>
    </ Value>
  </ Command Class>
</ Product>
```
Sorry for my english, is a google-translation from german

paulhartlieb, 2019-12-07:
I found some more information, but where I should add something I do not know:
`<ZWaveDevice xmlns="http://www.pepper1.net/zwavedb/xml-schemata/z-wave" schemaVersion="2">
<descriptorVersion>1</descriptorVersion>
<deviceData>
<manufacturerId value="0002"/>
<productType value="8007"/>
<productId value="0200"/>
<libType value="01"/>
<protoVersion value="02"/>
<protoSubVersion value="43"/>
<appVersion value="01"/>
<appSubVersion value="0e"/>
<basicClass value="02"/>
<genericClass value="16"/>
<specificClass value="01"/>
<optional value="false"/>
<listening value="true"/>
<routing value="true"/>
<beamSensor>0</beamSensor>
<rfFrequency>EU</rfFrequency>
<certId>13758</certId>
<certNumber>ZC08-10010012</certNumber>
</deviceData>
<deviceDescription>
<description>
<lang xml:lang="en">CCM module</lang>
</description>
<wakeupNote>
<lang xml:lang="en"/>
</wakeupNote>
<inclusionNote>
<lang xml:lang="en">Press the button.</lang>
</inclusionNote>
<productName>
Heat Recovery Ventilation - Computing and Communications Module
</productName>
<brandName>Danfoss</brandName>
<productLine>Danfoss Air</productLine>
<productCode>089F0257</productCode>
<productVersion>1.10</productVersion>
</deviceDescription>
<commandClasses>
<commandClass id="0020"/>
<commandClass id="0086"/>
<commandClass id="0091" controlled="true"/>
<commandClass id="0072"/>
<commandClass id="0037"/>
<commandClass id="0039"/>
</commandClasses>
<resourceLinks>
<deviceImage url="http://www.pepper1.net/zwavedb/uploads/resources/f2dba617fe5c98f08c94c1897192a555b978952b.jpg"/>
</resourceLinks>
</ZWaveDevice>`

Fishwaldo, 2019-12-08:
What didn’t work?

paulhartlieb, 2019-12-08:
it does not query the CommansClas I inserted in the air.xml file. The data is queried with iobroker, I insert a picture where you can see which data comes from the node.

[https://s19.directupload.net/images/191208/alyoe3j7.jpg](url)

 here is the log file

`2019-12-08 15:32:54.579 Always, OpenZwave Version 1.6-953-g79663e1e-dirty Starting Up
2019-12-08 15:32:54.619 Info, Loading Localization File /usr/local/etc/openzwave/Localization.xml
2019-12-08 15:32:54.621 Info, Loaded /usr/local/etc/openzwave/Localization.xml With Revision 8
2019-12-08 15:32:54.622 Always, Using Language Localization 
2019-12-08 15:32:54.661 Info, Loading NotificationCCTypes File /usr/local/etc/openzwave/NotificationCCTypes.xml
2019-12-08 15:32:54.662 Info, Loaded /usr/local/etc/openzwave/NotificationCCTypes.xml With Revision 9
2019-12-08 15:32:54.664 Info, Loading SensorMultiLevelCCTypes File /usr/local/etc/openzwave/SensorMultiLevelCCTypes.xml
2019-12-08 15:32:54.664 Info, Loaded /usr/local/etc/openzwave/SensorMultiLevelCCTypes.xml With Revision 3
2019-12-08 15:32:54.664 Info, Setting Up Provided Network Key for Secure Communications
2019-12-08 15:32:54.710 Info, Manufacturer_Specific.xml file Revision is 59
2019-12-08 15:32:55.133 Info, Unable to load config file /usr/local/etc/openzwave/danfoss/air.xml
2019-12-08 15:32:55.348 Info, Product name collision: FGR223 Roller Shutter Controller 3 type 303 id 1000 manufacturerid 10f, collides with FGRM223 Roller Shutter Controller 3, type 303 id 1000 manufacturerid 10f
2019-12-08 15:32:55.675 Info, Product name collision: ED2.0 Meter Adapter type 0 id 0 manufacturerid 128, collides with ED2.0 Meter Adapter, type 0 id 0 manufacturerid 128
2019-12-08 15:32:55.676 Info, Product name collision: ED2.0 Display type 128 id 0 manufacturerid 128, collides with ED2.0 Display, type 128 id 0 manufacturerid 128
2019-12-08 15:32:55.935 Info, Product name collision: GR105 Auto Valve type 202 id 511 manufacturerid 152, collides with GR105 Auto Valve, type 202 id 511 manufacturerid 152
2019-12-08 15:32:55.935 Info, Product name collision: GR105n Auto Valve type 3 id 512 manufacturerid 152, collides with GR105n Auto Valve, type 3 id 512 manufacturerid 152
2019-12-08 15:32:55.935 Info, Product name collision: ZSE02 Motion Sensor type 500 id 3 manufacturerid 152, collides with ZSE02 Motion Sensor, type 500 id 3 manufacturerid 152
2019-12-08 15:32:55.990 Info, Product name collision: ZWN-BPC Ceiling Mounted PIR Motion Sensor type 601 id 901 manufacturerid 11a, collides with ZWN-BPC Ceiling Mounted PIR Motion Sensor, type 601 id 901 manufacturerid 11a
2019-12-08 15:32:55.990 Info, Product name collision: ZWN-SC7 7-Button Scene Controller type 801 id b03 manufacturerid 11a, collides with ZWN-SC7 7-Button Scene Controller, type 801 id b03 manufacturerid 11a
2019-12-08 15:32:55.990 Info, Product name collision: ZW15S 15A On/Off Switch type 101 id 102 manufacturerid 11a, collides with ZW15S 15A On/Off Switch, type 101 id 102 manufacturerid 11a
2019-12-08 15:32:55.990 Info, Product name collision: ZW20R 20A TR Duplex Receptacle type 101 id 603 manufacturerid 11a, collides with ZW20R 20A TR Duplex Receptacle, type 101 id 603 manufacturerid 11a
2019-12-08 15:32:55.990 Info, Product name collision: ZWN-RSM1 PLUS-Smart Single Relay Switch Module type 111 id 605 manufacturerid 11a, collides with ZWN-RSM1 PLUS—Smart Single Relay Switch Module, type 111 id 605 manufacturerid 11a
2019-12-08 15:32:55.990 Info, Product name collision: ZW20RM 20A TR Smart Meter Duplex Receptacle type 111 id 101 manufacturerid 11a, collides with ZW20RM 20A TR Smart Meter Duplex Receptacle, type 111 id 101 manufacturerid 11a
2019-12-08 15:32:55.990 Info, Product name collision: ZW500D 500W In-Wall Preset Dimmer Switch type 102 id 201 manufacturerid 11a, collides with ZW500D 500W In-Wall Preset Dimmer Switch, type 102 id 201 manufacturerid 11a
2019-12-08 15:32:55.990 Info, Product name collision: ZW15RM Plus 15A TR Smart Meter Duplex Receptacle type 111 id 105 manufacturerid 11a, collides with ZW15RM Plus 15A TR Smart Meter Duplex Receptacle, type 111 id 105 manufacturerid 11a
2019-12-08 15:32:56.058 Info, Queuing Lookup on mfs.db.openzwave.com for Node 0
2019-12-08 15:32:56.058 Info,   Opening controller /dev/ttyACM0
2019-12-08 15:32:56.058 Info, Trying to open serial port /dev/ttyACM0 (attempt 1)
2019-12-08 15:32:56.058 Info, Starting DNSThread
2019-12-08 15:32:56.059 Info, Timer: thread starting
2019-12-08 15:32:56.059 Detail, Timer: waiting with timeout -1 ms
2019-12-08 15:32:56.059 Info, LookupTxT Checking mfs.db.openzwave.com
2019-12-08 15:32:56.059 Info, mgr,     Added driver for controller /dev/ttyACM0
2019-12-08 15:32:56.059 Info, Serial port /dev/ttyACM0 opened (attempt 1)
2019-12-08 15:32:56.059 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2019-12-08 15:32:56.059 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2019-12-08 15:32:56.059 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2019-12-08 15:32:56.059 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2019-12-08 15:32:56.059 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2019-12-08 15:33:01.074 Info, Lookup for mfs.db.openzwave.com returned 59
2019-12-08 15:33:01.079 Info, ManufacturerSpecificDB Initialized
[... cut ...]
2019-12-08 15:33:01.389 Warning, CheckCompletedNodeQueries all=1, deadFound=0 sleepingOnly=1
2019-12-08 15:33:01.389 Info,          Node query processing complete.
2019-12-08 15:33:01.389 Info, Saving Cache
2019-12-08 15:33:01.389 Info, Node001, Cache Save for Node 1 as its QueryStage_CacheLoad
2019-12-08 15:33:01.389 Info, Node003, Cache Save for Node 3 as its QueryStage_CacheLoad
2019-12-08 15:33:01.390 Detail, Node003, Notification: NodeQueriesComplete
2019-12-08 15:33:01.390 Detail, contrlr, Notification: AllNodesQueried
`

paulhartlieb, 2019-12-10:
This is a excerpt from a new log file
Now I have two more questions: why are the CommandClass 0x37 and CommandClass 0x39 NOT REQUIRED.

and what can I change so that they are REQUIRED

2019-12-10 21:16:22.750 Info, Node003, FUNC_ID_ZW_REQUEST_NODE_INFO Request successful.
2019-12-10 21:16:22.787 Detail, Node003,   Received: 0x01, 0x0f, 0x00, 0x49, 0x84, 0x03, 0x09, 0x02, 0x16, 0x01, 0x20, 0x72, 0x91, 0x37, 0x39, 0x86, 0x69
2019-12-10 21:16:22.787 Detail, 
2019-12-10 21:16:22.787 Info, Node003, UPDATE_STATE_NODE_INFO_RECEIVED from node 3
2019-12-10 21:16:22.787 Info, Node003,   Optional CommandClasses for node 3:
2019-12-10 21:16:22.788 Info, Node003,     COMMAND_CLASS_BASIC (Existing)
2019-12-10 21:16:22.788 Info, Node003,     COMMAND_CLASS_MANUFACTURER_SPECIFIC (Existing)
2019-12-10 21:16:22.788 Info, Node003,     COMMAND_CLASS_MANUFACTURER_PROPRIETARY
### 2019-12-10 21:16:22.788 Info, Node003,   CommandClass 0x37 - NOT REQUIRED
### 2019-12-10 21:16:22.788 Info, Node003,   CommandClass 0x39 - NOT REQUIRED
2019-12-10 21:16:22.788 Info, Node003,     COMMAND_CLASS_VERSION (Existing)
2019-12-10 21:16:22.788 Detail, Node003, AdvanceQueries queryPending=1 queryRetries=0 queryStage=NodeInfo live=1
2019-12-10 21:16:22.788 Detail, Node003,   Expected reply was received
2019-12-10 21:16:22.788 Detail, Node003,   Message transaction complete

petergebruers, 2019-12-10:
I don't know what a Danfoss Air is but it matches this device in the Alliance DB:

https://products.z-wavealliance.org/products/173

Seems is is a pre-Z-Wave Plus device... They don't list supported command classes but you have that log from OZW and it means...

0x37 = COMMAND_CLASS_HRV_STATUS = Heat Recovery Ventilation (HRV) Status Command Class

0x39 = COMMAND_CLASS_HRV_CONTROL = ... you can guess

I've never seen another device implement this and I am afraid OpenZWave has no support for those CCs...

When I "grep" tha Alliance DB I only get one hit on COMMAND_CLASS_HRV_CONTROL

https://products.z-wavealliance.org/products/179

IMHO this is a pretty unique device... It does not support any other interesting CC, I think you won't be able to use it unless someone implements the code for it.

Trivia... I saw you posted some info containing "http://www.pepper1.net/zwavedb" - pepper1 did certification of Z-Wave devices, that was a few years ago!

Fishwaldo, 2019-12-11:
As peter mentioned - We don't support those CommandClasses. I've never heard/seen them before.

paulhartlieb, 2019-12-11:
Is it very hard to implement? Are these the missing data?

```
 <cmd_class key="0x39" version="1" name="COMMAND_CLASS_HRV_CONTROL" help="Command Class Hrv Control" read_only="False" comment="">
    <cmd key="0x05" name="HRV_CONTROL_BYPASS_GET" help="Hrv Control Bypass  Get" comment="" />
    <cmd key="0x06" name="HRV_CONTROL_BYPASS_REPORT" help="Hrv Control Bypass Report" comment="">
      <param key="0x00" name="Bypass" type="BYTE" typehashcode="0x01" comment="">
        <valueattrib key="0x00" hasdefines="False" showhex="True" />
      </param>
    </cmd>
    <cmd key="0x04" name="HRV_CONTROL_BYPASS_SET" help="Hrv Control Bypass Set" comment="">
      <param key="0x00" name="Bypass" type="BYTE" typehashcode="0x01" comment="">
        <valueattrib key="0x00" hasdefines="False" showhex="True" />
      </param>
    </cmd>
    <cmd key="0x02" name="HRV_CONTROL_MODE_GET" help="Hrv Control Mode Get" comment="" />
    <cmd key="0x03" name="HRV_CONTROL_MODE_REPORT" help="Hrv Control Mode Report" comment="">
      <param key="0x00" name="Properties1" type="STRUCT_BYTE" typehashcode="0x07" comment="">
        <fieldenum key="0x00" fieldname="Mode" fieldmask="0x1F" shifter="0">
          <fieldenum value="Off" />
          <fieldenum value="Demand Automatic" />
          <fieldenum value="Schedule" />
          <fieldenum value="Energy Savings Mode" />
          <fieldenum value="Manual" />
        </fieldenum>
        <bitfield key="0x01" fieldname="Reserved" fieldmask="0xE0" shifter="5" />
      </param>
    </cmd>
    <cmd key="0x01" name="HRV_CONTROL_MODE_SET" help="Hrv Control Mode Set" comment="">
      <param key="0x00" name="Properties1" type="STRUCT_BYTE" typehashcode="0x07" comment="">
        <fieldenum key="0x00" fieldname="Mode" fieldmask="0x1F" shifter="0">
          <fieldenum value="Off" />
          <fieldenum value="Demand Automatic" />
          <fieldenum value="Schedule" />
          <fieldenum value="Energy Savings Mode" />
          <fieldenum value="Manual" />
        </fieldenum>
        <bitfield key="0x01" fieldname="Reserved" fieldmask="0xE0" shifter="5" />
      </param>
    </cmd>
    <cmd key="0x0A" name="HRV_CONTROL_MODE_SUPPORTED_GET" help="Hrv Control Mode Supported Get" comment="" />
    <cmd key="0x0B" name="HRV_CONTROL_MODE_SUPPORTED_REPORT" help="Hrv Control Mode Supported Report" comment="">
      <param key="0x00" name="Properties1" type="STRUCT_BYTE" typehashcode="0x07" comment="">
        <fieldenum key="0x00" fieldname="Manual Control Supported" fieldmask="0x0F" shifter="0">
          <fieldenum value="Bypass Open Close" />
          <fieldenum value="Bypass Auto" />
          <fieldenum value="Modulated Bypass" />
          <fieldenum value="Ventilation Rate" />
        </fieldenum>
        <bitfield key="0x01" fieldname="Reserved" fieldmask="0xF0" shifter="4" />
      </param>
      <param key="0x01" name="Bit Mask" type="BITMASK" typehashcode="0x06" comment="">
        <bitmask key="0x00" paramoffs="255" lenmask="0x00" lenoffs="0" />
      </param>
    </cmd>
    <cmd key="0x08" name="HRV_CONTROL_VENTILATION_RATE_GET" help="Hrv Control Ventilation Rate  Get" comment="" />
    <cmd key="0x09" name="HRV_CONTROL_VENTILATION_RATE_REPORT" help="Hrv Control Ventilation Rate  Report" comment="">
      <param key="0x00" name="Ventilation Rate" type="BYTE" typehashcode="0x01" comment="">
        <valueattrib key="0x00" hasdefines="False" showhex="True" />
      </param>
    </cmd>
    <cmd key="0x07" name="HRV_CONTROL_VENTILATION_RATE_SET" help="Hrv Control Ventilation Rate Set" comment="">
      <param key="0x00" name="Ventilation Rate" type="BYTE" typehashcode="0x01" comment="">
        <valueattrib key="0x00" hasdefines="False" showhex="True" />
      </param>
    </cmd>
  </cmd_class>
  <cmd_class key="0x37" version="1" name="COMMAND_CLASS_HRV_STATUS" help="Command Class Hrv Status" read_only="False" comment="">
    <cmd key="0x01" name="HRV_STATUS_GET" help="Hrv Status Get" comment="">
      <param key="0x00" name="Status Parameter" type="CONST" typehashcode="0x0B" comment="">
[... 11 lines cut ...]
        <const key="0x00" flagname="Outdoor Air temperature" flagmask="0x00" />
        <const key="0x01" flagname="Supply Air temperature" flagmask="0x01" />
        <const key="0x02" flagname="Exhaust Air temperature" flagmask="0x02" />
        <const key="0x03" flagname="Discharge Air temperature" flagmask="0x03" />
        <const key="0x04" flagname="Room temperature" flagmask="0x04" />
        <const key="0x05" flagname="Relative Humidity in room" flagmask="0x05" />
        <const key="0x06" flagname="Remaining filter life" flagmask="0x06" />
      </param>
      <param key="0x01" name="Properties1" type="STRUCT_BYTE" typehashcode="0x07" comment="">
        <bitfield key="0x00" fieldname="Size" fieldmask="0x07" shifter="0" />
        <bitfield key="0x01" fieldname="Scale" fieldmask="0x18" shifter="3" />
        <bitfield key="0x02" fieldname="Precision" fieldmask="0xE0" shifter="5" />
      </param>
      <param key="0x02" name="Value" type="VARIANT" typehashcode="0x0C" comment="">
        <variant paramoffs="255" showhex="True" signed="True" sizemask="0x00" sizeoffs="0" />
      </param>
    </cmd>
    <cmd key="0x03" name="HRV_STATUS_SUPPORTED_GET" help="Hrv Status Supported Get" comment="" />
    <cmd key="0x04" name="HRV_STATUS_SUPPORTED_REPORT" help="Hrv Status Supported Report" comment="">
      <param key="0x00" name="Bit Mask" type="BITMASK" typehashcode="0x06" comment="">
        <bitmask key="0x00" paramoffs="255" lenmask="0x00" lenoffs="0" />
      </param>
    </cmd>
  </cmd_class>
```

petergebruers, 2019-12-11:
Someone would have to write code in C++ based on data published by Silabs. I am sorry, but I am not going to do that...

paulhartlieb, 2019-12-11:
thanks for the information.

petergebruers, 2019-12-11:
For sake of completeness... That information is public, it is not "secret" in any way and I can point you to it, but you would have to find a volunteer to write code for you, then work with him to debug that code because you own the device and he probably does not have one to test... It is not impossible, but lots of hurdles to take.