OpenZWave/open-zwave #2035: Compilation error

nyxtom, 2019-12-13:
I'm seeing the following output when running `make`

```
cpp/src/command_classes/MultiInstance.h:87:9: error: private field 'm_useDestAsSrc' is not
      used [-Werror,-Wunused-private-field]
                bool            m_useDestAsSrc;

1 error generated.
make: *** [all] Error 2
```

petergebruers, 2019-12-13:
What version of OZW are you trying to build? It says so at the start of make.
I am asking because the current version is 1.6-992-g76e21d80 and variable "m_useDestAsSrc" is not in any source file. See [downloads](http://old.openzwave.com/downloads/)

nyxtom, 2019-12-13:
I was trying to build 1.6 based on this https://github.com/OpenZWave/open-zwave/commit/890f24b7e88f488eee464ed14c01fbceb276cf2a

nyxtom, 2019-12-13:
Looks like building from the http://old.openzwave.com/downloads/ appears to work instead of the releases shown in github.

closed 2019-12-13 by nyxtom