OpenZWave/open-zwave #2036: Nodes not detected in OZW1.6

Dinth, 2019-12-15:
Hey. Im trying to migrate to OZW1.6 (build:1.6-962-gcbe2b60f-dirty) using Zwave2MQTT 2.0.6-dev docker, unfortunately i found that it doesnt detect many of my devices correctly. 
For example, while all my Eurotronic Spirit-Z thermostats are detected correctly in OZW1.4 (Zwave2MQTT 2.0.6, not 2.0.6-dev), in OZW1.6 only half of them gets detected and other half is detected as "Generic Thermostat V2"
Here is a log from running Refresh Node info on one of the thermostats which doesnt get discovered properly on OZW1.6:

```
ZWave Detail, Node031, Received: 0x01, 0x07, 0x00, 0x13, 0x8d, 0x00, 0x00, 0x8c, 0xea
OpenZWave Detail, Node031, ZW_SEND_DATA Request with callback ID 0x8d received (expected 0x8d)
OpenZWave Info, Node031, Request RTT 1396 Average Request RTT 1061
OpenZWave Detail, Node031, Expected callbackId was received
2019-12-15T07:46:23.601Z z2m:Zwave Notification from node 31: Notification - TimeOut (1)
OpenZWave Error, Node031, ERROR: Dropping command, expected response not received after 1 attempt(s)
OpenZWave Detail, Node031, Removing current message
OpenZWave Detail, Node031, Notification: Notification - TimeOut
2019-12-15T07:46:23.602Z z2m:Zwave node 31 AVAILABLE: - (General Thermostat V2)
OpenZWave Detail, Node031, Query Stage Complete (ManufacturerSpecific2)
OpenZWave Detail, Node031, AdvanceQueries queryPending=0 queryRetries=0 queryStage=Versions live=1
OpenZWave Detail, Node031, QueryStage_Versions
OpenZWave Info, Node031, Requesting Versions
OpenZWave Info, Node031, Requesting Versions for COMMAND_CLASS_NO_OPERATION
OpenZWave Info, Node031, Requesting Versions for COMMAND_CLASS_BASIC
OpenZWave Info, Node031, Requesting Versions for COMMAND_CLASS_THERMOSTAT_MODE
OpenZWave Info, Node031, Requesting Versions for COMMAND_CLASS_THERMOSTAT_SETPOINT
OpenZWave Info, Node031, Requesting Versions for COMMAND_CLASS_MANUFACTURER_SPECIFIC
OpenZWave Info, Node031, Requesting Versions for COMMAND_CLASS_VERSION
OpenZWave Detail, Node031, QueryStage_Instances
OpenZWave Info, Node031, Essential node queries are complete
OpenZWave Detail, Node031, QueryStage_Static
OpenZWave Detail, Node031, Queuing (Query) ThermostatModeCmd_SupportedGet (Node=31): 0x01, 0x09, 0x00, 0x13, 0x1f, 0x02, 0x40, 0x04, 0x25, 0x8e, 0x17
OpenZWave Detail, Node031, Queuing (Query) ThermostatSetpointCmd_SupportedGet (Node=31): 0x01, 0x09, 0x00, 0x13, 0x1f, 0x02, 0x43, 0x04, 0x25, 0x8f, 0x15
OpenZWave Detail, Node031, Queuing (Query) ManufacturerSpecificCmd_Get (Node=31): 0x01, 0x09, 0x00, 0x13, 0x1f, 0x02, 0x72, 0x04, 0x25, 0x90, 0x3b
OpenZWave Detail, Node031, Queuing (Query) VersionCmd_Get (Node=31): 0x01, 0x09, 0x00, 0x13, 0x1f, 0x02, 0x86, 0x11, 0x25, 0x91, 0xdb
OpenZWave Detail, Node031, Queuing (Query) Query Stage Complete (Static)
OpenZWave Detail, Node031, Notification: EssentialNodeQueriesComplete
OpenZWave Detail,
OpenZWave Info, Node031, Sending (Query) message (Callback ID=0x8e, Expected Reply=0x04) - ThermostatModeCmd_SupportedGet (Node=31): 0x01, 0x09, 0x00, 0x13, 0x1f, 0x02, 0x40, 0x04, 0x25, 0x8e, 0x17
OpenZWave Info, Node031, Encrypted Flag is 0
OpenZWave Detail, Node031, Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
OpenZWave Detail, Node031, ZW_SEND_DATA delivered to Z-Wave stack
OpenZWave Detail, Node031, Received: 0x01, 0x07, 0x00, 0x13, 0x8e, 0x00, 0x00, 0x8e, 0xeb
OpenZWave Detail, Node031, ZW_SEND_DATA Request with callback ID 0x8e received (expected 0x8e)
OpenZWave Info, Node031, Request RTT 1432 Average Request RTT 1246
OpenZWave Detail, Node031, Expected callbackId was received
2019-12-15T07:46:33.601Z z2m:Zwave Notification from node 31: Notification - TimeOut (1)
OpenZWave Error, Node031, ERROR: Dropping command, expected response not received after 1 attempt(s)
OpenZWave Detail, Node031, Removing current message
OpenZWave Detail, Node031, Notification: Notification - TimeOut
OpenZWave Detail,
OpenZWave Info, Node031, Sending (Query) message (Callback ID=0x8f, Expected Reply=0x04) - ThermostatSetpointCmd_SupportedGet (Node=31): 0x01, 0x09, 0x00, 0x13, 0x1f, 0x02, 0x43, 0x04, 0x25, 0x8f, 0x15
OpenZWave Info, Node031, Encrypted Flag is 0
OpenZWave Detail, Node031, Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
OpenZWave Detail, Node031, ZW_SEND_DATA delivered to Z-Wave stack
OpenZWave Detail, Node031, Received: 0x01, 0x07, 0x00, 0x13, 0x8f, 0x00, 0x00, 0x86, 0xe2
OpenZWave Detail, Node031, ZW_SEND_DATA Request with callback ID 0x8f received (expected 0x8f)
OpenZWave Info, Node031, Request RTT 1342 Average Request RTT 1294
OpenZWave Detail, Node031, Expected callbackId was received
2019-12-15T07:46:43.602Z z2m:Zwave Notification from node 31: Notification - TimeOut (1)
OpenZWave Error, Node031, ERROR: Dropping command, expected response not received after 1 attempt(s)
OpenZWave Detail, Node031, Removing current message
OpenZWave Detail, Node031, Notification: Notification - TimeOut
OpenZWave Detail,
OpenZWave Info, Node031, Sending (Query) message (Callback ID=0x90, Expected Reply=0x04) - ManufacturerSpecificCmd_Get (Node=31): 0x01, 0x09, 0x00, 0x13, 0x1f, 0x02, 0x72, 0x04, 0x25, 0x90, 0x3b
OpenZWave Info, Node031, Encrypted Flag is 0
OpenZWave Detail, Node031, Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
OpenZWave Detail, Node031, ZW_SEND_DATA delivered to Z-Wave stack
OpenZWave Detail, Node031, Received: 0x01, 0x07, 0x00, 0x13, 0x90, 0x00, 0x00, 0x87, 0xfc
OpenZWave Detail, Node031, ZW_SEND_DATA Request with callback ID 0x90 received (expected 0x90)
OpenZWave Info, Node031, Request RTT 1361 Average Request RTT 1327
OpenZWave Detail, Node031, Expected callbackId was received
2019-12-15T07:46:53.603Z z2m:Zwave Notification from node 31: Notification - TimeOut (1)
OpenZWave Error, Node031, ERROR: Dropping command, expected response not received after 1 attempt(s)
[... 317 lines cut ...]
OpenZWave Info, Node034, Cache Save for Node 34 as its QueryStage_CacheLoad
OpenZWave Info, Node035, Cache Save for Node 35 as its QueryStage_CacheLoad
OpenZWave Info, Node036, Cache Save for Node 36 as its QueryStage_CacheLoad
OpenZWave Info, Node037, Cache Save for Node 37 as its QueryStage_CacheLoad
OpenZWave Info, Node042, Cache Save for Node 42 as its QueryStage_CacheLoad
OpenZWave Info, Node044, Cache Save for Node 44 as its QueryStage_CacheLoad
OpenZWave Info, Node048, Skipping Cache Save for Node 48 as its not past QueryStage_CacheLoad
OpenZWave Info, Node049, Cache Save for Node 49 as its QueryStage_CacheLoad
OpenZWave Info, Node050, Cache Save for Node 50 as its QueryStage_CacheLoad
OpenZWave Info, Node052, Cache Save for Node 52 as its QueryStage_CacheLoad
OpenZWave Info, Node055, Cache Save for Node 55 as its QueryStage_CacheLoad
OpenZWave Info, Node056, Skipping Cache Save for Node 56 as its not past QueryStage_CacheLoad
OpenZWave Info, Node057, Skipping Cache Save for Node 57 as its not past QueryStage_CacheLoad
OpenZWave Info, Node058, Skipping Cache Save for Node 58 as its not past QueryStage_CacheLoad
OpenZWave Info, Node059, Cache Save for Node 59 as its QueryStage_CacheLoad
OpenZWave Info, Node060, Cache Save for Node 60 as its QueryStage_CacheLoad
OpenZWave Info, Node062, Cache Save for Node 62 as its QueryStage_CacheLoad
OpenZWave Info, Node063, Cache Save for Node 63 as its QueryStage_CacheLoad
OpenZWave Info, Node064, Cache Save for Node 64 as its QueryStage_CacheLoad
OpenZWave Info, Node065, Skipping Cache Save for Node 65 as its not past QueryStage_CacheLoad
OpenZWave Info, Node066, Cache Save for Node 66 as its QueryStage_CacheLoad
OpenZWave Info, Node067, Skipping Cache Save for Node 67 as its not past QueryStage_CacheLoad
2019-12-15T07:58:46.749Z z2m:Zwave node 55 ready: - (Siren)
OpenZWave Detail, Node055, Notification: NodeQueriesComplete
```

petergebruers, 2019-12-15:
Please upgrade (or ask the Zwave2MQTT guys to update to latest version, which is 1.6-992-g76e21d80.

Then please do not post snippets from the OZW_Log.txt or any other log, but please zip the complete OZW_Log.txt and drag & drop it here, because the issue might be somewhere else in the log.

I see you've already started discussing that on the the other project, that's good, I post this to put a reference:

https://github.com/OpenZWave/Zwave2Mqtt/issues/172

petergebruers, 2019-12-15:
Forgot to mention, before creating and sending the OZW_Log.txt log, please delete the ozwcache* cache version, to force OZW to interview all devices (from scratch). But that means you'll have to wake up (manually, or wait until the wake up interval of the device passes) all battery nodes, they will remain unknown until you do that. That does not apply to the Eurotronic Spirit, because although it has batteries, it behaves (almost) like a mains powered device, it is a FLiRS (Frequently Listening Routing Slave). I did a quick search on "Popp Solar Powered Outdoor Siren" and that is probably a FLiRS too, the OZW_Log.txt will tell...

You might guess that 2 x issue with a FLiRS device would mean a problem with OpenZWave, but I recently tested a Z-Uno FLiRS and did not spot any problems.

Dinth, 2019-12-15:
Hi @petergebruers. Im not sure if that's a problem with FLiRS, as some of my mains powered devices show same behaviour, for example Fibaro Dimmers. I pasted Eurotronic TRVs as an example because those are odd - half of them show up and the other half does not. 

Here's a full log: https://paste.ubuntu.com/p/tRgsZWnvzp/

petergebruers, 2019-12-15:
> Im not sure if that's a problem with FLiRS

Probably not

> mains powered devices show same behaviour, for example Fibaro Dimmers.

That should not happen, but there was a change in the interview so that's why I'd like to see what happens with 1.6-992-g76e21d80

> Here's a full log: https://paste.ubuntu.com/p/x53csbbtXW/

Thanks. Please be patient, I am slow at this and I am a volunteer so it depends on available time...

> Im deleting ozwcache* on ever restart while im trying to figure it out why my nodes are not detected correctly.

Please only do that for testing purposes, the whole idea of the "cache file" is that OZW progresses through the interview per node, and as soon as one node finishes (certain stages), that data gets written to that file. Next time OZW starts, it will skip a lot of request/answer commands. This reduces startup time.

> Im pretty sure that Popp solar outdoor siren is actually behaving/reporting as mains powered device.

If it is, it will say so in the log, it is a very important property of the device. It will also show in the ozwcache file , see properties listening and frequentListening, eg `listening="false" frequentListening="false"`

petergebruers, 2019-12-15:
Okay, your log has plenty of these and that is not a good sign

`2019-12-14 19:24:00.167 Detail, Node050, CAN received...triggering resend`

I count hundreds of them

That usually indicates a hardware issue, a driver issue or something using the same serial port

You say HASS (so 1.4) is OK and ZWave2mqtt is not so I am guessing that is not on the same system?

Dinth, 2019-12-15:
@petergebruers it is same system, im just swapping Docker containers, stopping one, deleting zwcfg, restarting the system (seems that Zwave2MQTT does not free up all the resources when stopping the container) and starting a new container.
Im just deleting zwcfg for testing and due to frequent changes of versions. I have once tried restarting docker without deleting zwcfg but no new devices have been properly interviewed.

petergebruers, 2019-12-15:
@Fishwaldo I need a second opion on this. The 1.4 log is also full of CAN...

Not sure what to try next.

Fishwaldo, 2019-12-15:
I’m traveling right now so can’t see the logs but CAN = collisions. Could be a faulty node. I’d try powering off everything and progressively power on devices till the CAN’s start again. 

Either that or you have a node spamming the network with updates (power Meter?)

Fishwaldo, 2019-12-15:
> @petergebruers it is same system, im just swapping Docker containers, stopping one, deleting zwcfg, restarting the system (seems that Zwave2MQTT does not free up all the resources when stopping the container) and starting a new container.
> Here's a full log of running the other docker with OZW 1.4: https://paste.ubuntu.com/p/5HFK9Y28yt/
> Im just deleting zwcfg for testing and due to frequent changes of versions. I have once tried restarting docker without deleting zwcfg but no new devices have been properly interviewed.

Btw, OZW 1.4 uses zwcfg files and OZW 1.6 uses ozwcache files.

petergebruers, 2019-12-15:
Thanks @Fishwaldo 

Haven't had time myself to look into this in great detail but have gone through the first few nodes.

That is the log starting with `2019-12-14 19:22:26.108 Always, OpenZwave Version 1.6-962-gcbe2b60f-dirty Starting Up`

Node 6 seems unreachable.

Node 11 and 12 are sleeping.

Node 13 is mains and alive.

Node 14 and 15 are sleeping

Log at mains Node 16 gets interesting.

Node 56 is "chatty" and sends data while OZW is doing NoOp(s)... For example:

```
2019-12-14 19:23:00.234 Info, Node016, Received reply to FUNC_ID_ZW_GET_NODE_PROTOCOL_INFO
2019-12-14 19:23:00.234 Info, Node016,   Protocol Info for Node 16:
2019-12-14 19:23:00.234 Info, Node016,     Listening     = true

...

2019-12-14 19:23:00.235 Info, Node016, Sending (NoOp) message (Callback ID=0x10, Expected Reply=0x13) - NoOperation_Set (Node=16): 0x01, 0x09, 0x00, 0x13, 0x10, 0x02, 0x00, 0x00, 0x25, 0x10, 0xc2
2019-12-14 19:23:00.235 Info, Node016, Encrypted Flag is 0
2019-12-14 19:23:00.242 Detail, Node016,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-12-14 19:23:00.242 Detail, Node016,   ZW_SEND_DATA delivered to Z-Wave stack
2019-12-14 19:23:00.622 Detail, Node056,   Received: 0x01, 0x0a, 0x00, 0x04, 0x10, 0x38, 0x04, 0x30, 0x03, 0xff, 0x0c, 0x1d
2019-12-14 19:23:00.622 Detail, 
2019-12-14 19:23:00.622 Info, Node056, ApplicationCommandHandler - Unhandled Command Class 0x30
2019-12-14 19:23:00.622 Detail, Node016,   Expected callbackId was received
2019-12-14 19:23:00.663 Detail, Node016,   Received: 0x01, 0x07, 0x00, 0x13, 0x10, 0x00, 0x00, 0x2a, 0xd1
2019-12-14 19:23:00.663 Detail, Node016,   ZW_SEND_DATA Request with callback ID 0x10 received (expected 0x00)
2019-12-14 19:23:00.663 Warning, Node016, WARNING: Unexpected Callback ID received
2019-12-14 19:23:00.663 Detail, Node016,   Expected reply was received
2019-12-14 19:23:00.663 Detail, Node016,   Message transaction complete
2019-12-14 19:23:00.663 Detail, 
2019-12-14 19:23:00.663 Detail, Node016, Removing current message
```

There is something odd about this sequence of events imho...

- A NOP gets send (accepted for transmission) with Callback ID=0x10
- Then Node 56 sends data, OZW ignores it ('Unhandled Command') because that node has not been interviewed yet
- OZW receives 0x01, 0x07, 0x00, 0x13, 0x10, 0x00, 0x00, 0x2a, 0xd1 and correctly deduces "ZW_SEND_DATA Request with callback ID 0x10 received" - correct, the hex value after 0x13 is 0x10 - but then continues by saying: "(expected 0x00)". How did that happen?

Then the next node gets interrupted by "node 56" but this time the NOP test succeeds...

```
2019-12-14 19:23:00.666 Info, Node019, Sending (NoOp) message (Callback ID=0x11, Expected Reply=0x13) - NoOperation_Set (Node=19): 0x01, 0x09, 0x00, 0x13, 0x13, 0x02, 0x00, 0x00, 0x25, 0x11, 0xc0
2019-12-14 19:23:00.666 Info, Node019, Encrypted Flag is 0
2019-12-14 19:23:00.675 Detail, Node019,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-12-14 19:23:00.676 Detail, Node019,   ZW_SEND_DATA delivered to Z-Wave stack
2019-12-14 19:23:00.768 Detail, Node056,   Received: 0x01, 0x0a, 0x00, 0x04, 0x10, 0x38, 0x04, 0x30, 0x03, 0xff, 0x0c, 0x1d
2019-12-14 19:23:00.768 Detail, 
2019-12-14 19:23:00.768 Info, Node056, ApplicationCommandHandler - Unhandled Command Class 0x30
2019-12-14 19:23:00.808 Detail, Node056,   Received: 0x01, 0x0a, 0x00, 0x04, 0x10, 0x38, 0x04, 0x30, 0x03, 0xff, 0x0c, 0x1d
2019-12-14 19:23:00.808 Detail, 
2019-12-14 19:23:00.808 Info, Node056, ApplicationCommandHandler - Unhandled Command Class 0x30
2019-12-14 19:23:00.819 Detail, Node024,   Received: 0x01, 0x0a, 0x00, 0x04, 0x10, 0x18, 0x04, 0x30, 0x03, 0xff, 0x0c, 0x3d
2019-12-14 19:23:00.819 Detail, 
2019-12-14 19:23:00.843 Detail, Node056,   Received: 0x01, 0x0a, 0x00, 0x04, 0x10, 0x38, 0x04, 0x30, 0x03, 0xff, 0x0c, 0x1d
2019-12-14 19:23:00.844 Detail, 
2019-12-14 19:23:00.844 Info, Node056, ApplicationCommandHandler - Unhandled Command Class 0x30
2019-12-14 19:23:00.895 Detail, Node056,   Received: 0x01, 0x0a, 0x00, 0x04, 0x10, 0x38, 0x04, 0x30, 0x03, 0xff, 0x0c, 0x1d
2019-12-14 19:23:00.895 Detail, 
2019-12-14 19:23:00.895 Info, Node056, ApplicationCommandHandler - Unhandled Command Class 0x30
2019-12-14 19:23:01.288 Detail, Node056,   Received: 0x01, 0x0a, 0x00, 0x04, 0x10, 0x38, 0x04, 0x30, 0x03, 0xff, 0x0c, 0x1d
[... 45 lines cut ...]
2019-12-14 19:24:14.942 Detail, Node056, QueryStage_Probe
2019-12-14 19:24:14.942 Info, Node056, NoOperation::Set - Routing=true
2019-12-14 19:24:14.942 Detail, Node056, Queuing (NoOp) NoOperation_Set (Node=56): 0x01, 0x09, 0x00, 0x13, 0x38, 0x02, 0x00, 0x00, 0x25, 0x28, 0xd2
2019-12-14 19:24:14.942 Detail, Node056, Queuing (Query) Query Stage Complete (Probe)
2019-12-14 19:24:14.942 Detail, 
2019-12-14 19:24:14.942 Info, Node056, Sending (NoOp) message (Callback ID=0x28, Expected Reply=0x13) - NoOperation_Set (Node=56): 0x01, 0x09, 0x00, 0x13, 0x38, 0x02, 0x00, 0x00, 0x25, 0x28, 0xd2
2019-12-14 19:24:14.942 Info, Node056, Encrypted Flag is 0
2019-12-14 19:24:15.942 Error, Node056, ERROR: Dropping command, expected response not received after 1 attempt(s)
2019-12-14 19:24:15.942 Detail, Node056, Removing current message
2019-12-14 19:24:15.942 Detail, Node056, Notification: Notification - TimeOut
```

That is not looking good either.

But that might be caused by something else happing when we see the firs CAN, that is a bit before that log snippet:

```
2019-12-14 19:24:00.166 Detail, Node050, CAN received...triggering resend
```

@Dinth Can you power down node 56 (hex 0x38), remove ozwcache (so please try 1.6 because we do not support 1.4) and restart and see if the interview gets any better? 

I think @Fishwaldo may be on the right track when he said: "... or you have a node spamming the network with updates".

What about Node 6 and Node 21, they don't respond to a ping test, are powered and reachable?

Dinth, 2019-12-15:
Again, thanks both for looking into this i really do appreciate! 
Nodes 056 and 057 are "Shenzhen Neo Battery Powered PIR Sensor V2" sensors. I will try to remove them and repeat the process shortly.
Please ignore nodes 6, 11, 12 as those dont have batteries right now. Node 52 is weird and always was.
Node 1, 13, 16, 19, 25, 34, 35, 36, 37, 42, 44 seem to be discovered correctly. Im particularly interested in nodes 27, 28, 29, 31, 32. 49, 50, 55, which show correct "Type" but either nothing or Unknown node id 0000 in "Product". The remaining nodes dont show neither Type nor Product.

petergebruers, 2019-12-15:
> Nodes 056 and 057 are "Shenzhen Neo Battery Powered PIR Sensor V2" sensors. I will try to remove them and repeat the process shortly.

Right... Might be a "red herring" then - I expected those to be mains devices. Devices known to me to send quite a few updates, sometimes 5 per minute, are power reporting devices like a Fibaro Wall Plug with default settings and certain fluctuating nodes (that's just an example).

BTW iirc you had quite a few nodes in "secure" node, may I suggest if you re-include them to only do that if the devices can open/close doors, gates or blinds? The S0 protocol has a lot of overhead (instead of doing TX it can be 3-4 and sometimes more) and slows down your network.

> Im particularly interested in nodes 27, 28, 29, 31, 32. 49, 50, 55

Okay, the CAN issue starts at node 50 so that leaves 27 -> 49 to investigate. I'll try to do that tomorrow evening.

Dinth, 2019-12-15:
@petergebruers i took out batteries from nodes 52, 56, 57 as advised. This time Z2M only discovered "Product" of my first node (Controller).
Please see the log: https://paste.ubuntu.com/p/M4yWYWbTFK/

Dinth, 2019-12-16:
Please dsiregard the log above. 
Ive done some maintenance on my system and things seem to be little bit better (more devices got discovered correctly on initial interviews, **more importantly "Refresh Node Info" on the nodes which havent passed initial interview now works!**).
Still getting a lot of CAN errors which im failing to narrow down :( 
Here's the log: https://paste.ubuntu.com/p/Sc8Nq7JvCB/

petergebruers, 2019-12-16:
> Still getting a lot of CAN errors which im failing to narrow down :(

As long as you see a plenty of CAN, nothing is going to work reliably...

I was beginning to doubt my self, so I started my 10-node network and saved the OZW_Log.txt... No CAN, and the few timeouts you'll find can be explained.

[OZW_Log_H_ozwcp.txt.zip](https://github.com/OpenZWave/open-zwave/files/3967458/OZW_Log_H_ozwcp.txt.zip)

No CAN or resend issues.

That is OpenZwave Version 1.6-992-g76e21d80 on a Raspberry Pi 2 (two !) using a ZME USB1 dongle.

I think it is time for you to try something radically different. For example; install Domoticz Beta (which is based on a recent OZW 1.6 version) on a different system eg a Windows PC

https://www.domoticz.com/downloads/

Dinth, 2019-12-16:
Thanks for all help @petergebruers. Hopefully i will have some days off work over the christmas so i will try to fix the can problems, i will try different machine, software and if that wont help i will try excluding all the nodes i can access.

travishaagen, 2019-12-21:
I've had a similar experience. I previously used Home Assistant and it was able to identify node details (e.g., manufacturer), but using OZW 1.6 built from `master` has only been able to identify such details for the controller itself (Aeotec Z-Stick Gen5). I think values like manufacturer are being read as zero for all my devices (Aeotec zw120 door/win sensor, Kwikset Smartcode lock, and a generic toggle power switch).

[before.zip](https://github.com/OpenZWave/open-zwave/files/3991653/before.zip) this is my cache and logs by adding each device individually

[after.zip](https://github.com/OpenZWave/open-zwave/files/3991652/after.zip) then I deleted the cache/logs and started OZW again to see what the result was

Been pulling my hair out, but not sure where to go from here.

Dinth, 2019-12-21:
@petergebruers i have tried windows Domoticz beta, but i cannot find OZW logs literally anywhere. To the point, that i have asked on Domoticz forums and nobody could help me. 

Im also trying to troubleshoot those CAN errors, but i couldnt really find an explanation online why those happen. 
Can it be caused by a faulty node? Can it be caused by an working node but with signal level issues. Can it be caused by few nodes being too close to each other (apparently this can cause issues too). Shall i try to fix it by removing nodes from my mesh or rather add few new nodes to make mesh stronger.

Fishwaldo, 2019-12-22:
CAN = collisions at the RF level. It means two nodes trying to send at the same time. (In this case it’s the stick and one other node). 

Yes - removing nodes to try to isolate CAN issues can help. (Start with any non Zwave plus devices)

petergebruers, 2019-12-22:
> i have tried windows Domoticz beta, but i cannot find OZW logs literally anywhere. To the point, that i have asked on Domoticz forums and nobody could help me.

On the hardware tab, go to the OpenZWave hardware, click setup. Then you get a screen listing al your devices (nodes). Click on the controller and you'll find an option to enable debug logging.

You'll find OZW_Log.txt and ozwcache* in the "Config" folder of your Domoticz installation

> Can it be caused by few nodes being too close to each other (apparently this can cause issues too).

On paper it can cause issues, in real life not so much.

No, I own 45 mains powered nodes and most of them are mounted in pairs, very close to each other (7 - 10 cm between them), and I have checked performance with Zniffer. It is not an issue.
I've been on the Fibaro forum since 2013, I would have heard of such a problem.

petergebruers, 2019-12-22:
> Im also trying to troubleshoot those CAN errors, but i couldnt really find an explanation online why those happen.

That is because up to now that part of the protocol was not published as a "public" specification.

Silabs has said very recently, the will transfer all info to "The Alliance" and make Z-Wave a truly open spec.

If you want to details about SOF/ACK/NAK/CAN you can still obtain the relevant document if you register online. At the moment, I cannot post a copy of that document here yet...

INS12350-17 - Serial API Host Appl. Prg. Guide.pdf.pdf

I have to warn you, most people think SOF/ACK/NAK/CAN are the response of a Z-Wave device but that is only indirectly the case! The confusion arrises because 

Those four packets regulate the flow between the controller and OpenZWave, not between a device and and OpenZWave! But many online resources say Z-Wave has a "ACK" packet on the radio level which is true, but is unrelated to the SOF/ACK/NAK/CAN on the serial level.

The "ACK" on the device level is now as ZW_SEND_DATA "status"

If you  send data to a device... What users commonly understand as "ack" or "no ack" is

petergebruers, 2019-12-22:
Sorry, pressed "enter" too soon...

If you send data to a device... What users commonly understand as "ack" or "no ack" is everything related with ZW_SEND_DATA

For example, if you see this in OZW log:

```
Sending (NoOp) message (Callback ID=0x0a .......
(...)
ZW_SEND_DATA delivered to Z-Wave stack
(...)
WARNING: ZW_SEND_DATA failed. No ACK received - device may be asleep.
```

The "delivered" means the dongle has accepted the packet for transmission (and will transmit it soon)

The "ZW_SEND_DATA failed. No ACK received" means the dongle has reported and error

Technically speaking, although we use the word "ACK" for that it is the result of ZW_SEND_DATA Callback

```cpp
if (_data[3] != TRANSMIT_COMPLETE_OK)
 ````

And the most common failure is TRANSMIT_COMPLETE_NO_ACK

This is the complete list:

```cpp
/* Transmit complete codes */
#define TRANSMIT_COMPLETE_OK      0x00
#define TRANSMIT_COMPLETE_NO_ACK  0x01  /* retransmission error */
#define TRANSMIT_COMPLETE_FAIL    0x02  /* transmit error */
#define TRANSMIT_ROUTING_NOT_IDLE 0x03  /* transmit error */
/* Assign route transmit complete but no routes was found */
#define TRANSMIT_COMPLETE_NOROUTE 0x04  /* no route found in assignroute */
```

It comes as no surprise the names are confusing.

Anyway... If you see NAK/CAN as @Fishwaldo points out one of the possible causes is that OpenZWave and the controller (dongle) were sending data at the same time, and because the controller sends data to OpenZWave when it receives data from a device, that is a possible cause. But it is also possible for some reason the "serial protocol" lost sync. So far we have been able to identify thesed other causes of (lots of CAN/NAK):
- hardware issues
- Running "modemmanager" on Ubuntu, presumably because it "steals" ar "adds" bytes to the serial port
- Running a second instance of OpenZWave (should detect that on  recent version) on a Linux like system.

On windows, you get an exclusive lock on the COM port, that's one of the reasons why I've mentioned running Domoticz on windows.

It is not impossible to have a CAN or NAK on a healthy network, but the number should be very low.

petergebruers, 2019-12-22:
On a more practical level... If you don't own and use a Zniffer, you might _never_ be able to fully understand what is going on your network. OZW_Log is good but it is no substitute, a single line can turn into many Z-Wave packets, nothing we can do about that. The controller hides the details. 

Your best bet may be: properly delete (exclude) all your nodes... Reset your controller... Start from scratch. This means: start by adding your Z-Wave Plus mains devices first and add them so you spread them around your controller, because those are likely to form a good backbone. Do a heal so they are very happy nodes. Then add your battery devices. Go slowly and be careful when changing parameters on your devices.

To be honest... I don't fully understand what causes the behavior seen in your OZW_Log file. And although I have tried to replicate that, I don't get such behavior...

Dinth, 2019-12-23:
> On windows, you get an exclusive lock on the COM port, that's one of the reasons why I've mentioned running Domoticz on windows.

Hi. I finally managed to get OZW log from Windows Domoticz.
https://paste.ubuntu.com/p/yDd7dCVZt9/
Still CAN errors, but i believe little bit less (200+) ?
I just bought a fibaro controller (to update my fibaro nodes), after christmas i will start excluding nodes and updating them. 

Kind regards

petergebruers, 2019-12-25:
@Dinth thank you for trying that and posting the log. Indeed, still too many CANs and I am still convinced it has got to do with your "busy" network (meaning eg you have lots of sensors or modules reporting power) but I also noticed OZW may stay too long in CAN handling... I've opened a PR for that, if you know how to how to use your own OZW build with Zwave2MQTT you can try that branch on my PR (see #2049).

Fishwaldo, 2020-01-21:
I just merged the commit. Please test and let us know.

closed 2020-02-04 by no-response[bot]