OpenZWave/open-zwave #2039: Update config/2gig/ct101.xml

eresonance, 2019-12-18:
I'm going to update the Radio Thermostat CT101 xml file to include metadata and config stuff. I assume it's similar to the other devices under 2gig but the manual is pretty explicit too.

Will review this doc before I do: https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

Docs:
https://products.z-wavealliance.org/products/1301
https://static.wixstatic.com/ugd/94b5a7_23ba26a52a1e4f5498afa36d1552ed07.pdf

[image]

Horray for xmas holidays!

Fishwaldo, 2019-12-25:
Have you read this? https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices  When you made the changes you can open a PR

closed 2019-12-25 by Fishwaldo

eresonance, 2019-12-26:
Sorry, I was going to do step 3 of the "submitting a change to us":
> 3 if you had downloaded a copy of our stable/snapshot releases, then just create a new issue on our bug tracker, listing the Manufacturer, Product Type, Product ID, Product Name, and email any configuration file you had created to our mailing list
I was planning to test this via the ruimarinho/openzwave docker image so I don't have to try and run OZW from source.

melyux, 2020-06-25:
Did you ever get around to doing this? I wanted to do the same thing but happened upon this issue.

eresonance, 2020-06-30:
No, I couldn't get this project working from source. I was going to take a look at making the config change in the docker image then copy+paste it into the repo to make a PR, but decided to spend my time elsewhere. I'll eventually get back around to this.