OpenZWave/open-zwave #2042: [feat] Tag stable commits more frequently with OZW version

robertsLando, 2019-12-20:
It is not very clear how to keep ozw updated to latest version, I opened an issue some weeks ago about this and I ended up using releases from http://old.openzwave.com/downloads/ but seems they are not updated very frequently.

I suggest to tag commits more frequently with partials versions to allow users to identify stable commits, actually only minor versions are tagged, why not tag also patch versions?

Fishwaldo, 2019-12-21:
The releases from old.openzwave.com are updated after going through build tests etc. There is nothing wrong there. The last commit to the master branch was 9th of December. 

Its automated - and the build infra doesn't use Github releases etc for this. As mentioned countless times - stable, tested releases are on the downloads page.

closed 2019-12-21 by Fishwaldo

robertsLando, 2020-07-20:
@Fishwaldo Is there any possibility to at least to tag stable releases? For stable I means releases that are puublished on old.openzwave.com, if you tag commits or (even better) you create a `stable` branch that is updated everytime a new release goes out it would be lot easier to handle. Thoughts? 

https://github.com/OpenZWave/Zwave2Mqtt/issues/626