OpenZWave/open-zwave #2045: Popp Rain Sensor not detected properly on OZW 1.6

docbobo, 2019-12-21:
I own a Popp Rain Sensor, that OZW 1.4 detected correctly as such. OZW 1.6 failes to detect the device properly, reporting both manufacturer and product as Unknown: id=0000. In addition, the metadata contained in the ozwcache file is significantly smaller than the metadata contained in the old zwcfg file.

Can you please let me know which information you would need to further track this down?

petergebruers, 2019-12-22:
Did you wake up the device's after you've added it? It might require one or more wake-up. The manuals says "Push the button once. The LED blinks once indicating that Z- Weather is active."

If that does not help, please delete(exclude) the device. Set your software to produce a "debug log". Restart your software (Domoticz? Zwave2mqtt?). Add the device and wake it up a few times. Zip the OZW_Log.txt and drag and drop it here, it contains lots of lovely details.

docbobo, 2020-01-03:
Okay. I left it untouched for about a week, while I was on vacation. Now everything seems fine. Closing.

closed 2020-01-03 by docbobo