OpenZWave/open-zwave #2053: Heiman motion sensor HS1MS-Z problem

Tomasz01, 2019-12-27:
I test Heiman Smart Motion Sensor, but I've a problem. Sensors is working, react for motion, but variable Binary Sensor / Sensor_1 is always FALSE. I test several pieces, they all behave the same. The sensor works, for example the sleep variable changes when it detects motion. 

I use openzwave-1.6.986 with iobroker. Other devices z-wave works ok.

petergebruers, 2019-12-27:
Try to add the controller (1) to associaton group 3. According to the manual the lifeline uses notification instead of sensor binary. If that does not help, please zip your OZW_Log.txt of your test(s) and drag and drop it here for manual analysis. Please be very patient, I am on holiday.

Tomasz01, 2019-12-27:
If I add association, variable Sensor_1 becomes true, after react for motion, but only first. Later, variable not refresh, always true.

petergebruers, 2019-12-27:
What a weird device... The manual is in "chinglish" and I am nut sure I understand 😉 

```
grep "SensorBinary report" OZW_Log.txt
2019-12-27 17:10:18.218 Info, Node002, Received SensorBinary report: Sensor:12 State=Off
2019-12-27 17:10:18.259 Info, Node002, Received SensorBinary report: Sensor:12 State=Off
2019-12-27 17:10:21.223 Info, Node002, Received SensorBinary report: Sensor:8 State=On
2019-12-27 17:10:30.158 Info, Node002, Received SensorBinary report: Sensor:8 State=On
2019-12-27 17:10:37.318 Info, Node002, Received SensorBinary report: Sensor:8 State=On
2019-12-27 17:10:43.403 Info, Node002, Received SensorBinary report: Sensor:8 State=On
2019-12-27 17:10:52.264 Info, Node002, Received SensorBinary report: Sensor:8 State=On
2019-12-27 17:10:57.900 Info, Node002, Received SensorBinary report: Sensor:8 State=On
```

You have to know... "12" means "Motion" and "8" means Tamper. I expect you to have two "values" in iobroker but you only mention "Sensor_1". Do you have a "Sensor_2" as well?

Not sure if that fixes you problem because the reporting still looks a bit odd to me, it starts by saying "Motion Off" but never "On". And after that the device says "tamper On" several times, but never "Off". The decoding of the packets is correct, OZW can only tell to iobroker what your sensor reports to OpenZWave.

You say...

> I use openzwave-1.6.986 with iobroker. Other devices z-wave works ok.

But your logfile says something else...

OpenZwave Version 1.6-918-g51474a5a-dirty

Tomasz01, 2019-12-27:
Hmmmm... Im not remember how I installed openzwave. Now, I download for Debian 9:

openzwave-dbgsym_1.6.994-0_amd64.deb
libopenzwave1.6_1.6.994-0_amd64.deb
openzwave_1.6.994-0_amd64.deb

And install. All of them must install? I annex new log. 

I tested alternative iobroker.zwave2 package and adapter, and actually was two variables: Motion (always false) and Tamper (always true). With iobroker.zwave openzwave-shared@1.5.10 (tested also 1.5.7, 1.5.8), was only one variable Sensor_1 in Binary Sensor section.

[OZW_Log.txt.zip](https://github.com/OpenZWave/open-zwave/files/4005980/OZW_Log.txt.zip)

petergebruers, 2019-12-27:
> I tested alternative iobroker.zwave2 package and adapter, and actually was two variables: Motion (always false) and Tamper (always true).

Well, that matches the OZW_Log and if that is what the device reports... I am afraid I cannot help you fix the "binary reports".

In your previous log I saw "Alarm" notifications, I no longer see them in your new log.

Please remove the device from your network. Stop and start iobroker. Add the device. Trigger and wait for it to "un-trigger" eg make sure there is no motion for X minutes. Then try to add it to group 2, group 3 and trigger + wait again. Then zip the OZW_Log.txt. It will have more information because it also contains the "interview" of the device when you added it.

But to be honest... I might not be able to help you with this device... I am only a volunteer, I do not own this device and frankly I don't know why it behaves like that. So I might be wasting your time.

But maybe @Fishwaldo has an idea.

> Hmmmm... Im not remember how I installed openzwave. Now, I download for Debian 9:
> 
> openzwave-dbgsym_1.6.994-0_amd64.deb
> libopenzwave1.6_1.6.994-0_amd64.deb

Errr... Probably "libopenzwave" and that probably worked because now I see:

`OpenZwave Version 1.6.-1 Starting Up`

And that means the build server had trouble determining the version with "git describe" but that is not important here.

Tomasz01, 2019-12-27:
> Please remove the device from your network. Stop and start iobroker. Add the device. Trigger and wait for it to "un-trigger" eg make sure there is no motion for X minutes. Then try to add it to [...]

I did. New log.
[OZW_Log.txt.zip](https://github.com/OpenZWave/open-zwave/files/4006045/OZW_Log.txt.zip)

petergebruers, 2019-12-27:
Okay, interview looks clean to me.

In the log:

`Received SensorBinary report: Sensor:8 State=On`

Again... no "Off", and that is "tamper", not "motion".

Sorry, I won't be able to help you with this. I'm out...

Tomasz01, 2019-12-28:
Thank you for your help. Anyone else have any ideas?

Fishwaldo, 2020-01-21:
You may need to contact the manufacturer and ask them... we don't have this device so there is no way to say if this is "normal or not" but it does not appear to be like any other devices/vendors on the market.

closed 2020-01-21 by Fishwaldo

GitDuff, 2021-02-10:
I know it's an old closed thread, but I just installed one of these on my network a few days ago and thought it good to update the thread as I landed here when I had some issues trying to receive the state from the binary sensor.

By default, the binary sensor is always "off" regardless of motion, but after I add the controller (1) to associaton group 3, then the "on" state is received.

After a bit of testing, the "off" state is received ok, but it appears to only comes after 2 minutes of no motion. I can't see that it's possible to adjust the 2 minutes though from the options.