OpenZWave/open-zwave #2054: Qubino zmnhtdx/ zmnhxdx meter valueid for import (0x21) and export (0x41) energy

thecem, 2019-12-27:
I get for one valueid (in homeassistant) for both meters different energy values. I get for both values different hex values, and even the OZW_Log show as **Energy** and **Exporting Energy**. But the valueID is for both the same.

Node027, Received: 0x01, 0x12, 0x00, 0x04, 0x00, 0x1b, 0x0a, 0x32, 0x02, 0x21, 0x24, 0x00, 0x00, 0x00, 0x06, 0x00, 0x00, 0xc4, 0x00, 0x0f
Node027, Received Meter report from node 27: Energy=0.6kWh
Node027, Received: 0x01, 0x12, 0x00, 0x04, 0x00, 0x1b, 0x0a, 0x32, 0x02, 0x41, 0x24, 0x00, 0x00, 0x05, 0x65, 0x00, 0x00, 0xc4, 0x00, 0x09
Node027, Received Meter report from node 27: Exporting Energy=138.1kWh

Following the silabs standard Z-Wave Application Command Class Specification the values send from qubino device are correct:

0x32 ID 50 Command Class Meter
0x02 Command Meter_Report
_____Scale2(1)__Rate Type(2)__Meter Type (5)
0x41 0_________10___________00001
_______________Export_______Electric meter
or
0x21 0_________01___________00001
_______________Import________Electric meter

_____Precision(3)_Scale1(2)___Size
0x24 001_______00__________100
_______________kWh

Is the hex value 0x21 and 0x41 on this position included in the valueID calculation?
And if not, is there a possibility to distinguish between both values for import an export energy?
Is there a possibility to calculate the valueID following the standard ?

Thanks a lot!

Fishwaldo, 2020-01-21:
The export valueid indicates if the values represent producing or consuming energy. There is no need for seperate values as you can't produce and consume at the same time.

closed 2020-01-21 by Fishwaldo

thecem, 2020-01-21:
Thanks for your comment.
But with which parameter I will see if the valueID is presenting producing or consuming. Out of the standard I could distinguish, between both? Is there some information?

Fishwaldo, 2020-01-21:
The Exporting ValueID.

thecem, 2020-01-22:
Sorry for the question, and the Importing? With the same ValueID ? In the Logs of OZW I see the difference between import and export but with the same ValueID and HA is not able to distinguish between importing and exporting with the same ValueID.

Fishwaldo, 2020-01-22:
I suggest you use ozwcp to see it. I have no idea if it’s exposed in HASS. 

It’s a Boolean value. True or false If the other values represent exporting/importing values. 

If true, the values represent what you are producing, if false, the values represent what you are consuming.