OpenZWave/open-zwave #2062: Random negative values for Z-Wave plug power consumption

jacekpaszkowski, 2020-01-08:
Hi, im using Hass.io on RPi3 with Aeotec Z-Stick Gen 5.
HA is using OpenZwave Version 1.4.3452.
The problem is with these devices:
- Shenzhen Neo Electronics Co Ltd NAS-WR01Z Power plug 12A
- Shenzhen Neo Electronics Co Ltd NAS-WR01ZE Power plug 12A

I have issues with negative values reported for power consumption (kWh) entities in HA. I also see those values in OZW_Log.txt file.
I don't know if it is OZW or device issue? Or maybe there is a fix in never versions of OpenZwave?

Here is some log entries (and here is full log [OZW_Log_08.01.2019.zip](https://www.dropbox.com/s/n89fuanf6y8r0kd/OZW_Log_08.01.2019.zip?dl=0)):
```
2020-01-08 07:10:54.300 Detail, Node013,   Received: 0x01, 0x14, 0x00, 0x04, 0x00, 0x0d, 0x0e, 0x32, 0x02, 0x21, 0x44, 0x80, 0x00, 0x12, 0xe1, 0x00, 0x3d, 0x80, 0x00, 0x12, 0xe1, 0x84
2020-01-08 07:10:54.300 Detail, 
2020-01-08 07:10:54.301 Detail, Node013, Refreshed Value: old value=false, new value=false, type=bool
2020-01-08 07:10:54.301 Detail, Node013, Changes to this value are not verified
2020-01-08 07:10:54.301 Info, Node013, Received Meter report from node 13: Energy=-21474788.15kWh
2020-01-08 07:10:54.301 Detail, Node013, Refreshed Value: old value=-21474788.15, new value=-21474788.15, type=decimal
2020-01-08 07:10:54.301 Detail, Node013, Changes to this value are not verified
2020-01-08 07:10:54.301 Info, Node013,     Previous value was -21474788.15kWh, received 61 seconds ago.
2020-01-08 07:10:54.301 Detail, Node013, Refreshed Value: old value=-21474788.15, new value=-21474788.15, type=decimal
2020-01-08 07:10:54.301 Detail, Node013, Changes to this value are not verified
2020-01-08 07:10:54.301 Detail, Node013, Refreshed Value: old value=61, new value=61, type=int
2020-01-08 07:10:54.301 Detail, Node013, Changes to this value are not verified
2020-01-08 07:10:54.301 Detail, Node013, Notification: ValueChanged
2020-01-08 07:10:54.323 Detail, Node013, Notification: ValueChanged
2020-01-08 07:10:54.338 Detail, Node013, Notification: ValueChanged
2020-01-08 07:10:54.353 Detail, Node013, Notification: ValueChanged
[......]
2020-01-08 07:11:54.299 Detail, Node013,   Received: 0x01, 0x14, 0x00, 0x04, 0x00, 0x0d, 0x0e, 0x32, 0x02, 0x21, 0x44, 0x00, 0x00, 0x12, 0xe3, 0x00, 0x3d, 0x00, 0x00, 0x12, 0xe3, 0x84
2020-01-08 07:11:54.299 Detail, 
2020-01-08 07:11:54.299 Detail, Node013, Refreshed Value: old value=false, new value=false, type=bool
2020-01-08 07:11:54.299 Detail, Node013, Changes to this value are not verified
2020-01-08 07:11:54.299 Info, Node013, Received Meter report from node 13: Energy=48.35kWh
2020-01-08 07:11:54.299 Detail, Node013, Refreshed Value: old value=-21474788.15, new value=48.35, type=decimal
2020-01-08 07:11:54.299 Detail, Node013, Changes to this value are not verified
2020-01-08 07:11:54.299 Info, Node013,     Previous value was 48.35kWh, received 61 seconds ago.
2020-01-08 07:11:54.299 Detail, Node013, Refreshed Value: old value=-21474788.15, new value=48.35, type=decimal
2020-01-08 07:11:54.300 Detail, Node013, Changes to this value are not verified
2020-01-08 07:11:54.300 Detail, Node013, Refreshed Value: old value=61, new value=61, type=int
2020-01-08 07:11:54.300 Detail, Node013, Changes to this value are not verified
2020-01-08 07:11:54.300 Detail, Node013, Notification: ValueChanged
2020-01-08 07:11:54.327 Detail, Node013, Notification: ValueChanged
2020-01-08 07:11:54.342 Detail, Node013, Notification: ValueChanged
2020-01-08 07:11:54.357 Detail, Node013, Notification: ValueChanged
[......]
2020-01-08 18:12:47.603 Detail, Node005,   Received: 0x01, 0x14, 0x00, 0x04, 0x00, 0x05, 0x0e, 0x32, 0x02, 0x21, 0x54, 0x00, 0x00, 0x0a, 0xa6, 0x00, 0x3d, 0x00, 0x00, 0x0a, 0xbd, 0x87
2020-01-08 18:12:47.604 Detail, 
2020-01-08 18:12:47.604 Detail, Node005, Refreshed Value: old value=false, new value=false, type=bool
2020-01-08 18:12:47.604 Detail, Node005, Changes to this value are not verified
2020-01-08 18:12:47.604 Info, Node005, Received Meter report from node 5: Power=27.26W
2020-01-08 18:12:47.604 Detail, Node005, Refreshed Value: old value=27.49, new value=27.26, type=decimal
2020-01-08 18:12:47.604 Detail, Node005, Changes to this value are not verified
2020-01-08 18:12:47.604 Info, Node005,     Previous value was 27.49W, received 61 seconds ago.
2020-01-08 18:12:47.605 Detail, Node005, Refreshed Value: old value=27.18, new value=27.49, type=decimal
2020-01-08 18:12:47.605 Detail, Node005, Changes to this value are not verified
2020-01-08 18:12:47.605 Detail, Node005, Refreshed Value: old value=61, new value=61, type=int
2020-01-08 18:12:47.605 Detail, Node005, Changes to this value are not verified
2020-01-08 18:12:47.605 Detail, Node005, Notification: ValueChanged
2020-01-08 18:12:47.621 Detail, Node005, Notification: ValueChanged
2020-01-08 18:12:47.657 Detail, Node005, Notification: ValueChanged
2020-01-08 18:12:47.688 Detail, Node005, Notification: ValueChanged
[......]
2020-01-08 18:12:47.711 Detail, Node005,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x05, 0x0a, 0x32, 0x02, 0xa1, 0x42, 0x60, 0xde, 0x00, 0x3d, 0x60, 0x83, 0x57
2020-01-08 18:12:54.046 Detail, Node013,   Received: 0x01, 0x14, 0x00, 0x04, 0x00, 0x0d, 0x0e, 0x32, 0x02, 0x21, 0x44, 0x00, 0x00, 0x12, 0xeb, 0x00, 0x3d, 0x00, 0x00, 0x12, 0xeb, 0x84
2020-01-08 18:12:54.046 Detail, 
2020-01-08 18:12:54.046 Detail, Node013, Refreshed Value: old value=false, new value=false, type=bool
2020-01-08 18:12:54.046 Detail, Node013, Changes to this value are not verified
2020-01-08 18:12:54.046 Info, Node013, Received Meter report from node 13: Energy=48.43kWh
2020-01-08 18:12:54.046 Detail, Node013, Refreshed Value: old value=48.43, new value=48.43, type=decimal
2020-01-08 18:12:54.046 Detail, Node013, Changes to this value are not verified
[... 23 lines cut ...]
2020-01-08 18:13:47.541 Detail, Node005, Notification: ValueChanged
2020-01-08 18:13:47.557 Detail, Node005, Notification: ValueChanged
2020-01-08 18:13:47.573 Detail, Node005, Notification: ValueChanged
[.....]
2020-01-08 18:13:54.046 Detail, Node013,   Received: 0x01, 0x14, 0x00, 0x04, 0x00, 0x0d, 0x0e, 0x32, 0x02, 0x21, 0x44, 0x00, 0x00, 0x12, 0xeb, 0x00, 0x3d, 0x00, 0x00, 0x12, 0xeb, 0x84
2020-01-08 18:13:54.046 Detail, 
2020-01-08 18:13:54.046 Detail, Node013, Refreshed Value: old value=false, new value=false, type=bool
2020-01-08 18:13:54.046 Detail, Node013, Changes to this value are not verified
2020-01-08 18:13:54.046 Info, Node013, Received Meter report from node 13: Energy=48.43kWh
2020-01-08 18:13:54.046 Detail, Node013, Refreshed Value: old value=48.43, new value=48.43, type=decimal
2020-01-08 18:13:54.046 Detail, Node013, Changes to this value are not verified
2020-01-08 18:13:54.046 Info, Node013,     Previous value was 48.43kWh, received 61 seconds ago.
2020-01-08 18:13:54.047 Detail, Node013, Refreshed Value: old value=48.43, new value=48.43, type=decimal
2020-01-08 18:13:54.047 Detail, Node013, Changes to this value are not verified
2020-01-08 18:13:54.047 Detail, Node013, Refreshed Value: old value=61, new value=61, type=int
2020-01-08 18:13:54.047 Detail, Node013, Changes to this value are not verified
2020-01-08 18:13:54.047 Detail, Node013, Notification: ValueChanged
2020-01-08 18:13:54.074 Detail, Node013, Notification: ValueChanged
2020-01-08 18:13:54.089 Detail, Node013, Notification: ValueChanged
2020-01-08 18:13:54.105 Detail, Node013, Notification: ValueChanged
```

Hope somebody would help.

Jacek.

petergebruers, 2020-01-08:
It is a device or network issue, the packet data received by OZW is:

0x8000016e

That is

-2147483282

And the scale factor is 100 so you get

-21474832.82

OZW faithfully reports what it gets from the Z-Wave network... Nothing we can do about that. It has even checksum and checksum matches (otherwise OZW would not accept the data).

It has been mentioned before (usually on Neo Wall Plugs):

https://github.com/OpenZWave/open-zwave/issues/1724

Try to unplug those devices an see if the report sane values again.

closed 2020-01-08 by petergebruers

jacekpaszkowski, 2020-01-08:
thank you for clarification

markruys, 2021-03-25:
It seems to be a bug in this type of devices. I get these consecutive kWh metering values reported:
```
       37.86          0x00000ECA
-21474798.43  0xFFFFFFFF80000EDD
       38.26          0x00000EF2
-21474798.11  0xFFFFFFFF80000EFD
       38.40          0x00000F00
```
So what happens I think is that the the most significant bit is randomly being set. OZW expects a 32-bit signed int, hence the negative readings. According to the docs of this device, the maximum report value is  21474836.47 kWh, which is `0x7FFFFFFF`. The only work around to this device bug would be that somewhere in the software stack this most significant bit is reset to 0. Then we would have:
```
37.86
38.05
38.26
38.37
38.40
```